fwupd/fwupd #3564: X1C6 failing to update with `Device Error` message

gkelly, 2021-07-29:
**Describe the bug**
My Lenovo X1 Carbon Gen 6 is failing to apply updates to the ME, System Firmware, and UEFI Device Firmware. The `fwupdmgr update` command identifies and installs the pending updates. On reboot, the machine stays on the Lenovo splash screen for a few minutes, then flashes two lines _very quickly_ in the upper left:

```
WARNING: Could not apply capsule update: Device Error
WARNING: Could not apply capsules: Device Error
```

This is fast enough that I had to record my screen and step frame by frame through the video to read it.

**Steps to Reproduce**
Attempt a `fwupdmgr update`.

**Expected behavior**
Firmware updates.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
client version:	1.5.8
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.8
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): apt

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
20KH002KUS
│
├─Embedded Controller:
│     Device ID:          9698faabddf0d7b18925cfbbda95f8b0d0dacc53
│     Current version:    0.1.22
│     Minimum Version:    0.1.22
│     Vendor:             DMI:LENOVO
│     GUIDs:              3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│                         7090c94a-ee40-593e-91bc-8cfd5a8c0a5d
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
├─Intel Management Engine:
│     Device ID:          e563ad307df81c99f0de8c26292afd71cf409673
│     Current version:    184.80.3746
│     Minimum Version:    184.80.3746
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUIDs:              42a0a96e-c9f3-438f-9687-7826be33e4ce
│                         375afb87-ea51-5a6e-96d5-c35cc342dc65
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
├─Intel(R) Core™ i7-8650U CPU @ 1.90GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000ea
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
[... 418 lines cut ...]
├── initrd.img-5.11.0-25-generic
├── initrd.img-5.8.0-63-generic
├── initrd.img.old -> initrd.img-5.8.0-63-generic
├── lost+found [error opening dir]
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.11.0-25-generic
├── System.map-5.8.0-63-generic
├── vmlinuz -> vmlinuz-5.11.0-25-generic
├── vmlinuz-5.11.0-25-generic
├── vmlinuz-5.8.0-63-generic
└── vmlinuz.old -> vmlinuz-5.8.0-63-generic

5 directories, 297 files
```

**Additional questions**

- Operating system and version: Ubuntu 21.04
- Have you tried rebooting? Yes.
- Is this a regression? Yes.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled? *No*
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? It is a Lenovo, I disabled "Boot Order Lock' when trying to fix this issue.

superm1, 2021-07-29:
Can you please try to enable legacy option rom and see if it helps?

hughsie, 2021-07-29:
> Can you please try to enable legacy option rom and see if it helps?

Out of interest, what's your suspicion?