fwupd/fwupd #3568: FakeDevice update time

binaryNewts, 2021-07-29:
**Is your feature request related to a problem? Please describe.**
When doing tests on fwupd that require testing of something while an update is running, it's helpful to have something that takes a couple seconds to update. The FakeDevice integrated webcam provided, however, takes about a second to update, which is not very much time to do testing during.

**Describe the solution you'd like**
It would be nice to have either
- a way to set the amount of time that the device sleeps for
- a flag that would stop the update somewhere in the middle and continue when prompted

**Additional context**
Here's where the amount of time is currently set
https://github.com/fwupd/fwupd/blob/master/plugins/test/fu-plugin-test.c#L164

(@djcampello @bleungatchromium @g2pop)

hughsie, 2021-07-29:
> a way to set the amount of time that the device sleeps for

Could you parse the payload? e.g. parse it as a json or XML file with `<delay_seconds>130<delay_seconds>`

> a flag that would stop the update somewhere in the middle

Something like DeviceRequest() would work well for this.

superm1, 2021-07-29:
I think parsing the fake device payload is a great idea.
On that note - We are currently parsing `FWUPD_PLUGIN_TEST` to determine what tests to run.  If you shift this over to add delays from the payload, please adjust the plugin to get this information from the payload too.

bleungatchromium, 2021-07-29:
Hi @hughsie and @superm1 

Do you have an example where similar values are pulled from a payload metadata?

I assume this is coming from https://fwupd.org/downloads/3fab34cfa1ef97238fb24c5e40a979bc544bb2b0967b863e43e7d58e0d9a923f-fakedevice124.cab right?

So the idea is that if we want to run a test and customize the fake delay, we'd source a different .cab file with an extra field to tell the plugin how long to spin?

superm1, 2021-07-29:
>Do you have an example where similar values are pulled from a payload metadata?

Right now `data/installed-tests/fakedevice124.bin` is just ASCII, and a value.  The idea being proposed is to make that more.
```
<firmware>
<version>0x1020004</version>
<test>foo</test>
<delay>0</delay>
</firmware>
```

If that turns out to not scale as well as we were thinking, you can certainly say BAD IDEA, and then maybe an environment variable in the fwupd environment that the test plugin picks up is better/easier.

hughsie, 2021-07-30:
I think setting the environment variables once during self test startup is fine, but I think changing them during self test runtime is going to make it a bit like spaghetti code that's really hard to debug.

hughsie, 2021-07-30:
Does it have to be fakedevice124.cab? The sha1sum of various bits of that is kinda baked into other tests.

bleungatchromium, 2021-07-30:
Hi @hughsie 

Alright, can we go about the environment variable way? I see we use g_getenv in a few places, so presumably we could just set an env for FWUPD_PLUGIN_TEST_INSTALL_MS or something before invoking our test?

hughsie, 2021-07-30:
Yup, do the env var thing and if/when we need to make it less of a hack we can convert it then.

hughsie, 2022-01-21:
I think we did this in the other MR. Closing.

closed 2022-01-21 by hughsie