fwupd/fwupd #3573: "no valid release URIs" on XPS9300 for firmware update 1.6.3

blauch, 2021-07-30:
**Describe the bug**

I am using fwupdmgr for quite some time and have not had issue thus far. But now I am getting an error when trying to apply the latest firmware update that there are "no valid release URIs"

**Steps to Reproduce**
```shell
sudo fwupdmgr refresh
```
```
sudo fwupdmgr update

Devices with no available firmware updates: 
 • Integrated Webcam HD
 • PC611 NVMe SK hynix 512GB
 • UEFI dbx
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 1.5.0 to 1.6.3?                                 ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║ • Firmware updates to address the Intel Security Advisory.                   ║
║ • Firmware updates to address security vulnerabilities.                      ║
║ XPS 13 9300 must remain plugged into a power source for the duration of the  ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: Y
no valid release URIs
```

**Expected behavior**

Firmware update is applied

**fwupd version information**

```
$ pacman -Qs fwupd
local/fwupd 1.6.1-2
    Simple daemon to allow session software to update firmware
local/fwupd-efi 1.1-1
    EFI Application used by uefi-capsule plugin in fwupd
```

```shell
fwupdmgr --version
client version:	1.6.1
compile-time dependency versions
	gusb:	0.3.7

daemon version:	1.6.1
```

I installed it using `pacman`

**fwupd device information**

```
fwupdmgr get-devices --show-all-devices
XPS 13 9300
│
├─Integrated Webcam HD:
│     Device ID:          4295296d98b3ba38c72f6baa33d24f03a1d428f6
│     Current version:    96.28
│     Vendor:             CN0PW36V8LG0002MA8HRA00 (USB:0x0BDA)
│     GUIDs:              d784b200-4ead-55f2-8277-15e1618cbe7b
│                         9a1becb6-4d9f-5855-99d0-145b1700da10
│                         d35eb209-22f9-59a5-a918-33d7307edab9
│     Device Flags:       • Updatable
│   
├─Intel(R) Core™ i7-1065G7 CPU @ 1.30GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000a6
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         d70b62bc-d9c6-5156-bd86-4f7440ccb34d
│                         95b2821b-aba1-51cc-ac93-bf7518dbceb0
│     Device Flags:       • Internal device
│   
├─Iris Plus Graphics G7:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    07
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              9fd5b18d-ff8a-5316-80b6-6e859c3dcd01
│                         b6f99472-4218-524e-81e5-e149f0376b95
│                         c213be4c-faf8-5fe6-9430-b458722b2656
│                         43ede583-c357-55e5-854d-f6f15dab900d
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─PC611 NVMe SK hynix 512GB:
│     Device ID:          ac1ce3be70e444b92f99fe08c9b957efb9d0ee53
│     Summary:            NVM Express Solid State Drive
│     Current version:    11000111
│     Vendor:             SK hynix (NVME:0x1C5C)
│     GUIDs:              919c0a0c-88f2-518c-be84-39f0c34c5104
│                         89f21a68-b326-11e9-a2a3-2a2ae2dbcce4
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Current version:    1.5.0
│ │   Minimum Version:    1.5.0
│ │   Vendor:             Dell Inc. (DMI:Dell Inc.)
│ │   GUIDs:              e8292593-e66e-4878-b051-f152535ab130
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       29a2266e-9abb-57c9-b89e-58f2093f917b
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI Revocation Database
│       Current version:  77
│       Minimum Version:  77
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            00fe3755-a4d8-5ef7-ba5f-47979fbb3423
│                         4a6cd2cb-8741-5257-9d1f-89a275dacca7
│                         c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│     
└─TPM:
      Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
      Current version:    74.8.17568.5511
      Vendor:             ST Microelectronics (TPM:STM|PCI:0x1028)
      GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6
                          84df3581-f896-54d2-bd1a-372602f04c32
                          bfaed10a-bbc1-525b-a329-35da2f63e918
                          70b7b833-7e1a-550a-a291-b94a12d0f319
                          06f005e9-cb62-5d1a-82d9-13c534c53c48
      Device Flags:       • Internal device
    

```

**Additional questions**

- Operating system and version: 
Manjaro Linux Pahvo (21.1.0)

- Have you tried rebooting? 
Yes

- Is this a regression? 
No

hughsie, 2021-07-30:
Did you use `--ipfs` when running `fwupdmgr update`?

blauch, 2021-07-30:
no, I have not ever used this flag before. I use it as plain and simple as in my description :slightly_smiling_face:

hughsie, 2021-07-30:
Can you grab me the output of `fwupdmgr get-releases --verbose a45df35ac0e948ee180fe216a5f703f32dda163f` please?

blauch, 2021-07-30:
sure... here you go: 
```
fwupdmgr get-releases --verbose a45df35ac0e948ee180fe216a5f703f32dda163f

(fwupdmgr:16324): GLib-DEBUG: 10:55:32.892: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:16324): GLib-GIO-DEBUG: 10:55:32.907: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:16324): dconf-DEBUG: 10:55:32.907: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:16324): dconf-DEBUG: 10:55:32.908: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:16324): dconf-DEBUG: 10:55:32.908: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:16324): dconf-DEBUG: 10:55:32.908: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:16324): dconf-DEBUG: 10:55:32.908: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:16324): dconf-DEBUG: 10:55:32.908: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:16324): dconf-DEBUG: 10:55:32.908: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:16324): dconf-DEBUG: 10:55:32.908: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:16324): dconf-DEBUG: 10:55:32.908: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:16324): dconf-DEBUG: 10:55:32.908: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:16324): dconf-DEBUG: 10:55:32.911: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:16324): dconf-DEBUG: 10:55:32.911: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:16324): dconf-DEBUG: 10:55:32.911: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:16324): dconf-DEBUG: 10:55:32.911: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:16324): dconf-DEBUG: 10:55:32.911: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:16324): GLib-GIO-DEBUG: 10:55:32.914: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:16324): Fwupd-DEBUG: 10:55:32.917: Emitting ::status-changed() [idle]
  AppstreamId:          com.dell.uefie8292593.firmware
  RemoteId:             lvfs
  Summary:              Firmware for the Dell XPS 13 9300
  Description:          <p>This stable release fixes the following issues:</p><ul><li>Firmware updates to address the Intel Security Advisory.</li><li>Firmware updates to address security vulnerabilities.</li></ul>
  Version:              1.6.3
  Filename:             Dell XPS 13 9300 System BIOS_Ver. 1.6.3.cab
  Protocol:             org.uefi.capsule
  Checksum:             SHA1(d299803bfde308810a74864b755c4ff2fb5ed493)
  Checksum:             SHA256(c315fce8b8ccb724d2d68acfb1b4057a55345f9f78c4d3c8295578b52386bee8)
  License:              LicenseRef-proprietary
  Size:                 17,9 MB
  Created:              2021-05-24
  Uri:                  https://fwupd.org/downloads/c315fce8b8ccb724d2d68acfb1b4057a55345f9f78c4d3c8295578b52386bee8-Dell XPS 13 9300 System BIOS_Ver. 1.6.3.cab
  Homepage:             http://support.dell.com/
  Urgency:              critical
  Vendor:               Dell Inc.
  Flags:                is-upgrade

  AppstreamId:          com.dell.uefie8292593.firmware
  RemoteId:             lvfs
  Summary:              Firmware for the Dell XPS 13 9300
  Description:          <p>This stable release fixes the following issues:</p><ul><li>Fixed the issue where a user acceptance is prompted to proceed with the BIOSConnect recovery when network stack is disabled in the BIOS.</li><li>Fixed the issue where the BIOS Auto-Recovery and BIOS recovery through USB do not work.</li><li>Fixed the issue where the BIOS Auto-Recovery function does not work after the BIOS Auto-Recovery option in the BIOS setup is enabled.</li><li>Fixed the issue where the external monitor stops responding when it is connected to Dell Mobile Adapter Speakerphone MH3021P.</li><li>Fixed the issue where Dell dock WD19TB stops working when external devices are connected.</li><li>Fixed the issue where the Administrator and System Password cannot be set through Windows Management Instrumentation (WMI) command. This issue occurs when you enter more than 16 characters.</li><li>Fixed the issue where the BIOS updates fails when the system is connected to Dell dock WD19TB.</li><li>Fixed the issue where the power button does not work when you are in the operating system.</li></ul><p>Some new functionality has also been added:</p><ul><li>Added an option to add ownership date when it is not entered in the BIOS setup.</li></ul>
  Version:              1.5.0
  Filename:             Dell XPS 13 9300 System BIOS_Ver. 1.5.0.cab
  Protocol:             org.uefi.capsule
  Checksum:             SHA1(289280a9a992fe17fe6b0b86d1abf9f98ef198f5)
  Checksum:             SHA256(b022fe19616a5779b187778939c689acae68cb5f0f8b0b1d149277e6b84c407d)
  License:              LicenseRef-proprietary
  Size:                 14,7 MB
  Created:              2021-02-03
  Uri:                  https://fwupd.org/downloads/cbb98b7c5a0bd40c947d1dc0e33219e754d88651ef71f81a5e586a8c8b4411bf-Dell XPS 13 9300 System BIOS_Ver. 1.5.0.cab
  Homepage:             http://support.dell.com/
  Urgency:              critical
  Vendor:               Dell Inc.
  Flags:                none

  AppstreamId:          com.dell.uefie8292593.firmware
[... cut ...]
  Checksum:             SHA256(b44925c985923e23ea6556accab16db02eaea5d2f326facfa141c140cf5ad7f3)
  License:              LicenseRef-proprietary
  Size:                 13,9 MB
  Created:              2020-08-13
  Uri:                  https://fwupd.org/downloads/153c0ed1a19ecef730ec84494fc61c2a843af3ce9852989f46b243870f4e8f2d-XPS_13_9300_1.2.0.cab
  Homepage:             http://support.dell.com/
  Urgency:              critical
  Vendor:               Dell Inc.
  Flags:                is-downgrade|blocked-version
```

hughsie, 2021-07-30:
Okay, I can't explain that. Is there any chance you could test with git master, i.e. what will be 1.6.2 in a few days?

blauch, 2021-07-30:
yes, I can install `fwupd-git` and `fwupd-efi-git` once it made it to the repos

kaputtnik, 2021-07-30:
I got the same issue with my XPS 15 9560 and Arch Linux.

```
fwupdmgr get-releases --verbose a45df35ac0e948ee180fe216a5f703f32dda163f

(fwupdmgr:25438): GLib-DEBUG: 15:52:45.401: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:25438): GLib-GIO-DEBUG: 15:52:45.415: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:25438): dconf-DEBUG: 15:52:45.415: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:25438): dconf-DEBUG: 15:52:45.415: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:25438): dconf-DEBUG: 15:52:45.415: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:25438): dconf-DEBUG: 15:52:45.415: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:25438): dconf-DEBUG: 15:52:45.415: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:25438): dconf-DEBUG: 15:52:45.415: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:25438): dconf-DEBUG: 15:52:45.415: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:25438): dconf-DEBUG: 15:52:45.415: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:25438): dconf-DEBUG: 15:52:45.415: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:25438): dconf-DEBUG: 15:52:45.416: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:25438): dconf-DEBUG: 15:52:45.418: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:25438): dconf-DEBUG: 15:52:45.418: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:25438): dconf-DEBUG: 15:52:45.418: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:25438): dconf-DEBUG: 15:52:45.418: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:25438): dconf-DEBUG: 15:52:45.418: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:25438): GLib-GIO-DEBUG: 15:52:45.422: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:25438): Fwupd-DEBUG: 15:52:45.424: Emitting ::status-changed() [idle]
  AppstreamId:          com.dell.uefi34578c72.firmware
  RemoteId:             lvfs
  Summary:              Firmware for the Dell XPS 15 9560/Precision 5520
  Description:          <p>This stable release fixes the following issues:</p><ul><li>Firmware updates to address the Intel Security Advisory.</li><li>Firmware updates to address security vulnerabilities.</li></ul>
  Version:              1.23.1
  Filename:             Dell XPS 15 9560&Precision 5520 System BIOS_Ver.1.23.1.cab
  Protocol:             org.uefi.capsule
  Checksum:             SHA1(6e930a0f24efb6b1bdf88ca5b479ca7d43135268)
  Checksum:             SHA256(bc77e6e406022e4c0cea7e06fbbb7b545e261b65d227574c5a82afd5daeb3f67)
  License:              LicenseRef-proprietary
  Size:                 13,9 MB
  Created:              2021-05-25
  Uri:                  https://fwupd.org/downloads/bc77e6e406022e4c0cea7e06fbbb7b545e261b65d227574c5a82afd5daeb3f67-Dell XPS 15 9560&Precision 5520 System BIOS_Ver.1.23.1.cab
  Homepage:             http://support.dell.com/
  Urgency:              critical
  Vendor:               Dell Inc.
  Flags:                is-upgrade

  AppstreamId:          com.dell.uefi34578c72.firmware
  RemoteId:             lvfs
  Summary:              Firmware for the Dell XPS 15 9560/Precision 5520
  Description:          <p>This stable release fixes the following issues:</p><ul><li>Updated the video controller name in the BIOS to match with the video controller name in the operating system.</li></ul>
  Version:              1.22.1
  Filename:             firmware.cab
  Protocol:             org.uefi.capsule
  Checksum:             SHA1(ed35616d649531df5545b48f4b45adc380d91170)
  Checksum:             SHA256(b44e5f08e4a0da19b6ede573bef77abcbea52a66170bd5b573f56ea96b48c2fa)
  License:              LicenseRef-proprietary
  Size:                 10,6 MB
  Created:              2021-02-23
  Uri:                  https://fwupd.org/downloads/a3fa81b40924dcba6ee67aef629596665688179a9d9d8eba798aa4450c665aac-firmware.cab
  Homepage:             http://support.dell.com/
  Urgency:              critical
  Vendor:               Dell Inc.
  Flags:                none

  AppstreamId:          com.dell.uefi34578c72.firmware
  RemoteId:             lvfs
  Summary:              Firmware for the Dell XPS 15 9560/Precision 5520
  Description:          <p>This stable release fixes the following issues:</p><ul><li>Firmware updates to address the Intel Security Advisory</li><li>Firmware updates to address the Intel Technical Advisory</li></ul>
  Version:              1.21.0
  Filename:             firmware.cab
  Protocol:             org.uefi.capsule
  Checksum:             SHA1(6f5550e349fc5904c43d3214a8863c3e8844cb7a)
  Checksum:             SHA256(e8d08be55cfb22d870c2d2f619d70260947c245c1612b8aff0385764e0d4fc98)
  License:              LicenseRef-proprietary
[... 20 lines cut ...]
  Uri:                  https://fwupd.org/downloads/972abc70ab7d74aec756bdc2a49cecaf4da5f5a7a9197205afb17227fcf72a43-Dell XPS 15 9560&Precision 5520 System BIOS_Ver.1.20.0.cab
  Homepage:             http://support.dell.com/
  Urgency:              critical
  Vendor:               Dell Inc.
  Flags:                is-downgrade|blocked-version

[... cut ...]
  Checksum:             SHA256(e7297994e295d6e92f4369853ba49efc09c85f5ae4b67346b842614d5f33e34f)
  License:              LicenseRef-proprietary
  Size:                 10,4 MB
  Created:              2020-05-22
  Uri:                  https://fwupd.org/downloads/6d7b65ec0f4273e7adee2474c8a1eb290cdaa259880ea3b2ae9c9d79f4641399-xps_15_9560_1.19.2.cab
  Homepage:             http://support.dell.com/
  Urgency:              critical
  Vendor:               Dell Inc.
  Flags:                is-downgrade|blocked-version

```

hughsie, 2021-07-30:
@kaputtnik does it work better with git master?

piyueh, 2021-07-30:
I got the same issue with Dell XPS 13 7390. I'm using the master branch (commit f11abafed7eb798ba81a120398d3209fdebb510e). My system firmware was originally 1.7.0. I was able to upgrade from 1.7.0 to 1.8.0 using downloaded firmware file ([here](https://fwupd.org/downloads/6ebfa3db299402d637f4919702d854bf2bc747218d018569be39236c4ba98850-Dell%20XPS%2013%207390%20System%20BIOS_Ver.1.8.0.cab)). However, I wasn't able to upgrade to 1.9.0 even with the downloaded file ([here](https://fwupd.org/downloads/e52918a952cd1f754ce53a7a9678db1e8d2ff5d0df6b0923ae98c14625363b8b-Dell%20XPS%2013%207390%20System%20BIOS_Ver.1.9.0.cab)).

I tried to use `get-details` to check the file of v1.9.0, but got errors:

```
$ fwupdmgr get-details 4ecfcd7dacdabca09d276e6fad0686ceaefbdeb155da3b0da1181545ba88b675-Dell\ XPS\ 13\ 7390\ System\ BIOS_Ver.1.9.0.cab
----

Decompressing…           [   -                                   ]failed to compile 633f56b634847160b2ba9655fac268f9f0604a8a:filename=firmware.metainfo.xml:prefix=components: Error on line 58 char 2: Document must begin with an element (e.g. <book>)
```

blauch, 2021-08-02:
@hughsie just a wild guess... there are whitespace in between - could it be that the URI is not escaped properly? :man_shrugging: 

`https://fwupd.org/downloads/cae240dac84f228948c0ec3ee9a0fc1607dbb6bf4f7a6842b9d10f73915cd08c-Dell XPS 13 9300 System BIOS_Ver. 1.6.3.cab`

hughsie, 2021-08-02:
I was also surprised to see the spaces too.

cloventt, 2021-08-02:
I see this same issue for a Dell XPS 15.

OS is Arch Linux, installed via `pacman`.
```~ fwupdmgr --version
client version:	1.6.1
compile-time dependency versions
	gusb:	0.3.7

daemon version:	1.6.1```

```fwupdmgr get-releases --verbose a45df35ac0e948ee180fe216a5f703f32dda163f
(fwupdmgr:22026): GLib-DEBUG: 11:08:01.564: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:22026): GLib-GIO-DEBUG: 11:08:01.575: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:22026): dconf-DEBUG: 11:08:01.575: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:22026): dconf-DEBUG: 11:08:01.575: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:22026): dconf-DEBUG: 11:08:01.575: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:22026): dconf-DEBUG: 11:08:01.575: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:22026): dconf-DEBUG: 11:08:01.575: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:22026): dconf-DEBUG: 11:08:01.575: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:22026): dconf-DEBUG: 11:08:01.575: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:22026): dconf-DEBUG: 11:08:01.575: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:22026): dconf-DEBUG: 11:08:01.575: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:22026): dconf-DEBUG: 11:08:01.575: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:22026): dconf-DEBUG: 11:08:01.578: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:22026): dconf-DEBUG: 11:08:01.578: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:22026): dconf-DEBUG: 11:08:01.578: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:22026): dconf-DEBUG: 11:08:01.578: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:22026): dconf-DEBUG: 11:08:01.578: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:22026): GLib-GIO-DEBUG: 11:08:01.594: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:22026): Fwupd-DEBUG: 11:08:01.597: Emitting ::status-changed() [idle]
  AppstreamId:          com.dell.uefi57bd7271.firmware
  RemoteId:             lvfs
  Summary:              Firmware for the Dell XPS 15 9500, Precision 5550
  Description:          <p>This stable release fixes the following issues:</p><ul><li>Firmware updates to address security vulnerabilities.</li><li>Firmware updates to address the Intel Security Advisory</li></ul>
  Version:              1.8.1
  Filename:             Dell XPS 15 9500&Precision 5550 System BIOS_Ver.1.8.1.cab
  Protocol:             org.uefi.capsule
  Checksum:             SHA1(06435f3cd3e5298823acc9d722bcf5b977693e4e)
  Checksum:             SHA256(a47cfba226aa4faf1792b5f3bda9845ee3d24d29d7f8222e4f0e9088a829d9a6)
  License:              LicenseRef-proprietary
  Size:                 27.9 MB
  Created:              2021-05-26
  Uri:                  https://fwupd.org/downloads/a47cfba226aa4faf1792b5f3bda9845ee3d24d29d7f8222e4f0e9088a829d9a6-Dell XPS 15 9500&Precision 5550 System BIOS_Ver.1.8.1.cab
  Homepage:             http://support.dell.com/
  Urgency:              critical
  Vendor:               Dell Inc.
  Flags:                is-upgrade

  AppstreamId:          com.dell.uefi57bd7271.firmware
  RemoteId:             lvfs
  Summary:              Firmware for the Dell XPS 15 9500, Precision 5550
  Description:          <p>This stable release fixes the following issues:</p><ul><li>Fixed the issue where a user acceptance is prompted to proceed with the BIOSConnect recovery when network stack is disabled in the BIOS.</li><li>Fixed the issue where the Administrator and System Password cannot be set through Windows Management Instrumentation (WMI) command. This issue occurs when you enter more than 16 characters.</li></ul><p>Some new functionality has also been added:</p><ul><li>Supports Dell DA310 USB-C mobile adapter.</li><li>Updated the video controller name in the BIOS to match with the video controller name in the operating system.</li><li>Added an option to add ownership date when it is not entered in the BIOS setup.</li></ul>
  Version:              1.7.1
  Filename:             XPS9500_Precision5550_1.7.1.cab
  Protocol:             org.uefi.capsule
  Checksum:             SHA1(39ae82189e99619a54e4bec2780493235d729133)
  Checksum:             SHA256(dc6b15c383fe0bbff21f933e5fa54f4b2459b4a7454395e382ef002fe3f5f32a)
  License:              LicenseRef-proprietary
  Size:                 24.4 MB
  Created:              2021-03-29
  Uri:                  https://fwupd.org/downloads/82eae45d6e4f5efc284294e4bb850a9de890de9f220002e0c9008d7204bfa5c8-XPS9500_Precision5550_1.7.1.cab
  Homepage:             http://support.dell.com/
  Urgency:              critical
  Vendor:               Dell Inc.
  Flags:                none

  AppstreamId:          com.dell.uefi57bd7271.firmware
  RemoteId:             lvfs
  Summary:              Firmware for the Dell XPS 15 9500, Precision 5550
[... cut ...]
  Checksum:             SHA256(32be5fea4022d1ead0b1707da529f2083bc73b438d39d14cecadcd2faeb0f48f)
  License:              LicenseRef-proprietary
  Size:                 23.3 MB
  Created:              2020-07-20
  Uri:                  https://fwupd.org/downloads/31e9f6485b969365b55b3a82dcca67474fd2a142024277f269e0a7d779ee70d4-FioranoCml_14.0_PROD_QS_firmware.cab
  Homepage:             http://support.dell.com/
  Urgency:              critical
  Vendor:               Dell Inc.
  Flags:                is-downgrade|blocked-version
```

Spaces in the URL are probably the issue.

hughsie, 2021-08-03:
@cloventt can you try 1.6.2 which I released yesterday? If that still fails then I need to prioritize this accordingly.

JeffHenson, 2021-08-04:
I'm still getting the error on 1.6.2:

```
❯ fwupdmgr --version
client version:	1.6.2
compile-time dependency versions
	gusb:	0.3.7

daemon version:	1.6.2
```

```
❯ fwupdmgr get-releases --verbose a45df35ac0e948ee180fe216a5f703f32dda163f
(fwupdmgr:2751): GLib-DEBUG: 18:09:54.310: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:2751): GLib-GIO-DEBUG: 18:09:54.320: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:2751): dconf-DEBUG: 18:09:54.320: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:2751): dconf-DEBUG: 18:09:54.321: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:2751): dconf-DEBUG: 18:09:54.321: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:2751): dconf-DEBUG: 18:09:54.321: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:2751): dconf-DEBUG: 18:09:54.321: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:2751): dconf-DEBUG: 18:09:54.321: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:2751): dconf-DEBUG: 18:09:54.321: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:2751): dconf-DEBUG: 18:09:54.321: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:2751): dconf-DEBUG: 18:09:54.321: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:2751): dconf-DEBUG: 18:09:54.321: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:2751): dconf-DEBUG: 18:09:54.324: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:2751): dconf-DEBUG: 18:09:54.324: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:2751): dconf-DEBUG: 18:09:54.324: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:2751): dconf-DEBUG: 18:09:54.324: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:2751): dconf-DEBUG: 18:09:54.324: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:2751): GLib-GIO-DEBUG: 18:09:54.324: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:2751): Fwupd-DEBUG: 18:09:54.326: Emitting ::status-changed() [idle]
  AppstreamId:          com.dell.uefie4d7bdc4.firmware
  RemoteId:             lvfs
  Summary:              Firmware for the Dell XPS 15 9570/Precision 5530
  Description:          <p>This stable release fixes the following issues:</p><ul><li>Firmware updates to address the Intel Security Advisory.</li><li>Fixed the issue where the system displays a black screen when restarting the system after installing the Intel Rapid Storage Technology driver.</li><li>Firmware updates to address security vulnerabilities.</li></ul>
  Version:              1.20.0
  Filename:             Dell XPS 15 9570&Precision 5530 System BIOS_ver.1.20.0.cab
  Protocol:             org.uefi.capsule
  Checksum:             SHA1(7e5b279d70e2d5c9fba652341d8d6bb9fb7c4880)
  Checksum:             SHA256(be0c2a882f05e137b7e273ac60f294e56cadd71dccabb65377be94afc973a24c)
  License:              LicenseRef-proprietary
  Size:                 21.5 MB
  Created:              2021-06-08
  Uri:                  https://fwupd.org/downloads/be0c2a882f05e137b7e273ac60f294e56cadd71dccabb65377be94afc973a24c-Dell XPS 15 9570&Precision 5530 System BIOS_ver.1.20.0.cab
  Homepage:             http://support.dell.com/
  Urgency:              critical
  Vendor:               Dell Inc.
  Flags:                is-upgrade

  AppstreamId:          com.dell.uefie4d7bdc4.firmware
  RemoteId:             lvfs
  Summary:              Firmware for the Dell XPS 15 9570/Precision 5530
  Description:          <p>This stable release fixes the following issues:</p><ul><li>Fixed the issue where the Administrator and System Password cannot be set through Windows Management Instrumentation (WMI) command. This issue occurs when you enter more than 16 characters.</li><li>Fixed the issue where a user acceptance is prompted to proceed with the BIOSConnect recovery when network stack is disabled in the BIOS.</li><li>Fixed the issue where Enhanced Pre-Boot System Assessment (ePSA) fails to detect disconnected LCD cable issue during the test.</li><li>Firmware updates to address the Intel Technical Advisory</li></ul><p>Some new functionality has also been added:</p><ul><li>Updated the video controller name in the BIOS to match with the video controller name in the operating system.</li></ul>
  Version:              1.19.0
  Filename:             Dell XPS 15 9570&Precision 5530 System BIOS_ver.1.19.0.cab
  Protocol:             org.uefi.capsule
  Checksum:             SHA1(de1c17daaf7db3965f6df718e1bf179febf078b3)
  Checksum:             SHA256(ed20b3aedf652ca98b26f844df6ef9e3312ca2b75ac2679bc08dc4be148f2fdb)
  License:              LicenseRef-proprietary
  Size:                 18.1 MB
  Created:              2021-03-17
  Uri:                  https://fwupd.org/downloads/41038b03bd7477d2a99e176699e5a99729f80f41135dbd30cd6dd463e2d54865-Dell XPS 15 9570&Precision 5530 System BIOS_ver.1.19.0.cab
  Homepage:             http://support.dell.com/
  Urgency:              critical
  Vendor:               Dell Inc.
  Flags:                none

  AppstreamId:          com.dell.uefie4d7bdc4.firmware
  RemoteId:             lvfs
[... 49 lines cut ...]
  Vendor:               Dell Inc.
[... cut ...]
║ restarting the system after installing the Intel Rapid Storage Technology    ║
║ driver.                                                                      ║
║ • Firmware updates to address security vulnerabilities.                      ║
║ Precision 5530 must remain plugged into a power source for the duration of   ║
║ the update to avoid damage.                                                  ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
no valid release URIs
```

hughsie, 2021-08-04:
It's the ampersand. I'm working on a fix LVFS-side.

hughsie, 2021-08-04:
https://gitlab.com/fwupd/lvfs-website/-/merge_requests/1055

blauch, 2021-08-05:
> @hughsie just a wild guess... there are whitespace in between - could it be that the URI is not escaped properly? 
> 
> `https://fwupd.org/downloads/cae240dac84f228948c0ec3ee9a0fc1607dbb6bf4f7a6842b9d10f73915cd08c-Dell XPS 13 9300 System BIOS_Ver. 1.6.3.cab`

@hughsie I wonder if the ampersand is the only culprit that is causing this problem since there is none in the URI for my laptop?!

hughsie, 2021-08-05:
Ohh interesting. The lvfs-website fix will replace the spaces too, so fingers crossed.

hughsie, 2021-08-05:
I've deployed that, run a fsck, and the signing instance is running at 100% :) I'll let you know when it's all done.

hughsie, 2021-08-05:
Okay, it just completed. Can you try "fwupdmgr refresh --force" and try again please.

shroudedcode, 2021-08-05:
Still running into the same kind of issue after running `fwupdmgr refresh --force` with the same device as @cloventt:

```
$ fwupdmgr --version
client version:	1.6.2
compile-time dependency versions
	gusb:	0.3.7

daemon version:	1.6.2
```

```
$ fwupdmgr update
Devices with no available firmware updates: 
 • SSDPEMKF512G8 NVMe INTEL 512GB
 • UEFI dbx
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 1.9.1 to 1.12.0?                                ║
╠══════════════════════════════════════════════════════════════════════════════╣
[...]

Perform operation? [Y|n]: 
no valid release URIs
```

```
$ fwupdmgr get-releases --verbose a45df35ac0e948ee180fe216a5f703f32dda163f
  AppstreamId:          com.dell.uefi417d4c2a.firmware
  RemoteId:             lvfs
  Summary:              Firmware for the Dell XPS 15 7590/Precision 5540
  Description:          <p>This stable release fixes the following issues:</p><ul><li>Firmware updates to address the Intel Security Advisory.</li><li>Firmware updates to address security vulnerabilities.</li></ul>
  Version:              1.12.0
  Filename:             Dell XPS7590&Precision5540 BIOS_Ver. 1.12.0.cab
  Protocol:             org.uefi.capsule
  Checksum:             SHA1(4a66bdc2c9488ec551fda69ee05e82643d6800a7)
  Checksum:             SHA256(e5546e3f97efc23a65608ed1b70fb45ea9539fdd95714656c169aef1d363803b)
  License:              LicenseRef-proprietary
  Size:                 22.9 MB
  Created:              2021-06-10
  Uri:                  https://fwupd.org/downloads/e5546e3f97efc23a65608ed1b70fb45ea9539fdd95714656c169aef1d363803b-Dell XPS7590&Precision5540 BIOS_Ver. 1.12.0.cab
  Homepage:             http://support.dell.com/
  Urgency:              critical
  Vendor:               Dell Inc.
  Flags:                is-upgrade

[...]
```

Again, spaces and ampersands in the URL seem to be at fault.

hughsie, 2021-08-05:
Ohh that's the old name still. Maybe try again in 2 hours when the CDN is invalidated?

mastertinner, 2021-08-05:
For me, that fixed the issue. Thanks, @hughsie! I was able to upgrade my Dell XPS 13 7390 system  firmware from 1.8.0 to 1.9.0.

sileht, 2021-08-05:
It's allow fixed for my XPS 13 9300 with firmware from 1.5.0 to 1.6.3.

williamkray, 2021-08-05:
third confirmation, i was able to update my XPS 13 9310. thank you!

chrisjbillington, 2021-08-07:
My Dell Precision 5520 was also able to get a firmware update after this fix, having previously suffered the same issue described upthread. Thanks for the fix!