fwupd/fwupd #3576: Problem detecting the Fingerprint Reader by Synaptics on Lenono P52

erlisdhima, 2021-07-30:
fwupd isn't able to detect the fingerprint device, but lsusb does.

```
$ lsusb | grep "Synaptics"
Bus 001 Device 004: ID 06cb:009a Synaptics, Inc. 
```

**My fwupd version:**
```
$ fwupdmgr --version                                    
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11
```

I also tried removing fwupd and installing the snap and flatpak version, but same issue.
```
$ sudo -i flatpak run org.freedesktop.fwupd --version                         
client version:	1.6.1
compile-time dependency versions
	gusb:	0.3.7

```

**Detected devices:**
```
$ fwupdmgr get-devices                                  
20M9CTO1WW
│
├─Thunderbolt Controller:
│     Device ID:           74437a46ad5476328c46fe8316ebd905f646d9f3
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     62.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               c4d055aa-3295-5ff3-91a9-e576ebe209c0
│                          70470281-1b06-5da6-9ee7-ac4d96ea828d
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─INTEL SSDPEKKF512G8L:
│     Device ID:           c6a0cfba7c7d81e253fce571e1d1e9f6003ae1c7
│     Summary:             NVM Express Solid State Drive
│     Current version:     L14P
│     Vendor:              Intel Corporation (NVME:0x8086)
│     GUIDs:               79517f86-8df8-5d6e-a18b-33f0b36a78e9
│                          68db11e5-b0cf-5bc9-a94e-17e28496e505
│                          87fd9fec-f447-5ec6-a1cd-d55fac0d3d30
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Samsung SSD 970 EVO 250GB:
│     Device ID:           03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Summary:             NVM Express Solid State Drive
│     Current version:     2B2QEXE7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d
│                          47335265-a509-51f7-841e-1c94911af66b
│                          c198ada2-cf92-5e2e-a7d0-fb7fa0fde599
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           65b6a9dc7b7df18bdff003584b51bf21373e3aa6
│     Current version:     0.1.42
│     Minimum Version:     0.1.27
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                1e1fe415-74e8-49e1-9508-106b3d13d50d
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           619057f1d0da67b3fa4e9f74e09fc9218ce32643
│     Current version:     192.81.1753
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                971682af-21a0-46e3-91c0-825702c2ba70
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           0f3c64df11304fc5711ef77a93e3e042e2738907
│     Current version:     0.1.16
│     Minimum Version:     0.1.16
│     Vendor:              DMI:LENOVO
│     GUID:                88b8ba7e-296b-4f9e-929f-dcc19318dbcf
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─WDC WD10JPLX-00MBPT0:
      Device ID:           64ec4cd1e1c9565e79b00f4e7221b1c689b33e96
      Summary:             ATA Drive
      Current version:     01.01H01
      GUIDs:               2c27337b-72c6-5e84-a962-dac289e327a9
                           9a5ea9a9-9b18-569f-a251-cd4059db7115
                           807dc9ac-688b-5f0e-b8d5-1bb711146ef6
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
```

**My system:**
```
Operating System: Ubuntu 20.04.2 LTS
Kernel: Linux 5.4.0-80-generic
Architecture: x86-64
```

The following errors are thrown by the`fprintd` service:
```
systemd[1]: Starting Fingerprint Authentication Daemon...
kernel: [   93.326475] usb 1-9: reset full-speed USB device number 4 using xhci_hcd
fprintd[6581]: Firmware version 6.7 with build number 164 is unsupported
fprintd[6581]: Ignoring device due to initialization error: Unsupported firmware version (6.7 with build number 164)
dbus-daemon[1118]: [system] Successfully activated service 'net.reactivated.Fprint'
systemd[1]: Started Fingerprint Authentication Daemon.
systemd[1]: fprintd.service: Succeeded.
```

`hwinfo` will output this:
```
  usb device: name = 1-9:1.0
    path = /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0
    modalias = "usb:v06CBp009Ad0164dcFFdsc10dpFFicFFisc00ip00in00"
    bInterfaceNumber = 0
    bInterfaceClass = 255
    bInterfaceSubClass = 0
    bInterfaceProtocol = 0
    if: 1-9:1.0 @ /devices/pci0000:00/0000:00:14.0/usb1/1-9
    bDeviceClass = 255
    bDeviceSubClass = 16
    bDeviceProtocol = 255
    idVendor = 0x06cb
    idProduct = 0x009a
    serial = "108ba4076661"
    bcdDevice = 0164
    speed = "12"

58: USB 00.0: 0000 Unclassified device
  [Created at usb.122]
  Unique ID: POWV.J_8dhmy_9X7
  Parent ID: k4bc.2DFUsyrieMD
  SysFS ID: /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0
  SysFS BusID: 1-9:1.0
  Hardware Class: unknown
  Model: "Synaptics Unclassified device"
  Hotplug: USB
  Vendor: usb 0x06cb "Synaptics, Inc."
  Device: usb 0x009a 
  Revision: "1.64"
  Serial ID: "108ba4076661"
  Speed: 12 Mbps
  Module Alias: "usb:v06CBp009Ad0164dcFFdsc10dpFFicFFisc00ip00in00"
  Config Status: cfg=new, avail=yes, need=no, active=unknown
  Attached to: #67 (Hub)
```

Help or feedback is much appreciated!

erlisdhima, 2021-07-30:
I also opened an issue here for reference https://gitlab.freedesktop.org/libfprint/fprintd/-/issues/113

hughsie, 2021-07-30:
1.3.11 is now EOL. Can you try with a supported 1.5.x or 1.6.x release please?

erlisdhima, 2021-07-30:
@hughsie , I tried, but same with both the flatpak and snap package (version 1.6.1).

superm1, 2021-07-30:
Can we see the the verbose daemon log from the newer versions?

JiangXL, 2021-08-07:
Same issues in Synaptics(06cb:00c9) on HP Spectre X360 2021. 
Here is 1.62 log output:
[synaptics.log](https://github.com/fwupd/fwupd/files/6949257/synaptics.log)

erlisdhima, 2021-08-13:
@superm1 , this is the output from 1.6.2

```
09:23:09:0911 FuDebug              Verbose debugging enabled (on console 1)
09:23:09:0912 FuCommon             creating path /run/lock
09:23:09:0912 FuMain               locked /run/lock/fwupdtool
09:23:09:0912 FuConfig             loading config values from /app/etc/fwupd/daemon.conf
09:23:09:0912 FuConfig             using autodetected max archive size 18.4 EB
09:23:09:0917 GVFS                 org.gtk.vfs.MountTracker.listMountableInfo call failed: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: org.freedesktop.DBus.Error.ServiceUnknown (g-dbus-error-quark, 2)

09:23:09:0917 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
09:23:09:0917 FuRemoteList         loading /app/share/fwupd/metainfo
09:23:09:0917 XbSilo               attempting to load /var/data/cache/fwupd/metainfo.xmlb
09:23:09:0917 XbSilo               file: 9faa27b2-d040-07dd-b4c8-d817993eac12, current:9faa27b2-d040-07dd-b4c8-d817993eac12, cached: (null)
09:23:09:0917 XbSilo               loading silo with file contents
09:23:09:0917 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/dell-esrt.conf
09:23:09:0917 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/vendor.conf
09:23:09:0917 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/lvfs.conf
09:23:09:0917 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/lvfs-testing.conf
09:23:09:0918 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/vendor-directory.conf
09:23:09:0918 FuRemoteList         ignoring unfound remote fwupd
09:23:09:0918 FuRemoteList         ordering lvfs-testing=lvfs+1
09:23:09:0918 Jcat                 trying to load certificate from /app/etc/pki/fwupd/LVFS-CA.pem
09:23:09:0918 Jcat                 reading /app/etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
09:23:09:0918 Jcat                 loaded 1 certificates
09:23:09:0918 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
09:23:09:0918 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
09:23:09:0918 Jcat                 trying to load certificate from /app/etc/pki/fwupd-metadata/LVFS-CA.pem
09:23:09:0918 Jcat                 reading /app/etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
09:23:09:0918 Jcat                 loaded 1 certificates
09:23:09:0918 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
09:23:09:0918 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
09:23:09:0918 Jcat                 reading /var/data/lib/fwupd/pki/secret.key with 2459 bytes
09:23:09:0918 Jcat                 reading /var/data/lib/fwupd/pki/client.pem with 1383 bytes
09:23:09:0923 FuEngine             client certificate exists and working
09:23:09:0923 FuHistory            trying to open database '/var/data/lib/fwupd/pending.db'
09:23:09:0923 FuHistory            got schema version of 6
09:23:09:0923 FuSmbios             ignoring Failed to open file “/sys/firmware/dmi/tables/smbios_entry_point”: Permission denied
09:23:09:0923 FuSmbios             trying to read /sys/class/dmi/id
09:23:09:0923 FuSmbios             unable to read SMBIOS data from /sys/class/dmi/id/product_serial: Failed to open file “/sys/class/dmi/id/product_serial”: Permission denied
09:23:09:0923 FuSmbios             unable to read SMBIOS data from /sys/class/dmi/id/board_serial: Failed to open file “/sys/class/dmi/id/board_serial”: Permission denied
09:23:09:0923 FuSmbios             unable to read SMBIOS data from /sys/class/dmi/id/chassis_serial: Failed to open file “/sys/class/dmi/id/chassis_serial”: Permission denied
09:23:09:0923 FuHwids              smbios property Manufacturer=LENOVO
09:23:09:0924 FuHwids              smbios property EnclosureKind=a
09:23:09:0924 FuHwids              smbios property Family=ThinkPad P52
09:23:09:0924 FuHwids              smbios property ProductName=20M9CTO1WW
09:23:09:0924 FuHwids              smbios property ProductSku=LENOVO_MT_20M9_BU_Think_FM_ThinkPad P52
09:23:09:0924 FuHwids              smbios property BiosVendor=LENOVO
09:23:09:0924 FuHwids              smbios property BiosVersion=N2CET59W (1.42 )
09:23:09:0924 FuHwids              ignoring BiosMajorRelease: offset bigger than size 9
09:23:09:0924 FuHwids              ignoring BiosMinorRelease: offset bigger than size 9
09:23:09:0924 FuHwids              ignoring FirmwareMajorRelease: offset bigger than size 9
09:23:09:0924 FuHwids              ignoring FirmwareMinorRelease: offset bigger than size 9
09:23:09:0924 FuHwids              smbios property BaseboardManufacturer=LENOVO
09:23:09:0924 FuHwids              smbios property BaseboardProduct=20M9CTO1WW
09:23:09:0924 FuHwids              HardwareID-0 is not available, not available as 'BiosMajorRelease' unknown
09:23:09:0924 FuHwids              HardwareID-1 is not available, not available as 'BiosMajorRelease' unknown
09:23:09:0924 FuHwids              HardwareID-2 is not available, not available as 'BiosMajorRelease' unknown
09:23:09:0924 XbSilo               attempting to load /var/data/cache/fwupd/metadata.xmlb
09:23:09:0924 XbSilo               file: 75c5b63b-640b-1c89-dc60-c77fffce1a2b, current:75c5b63b-640b-1c89-dc60-c77fffce1a2b, cached: (null)
09:23:09:0924 XbSilo               loading silo with file contents
[... cut ...]
09:23:10:0338 FuPlugin             destroy(modem_manager)
09:23:10:0338 FuPlugin             destroy(linux_lockdown)
09:23:10:0338 FuPlugin             destroy(tpm_eventlog)
09:23:10:0339 FuPlugin             destroy(dell)
09:23:10:0339 FuPlugin             destroy(powerd)
09:23:10:0339 FuPlugin             destroy(linux_swap)
09:23:10:0339 FuPlugin             destroy(linux_tainted)
09:23:10:0339 FuPlugin             destroy(redfish)
09:23:10:0339 FuPlugin             destroy(uefi_capsule)
```

superm1, 2021-08-13:
It's still not showing the probe error for synaptics for either of you guys.  Can you please try to set `FWUPD_PROBE_VERBOSE=1` in the environment before running fwupd daemon?

erlisdhima, 2021-08-13:
@superm1, using the verbose env variable, I see this result:
```
14:16:47:0940 FuEngine             usb added FuUsbDevice:
Unknown Device
  Guid:                 aaacee1a-8f59-5fae-b888-49f3ebb29c7d ← USB\VID_06CB&PID_009A&REV_0164 ⚠
  Guid:                 43a27de8-192d-5273-a3b5-7b17e2980f4a ← USB\VID_06CB&PID_009A ⚠
  Flags:                none
  VendorId:             USB:0x06CB
  Version:              1.64
  VersionFormat:        bcd
  PhysicalId:           usb:01:00:09
  ParentPhysicalIds:    usb:01:00,usb:01
  UsbDeviceClass:       vendor-specific
```

See full copy of the console output.
[fwupd.log](https://github.com/fwupd/fwupd/files/6983059/fwupd.log)

Thanks a lot for working on this!

superm1, 2021-08-13:
Yeah I think the issue is related to missing a quirk for that device.  Try the PR I linked and see if that's enough for it to work.

erlisdhima, 2021-08-13:
I built that specific branch, but I'm getting this error on the `fwupd.service` (`/usr/local/libexec/fwupd/fwupd`)
```
15:51:07:0567 FuEngine             failed to add device usb:01:00:09: device 48 is not supported by this plugin
15:51:08:0144 FuMain               another service has claimed the dbus name org.freedesktop.fwupd
```

When I run `fwupdmgr get-devices` it hangs until it times out:
```
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
```

Please can you help what might be going wrong?

Thanks!

superm1, 2021-08-13:
Try to run `./contrib/prepare-system /usr/local install`.  This will set it up for your system to allow running fwupd from `/usr/local`.  If you want to revert it, you can run `./contrib/prepare-system /usr/local remove`

erlisdhima, 2021-08-13:
Thanks! It worked.
The problem now is that it doesn't list the device at all when running: `FWUPD_PROBE_VERBOSE=1 fwupdmgr get-devices -v`
It was listing it before as an 'Unknown Device' (see above).

superm1, 2021-08-13:
Can use I share the verbose daemon logs again with the patch in place?

erlisdhima, 2021-08-13:
Yes, please see attached. 
[fwupd-1.log](https://github.com/fwupd/fwupd/files/6983870/fwupd-1.log)

superm1, 2021-08-13:
That's the client log, i'm looking for the daemon log (as in `/usr/local/libexec/fwupd/fwupd -v`)

erlisdhima, 2021-08-13:
@superm1 
[fwupd-daemon.log](https://github.com/fwupd/fwupd/files/6984994/fwupd-daemon.log)

superm1, 2021-08-13:
Here is the reason for the failure. I think we'll need information from synaptics if the existing code can support this product ID. 

```
20:16:02:0843 FuPluginSynapticsPrometheus product ID is 48, version=6.7, buildnum=164 prod=32
```

superm1, 2021-08-15:
Synaptics will need to confirm what changes needed to support product ID 48.

erlisdhima, 2021-08-15:
@superm1 I found [this reddit post](https://www.reddit.com/r/thinkpad/comments/ihotav/linux_fingerprint_sensor_support_for_synaptics/), which has a link to a [driver (build in python)](https://github.com/uunicorn/python-validity) supposedly supporting this specific fingerprint reader.
I'm not sure what they are doing to support it, but will let you know as soon as I try it if its working. 

It would be great if it was actually supported by fwupd.

superm1, 2021-08-16:
I think we need to understand if this is a "Prometheus" family of device or something else to start. If it's something else then we're talking brand new plugin and it needs synaptics help in the form of specifications or code, not just a reverse engineered project.

superm1, 2021-08-17:
Synaptics confirmed this is not a `prometheus` device, it's an older device and the current plugin won't work with it.  That also likely means that no firmware is being published for it .

closed 2021-08-17 by superm1

erlisdhima, 2021-08-17:
Thanks for confirming! Tried with the above mentioned package and it worked.