fwupd/fwupd #3577: CI failure on snap from recent changes

superm1, 2021-07-30:
```
FAILED: plugins/powerd/libfu_plugin_powerd.so 
cc  -o plugins/powerd/libfu_plugin_powerd.so 'plugins/powerd/eceb772@@fu_plugin_powerd@sha/fu-plugin-powerd.c.o' -I/root/project/stage/include -L/root/project/parts/fwupd/install/lib -L/root/project/parts/fwupd/install/usr/lib -L/root/project/parts/fwupd/install/lib/x86_64-linux-gnu -L/root/project/parts/fwupd/install/usr/lib/x86_64-linux-gnu -L/root/project/stage/lib -L/root/project/stage/usr/lib -L/root/project/stage/lib/x86_64-linux-gnu -L/root/project/stage/usr/lib/x86_64-linux-gnu -Wl,--as-needed -Wl,--allow-shlib-undefined -shared -fPIC -Wl,--start-group -Wl,-soname,libfu_plugin_powerd.so -Wl,-z,relro -Wl,-z,defs -Wl,-z,now -Wl,-z,ibt,-z,shstk libfwupd/libfwupd.so.2.0.0 libfwupdplugin/libfwupdplugin.so.2.0.0 subprojects/libxmlb/src/libxmlb.so.2.0.0 subprojects/gusb/gusb/libgusb.so.2.0.10 /usr/lib/x86_64-linux-gnu/libgio-2.0.so /usr/lib/x86_64-linux-gnu/libgobject-2.0.so /usr/lib/x86_64-linux-gnu/libglib-2.0.so -Wl,--export-dynamic /usr/lib/x86_64-linux-gnu/libgmodule-2.0.so -pthread /usr/lib/x86_64-linux-gnu/libgudev-1.0.so /usr/lib/x86_64-linux-gnu/libjson-glib-1.0.so /usr/lib/x86_64-linux-gnu/libusb-1.0.so /usr/lib/x86_64-linux-gnu/libarchive.so -Wl,--end-group '-Wl,-rpath,$ORIGIN/../../libfwupd:$ORIGIN/../../subprojects/libjcat/libjcat:$ORIGIN/../../libfwupdplugin:$ORIGIN/../../subprojects/libxmlb/src:$ORIGIN/../../subprojects/gusb/gusb:/root/project/stage/lib' -Wl,-rpath-link,/root/project/parts/fwupd/build/snapbuild/libfwupd -Wl,-rpath-link,/root/project/parts/fwupd/build/snapbuild/subprojects/libjcat/libjcat -Wl,-rpath-link,/root/project/stage/lib -Wl,-rpath-link,/root/project/parts/fwupd/build/snapbuild/libfwupdplugin -Wl,-rpath-link,/root/project/parts/fwupd/build/snapbuild/subprojects/libxmlb/src -Wl,-rpath-link,/root/project/parts/fwupd/build/snapbuild/subprojects/gusb/gusb
plugins/powerd/eceb772@@fu_plugin_powerd@sha/fu-plugin-powerd.c.o: In function `fu_plugin_startup':
/root/project/parts/fwupd/build/snapbuild/../plugins/powerd/fu-plugin-powerd.c:98: undefined reference to `G_SOURCE_FUNC'
```

It appears to me that CI is failing in snap now because of https://github.com/fwupd/fwupd/commit/1a58398afb242e177ae4be7aa5614028cec40fa8

CC @g2pop

superm1, 2021-07-31:
I fixed it by https://github.com/fwupd/fwupd/pull/3579
In the future, please try to avoid breakage for any CI before merging.

closed 2021-07-31 by superm1

djcampello, 2021-07-31:
Sorry about this. I just saw this! Also, I did not see the CI failure when the PR was run :S Thank you for the fix!