fwupd/fwupd #3583: 1.6.2: test suite is failing

kloczek, 2021-08-02:
After sucessfull build ..
```console
+ cd fwupd-1.6.2
+ /usr/bin/meson test -C x86_64-redhat-linux-gnu --num-processes 48 --print-errorlogs
ninja: Entering directory `/home/tkloczko/rpmbuild/BUILD/fwupd-1.6.2/x86_64-redhat-linux-gnu'
ninja: no work to do.
 1/32 fwupd-exported-api                     OK               0.04s
 2/32 fwupdplugin-exported-api               OK               0.02s
 3/32 fwupd-context-test                     OK               0.05s
 4/32 fwupd-thread-test                      OK               0.11s
 5/32 fwupd-self-test                        OK               0.12s
 6/32 fwupdplugin-self-test                  OK               2.27s
 7/32 fu-self-test                           OK               5.70s
 8/32 acpi-dmar-self-test                    OK               0.20s
 9/32 acpi-facp-self-test                    OK               0.20s
10/32 acpi-phat-self-test                    OK               0.19s
11/32 ata-self-test                          OK               0.18s
12/32 bcm57xx-self-test                      OK               0.17s
13/32 ccgx-self-test                         OK               0.17s
14/32 dell-self-test                         OK               0.16s
15/32 fu-dfu-self-test                       OK               0.16s
16/32 elantp-self-test                       OK               0.15s
17/32 logitech-hidpp-self-test               OK               0.13s
18/32 nitrokey-self-test                     OK               0.12s
19/32 nvme-self-test                         OK               0.11s
20/32 pxi-self-test                          OK               0.11s
21/32 synaptics-mst-self-test                OK               0.09s
22/32 synaptics-prometheus-self-test         OK               0.09s
23/32 synaptics-rmi-self-test                OK               0.08s
24/32 tpm-eventlog-self-test                 OK               0.07s
25/32 uefi-dbx-self-test                     OK               0.05s
26/32 vli-self-test                          OK               0.05s
27/32 wacom-usb-self-test                    OK               0.04s
28/32 redfish-self-test                      OK               0.12s
29/32 linux-swap-self-test                   FAIL             0.32s   killed by signal 6 SIGABRT
>>> MALLOC_PERTURB_=232 /home/tkloczko/rpmbuild/BUILD/fwupd-1.6.2/x86_64-redhat-linux-gnu/plugins/linux-swap/linux-swap-self-test
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――― ✀  ―――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――
stdout:
# random seed: R02Sc92eac4ad0e4253b245a71a74e17ce30
1..3
# Start of linux-swap tests
ok 1 /linux-swap/none
Bail out! ERROR:../plugins/linux-swap/fu-self-test.c:44:fu_linux_swap_plain_func: assertion failed (error == NULL): failed to call org.freedesktop.UDisks2.Manager.GetBlockDevices(): Could not activate remote peer. (g-dbus-error-quark, 3)
stderr:
**
ERROR:../plugins/linux-swap/fu-self-test.c:44:fu_linux_swap_plain_func: assertion failed (error == NULL): failed to call org.freedesktop.UDisks2.Manager.GetBlockDevices(): Could not activate remote peer. (g-dbus-error-quark, 3)
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――

30/32 thunderbolt-self-test                  OK               3.54s
31/32 lenovo-thinklmi-self-test              FAIL            25.31s   killed by signal 5 SIGTRAP
>>> MALLOC_PERTURB_=32 /home/tkloczko/rpmbuild/BUILD/fwupd-1.6.2/x86_64-redhat-linux-gnu/plugins/lenovo-thinklmi/lenovo-thinklmi-self-test
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――― ✀  ―――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――
stdout:
# random seed: R02S1ae91aeba71bf31012431df0f11b389e
# GLib-GIO-DEBUG: _g_io_module_get_default: Found default implementation gvfs (GDaemonVfs) for ?gio-vfs?
# XbSilo-DEBUG: attempting to load /tmp/.HHI760
# XbSilo-DEBUG: failed to load silo: blob too small
# FuPlugin-DEBUG: init(/home/tkloczko/rpmbuild/BUILD/fwupd-1.6.2/x86_64-redhat-linux-gnu/plugins/lenovo-thinklmi/../uefi-capsule/libfu_plugin_uefi_capsule.so)
# FuPlugin-DEBUG: startup(uefi_capsule)
# FuPlugin-DEBUG: init(/home/tkloczko/rpmbuild/BUILD/fwupd-1.6.2/x86_64-redhat-linux-gnu/plugins/lenovo-thinklmi/libfu_plugin_lenovo_thinklmi.so)
# FuPlugin-DEBUG: startup(lenovo_thinklmi)
1..2
# Start of fwupd tests
# GLib-DEBUG: setenv()/putenv() are not thread-safe and should not be used after threads are created
# FuPlugin-DEBUG: coldplug(uefi_capsule)
# FuPluginUefiCapsule-DEBUG: not using CapsuleOnDisk support: failed to read EFI variable: Error opening file /home/tkloczko/rpmbuild/BUILD/fwupd-1.6.2/plugins/lenovo-thinklmi/tests/efi/efivars/OsIndicationsSupported-8be4df61-93ca-11d2-aa0d-00e098032b8c: No such file or directory
# GLib-DEBUG: setenv()/putenv() are not thread-safe and should not be used after threads are created
Bail out! FATAL-CRITICAL: failed to allocate dbus proxy object: Error calling StartServiceByName for com.intel.tss2.Tabrmd: Timeout was reached
stderr:

[... 100 lines cut ...]
(/home/tkloczko/rpmbuild/BUILD/fwupd-1.6.2/x86_64-redhat-linux-gnu/plugins/uefi-capsule/uefi-self-test:1201933): FuQuirks-DEBUG: 15:43:03.672: invalid key names: CcgxFlashRowSize,CcgxFlashSize,CcgxImageKind,CxaudioChipIdBase,CxaudioPatch1ValidAddr,CxaudioPatch2ValidAddr,DellDockBlobBuildOffset,DellDockBlobMajorOffset,DellDockBlobMinorOffset,DellDockBlobVersionOffset,DellDockBoardMin,DellDockHubVersionLowest,DellDockInstallDurationI2C,DellDockUnlockTarget,DellDockVersionLowest,DfuAltName,DfuForceTimeout,DfuForceVersion,ElantpI2cTargetAddress,ElantpIapPassword,ElantpIcPageCount,FlashromProgrammer,JabraMagic,PciBcrAddr,Rts54I2cSpeed,Rts54RegisterAddrLen,Rts54TargetAddr,SuperioChipsets,SuperioId,SuperioPort,SynapticsMstDeviceKind,VliDeviceKind,VliSpiCmdChipErase,VliSpiCmdSectorErase,WacomI2cFlashBaseAddr,WacomI2cFlashBlockSize,WacomI2cFlashSize
# GLib-DEBUG: setenv()/putenv() are not thread-safe and should not be used after threads are created
(/home/tkloczko/rpmbuild/BUILD/fwupd-1.6.2/x86_64-redhat-linux-gnu/plugins/uefi-capsule/uefi-self-test:1201933): GLib-DEBUG: 15:43:03.673: setenv()/putenv() are not thread-safe and should not be used after threads are created
Bail out! FATAL-CRITICAL: failed to allocate dbus proxy object: Error calling StartServiceByName for com.intel.tss2.Tabrmd: Timeout was reached
stderr:

** (/home/tkloczko/rpmbuild/BUILD/fwupd-1.6.2/x86_64-redhat-linux-gnu/plugins/uefi-capsule/uefi-self-test:1201933): CRITICAL **: 15:43:28.701: failed to allocate dbus proxy object: Error calling StartServiceByName for com.intel.tss2.Tabrmd: Timeout was reached
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――

Summary of Failures:

29/32 linux-swap-self-test           FAIL             0.32s   killed by signal 6 SIGABRT
31/32 lenovo-thinklmi-self-test      FAIL            25.31s   killed by signal 5 SIGTRAP
32/32 uefi-self-test                 FAIL            25.31s   killed by signal 5 SIGTRAP

Ok:                 29
Expected Fail:      0
Fail:               3
Unexpected Pass:    0
Skipped:            0
Timeout:            0

Full log written to /home/tkloczko/rpmbuild/BUILD/fwupd-1.6.2/x86_64-redhat-linux-gnu/meson-logs/testlog.txt
```
I'm not sure but does it mean that I should try to run test suite in separated dbus session?

hughsie, 2021-08-02:
Well, we run the self tests in Fedora too, and that doesn't have dbus set up. How exactly are you running the tests?

kloczek, 2021-08-02:
> Well, we run the self tests in Fedora too, and that doesn't have dbus set up. How exactly are you running the tests?

Executed commands are prefixed by "+"
```console
+ cd fwupd-1.6.2
+ /usr/bin/meson test -C x86_64-redhat-linux-gnu --num-processes 48 --print-errorlogs
```

hughsie, 2021-08-02:
No, I meant, what kind of environment are you using? e.g. a bubblewrap, a jail, or something else?

kloczek, 2021-08-02:
Just regular bare metal system with installed all necessary build requirements.

hughsie, 2021-08-02:
com.intel.tss2.Tabrmd doesn't seem to want to activate. Is there some kind of security policy preventing this?

kloczek, 2021-08-02:
How can I check that?
Just in case .. I'm building that package in system with disabled SELinux.

hughsie, 2021-08-03:
> Bail out! FATAL-CRITICAL: failed to allocate dbus proxy object: Error calling StartServiceByName for com.intel.tss2.Tabrmd: Timeout was reached

This is from the tpm2 libraries, there's nothing fwupd can do here. Does `sudo tpm2_pcrread` also give you the same error?

superm1, 2021-08-03:
I think you can uninstall the tpm2-abrmd package and try it using kernel resource manager.

kloczek, 2021-08-04:
After uninstall tpm2-abrmd I've manage to pass fwupd test suite correctly.
I had some ABI mess between libblockdev and udisk and that was cause of the failing linux-swap-self-test unit.

Still my chrome crashes because TPM2 issues but that is another story.
Still as well don't know generally why not working TPM started causing all those issues and going to ask tpm2-abrmd maintainer because such connection looks a bit odd/unusual to me.

Closing.
Thank you for the suggestion with tpm2-abrmd.

closed 2021-08-04 by kloczek