fwupd/fwupd #3584: 'UEFI capsule updates not available or [not] enabled in firmware setup': ThinkPad X230

LinuxOnTheDesktop, 2021-08-02:
**Describe the question**
I see the message 'UEFI capsule updates not available or [not] enabled in firmware setup' (my interpolation). Yet, my settings are thus:

[image]

Indeed I see the message even if enable 'rollback prevention'. Also, though, just now I saw a message on boot about an 'unreadable database' or some such. I am unsure I have seen this message before (and just now I disabled that rolback prevention but, also, upgraded fwupd).

**fwupd version information**
Please provide the version of the daemon and client if applicable to your current installation of `fwupd`.

```shell
$ fwupdmgr --version
client version:	1.5.11
compile-time dependency versions
	gusb:	0.3.4

daemon version:	1.5.11
```

Secure boot is disabled.

ThinkPad X230
Linux Mint 20.2 Cinnamon

hughsie, 2021-08-02:
Are you booting in UEFI mode or CSM BIOS legacy compat?

LinuxOnTheDesktop, 2021-08-02:
UEFI mode, I think. My relevant BIOS settings (well, I think they are the relevant settings) are as follows.

UEFI/Legacy Boot:
- UEFI/Legacy Boot Priority: UEFI First
- CSM Support: Yes
Boot Order Lock: disabled

hughsie, 2021-08-02:
Does /sys/firmware/efi/esrt/ exist? What about /sys/firmware/efi/efivars/ ?

LinuxOnTheDesktop, 2021-08-02:
`/sys/firmware/efi/esrt/`: does not exist.

`/sys/firmware/efi/efivars/`: exists.

And the message I get on boot is 'could not parse dbx signatures' (as against what I said it was above; sorry).

hughsie, 2021-08-02:
> /sys/firmware/efi/esrt/: does not exist.

Can you try booting in UEFI-only mode? I think you're booting in Legacy.

LinuxOnTheDesktop, 2021-08-02:
Thanks. But (1) should my existing BIOS settings not suffice to boot in UEFI-only mode? And (2) just now I set the boot mode to UEFI-only and nothing changed - I still saw the 'signatures' message and fwup still reports capsule updates as being either unavailable or disabled.

hughsie, 2021-08-02:
Maybe it's just too old to support the ESRT table? You could try flashing a firmware using Windows or a LiveCD manually and then see if that "unlocks" the newer EFI stuff.

LinuxOnTheDesktop, 2021-08-02:
Flashing what firmware, please? I think the PC has the latest available Lenovo BIOS for its model. Note also that the X230 is a popular enthusiast laptop, so, if the problem does not owe to anything peculiar in my setup, then many people will have problems.

I suspect though that the problem has to do with secure boot somehow. For, as is fairly common, I had terrible trouble (some time ago) with secure boot and ended up deleting secure boot keys and messing around with awful 'mokutil' on-boot interface things. And/or perhaps fwup is the problem - because, until I updated to fwup 1.5.11, I might have had that 'signature' message on boot (I am unsure) but fwup did not report a problem.

LinuxOnTheDesktop, 2021-08-02:
Also or at any rate: [the webpage of yours that is at issue](https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported) should make clear the effect of the problem. That is: if these 'UEFI capsules' cannot be got to work, what does that entail? That - so long as that situation obtains - some updates offered by fwupd are not installable? That no updates offered by fwupd are installable? The page does say, 'This warning can be ignored if UEFI firmware updates are not desired'. But what does that mean? that a _method_ of updating is unavailable? That _some updates_ are unavailable?

hughsie, 2021-08-03:
> I think the PC has the latest available Lenovo BIOS for its model.

I think you're out of luck then I'm afraid.

> I suspect though that the problem has to do with secure boot somehow. 

If there's no ESRT table that's nothing secure boot related.

>  the webpage of yours that is at issue 

Dude, it's a wiki. How about adding something like "Some older hardware (usually produced before 2015) does not support the ACPI ESRT table. It's possible that flashing the latest vendor BIOS using Windows or a LiveCD manually will add support for this table (which fixes the error message), but in most cases UEFI updates are unavailable for these platforms."

LinuxOnTheDesktop, 2021-08-19:
@hughsie 

I take it that you mean: why don't I add something that text to the wiki page? Answer: because I am not confident that the text is germane. Here is why.

1) Until recently I saw no such error message on fwupd on that machine.

2) I imagine many people use the same model of machine (ThinkPad X230) with fwupd (and do so successfully).

3) fwupd and its supporting infrastructure (including the 'shim' stuff on Ubuntu) has various problems - one of those problems could be causing the difficulty I face.

LinuxOnTheDesktop, 2021-09-19:
I have added text to the wiki. That text is similar to the text @hughsie suggested.

However, I am unconvinced that the text - even as ammended - gives the correct diagnosis of the problem that I reported here. For, I started to receive the error message at issue only fairly recently.

closed 2021-10-02 by stale[bot]

zakutin, 2022-11-19:
I also have exact the same issue with my x230 laptop. I'm getting the message "couldn't parse dbx signatures: -74", no matter what I do with UEFI BIOS settings.

@LinuxOnTheDesktop, did you manage the installation issue of the dbx updates somehow?

@hughsie, from your experience, is there any way to install these dbx updates without ACPI ESRT table support (in my case "Upgrade UEFI dbx from 78 to 217")? It seems like the latest UEFI BIOS version ever possible on x230 laptops is G2ETB7WW (2.77) from 2019, which has no ACPI ESRT option.

LinuxOnTheDesktop, 2022-11-19:
@zakutin : I still see a message about the updates not being available, but now the message contains the URL of [this webpage](https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported). That webpage is hard to understand, at least if read quickly, but from it I infer, tentatively, the following. fwupd it trying to tell me that my BIOS is too old to support fwupd (or, possibly, too old to support some of fwupd's functionality).

hughsie, 2022-11-21:
> couldn't parse dbx signatures

Is that from fwupd?

> is there any way to install these dbx updates without ACPI ESRT table support

You don't need ESRT to install the dbx, it's a different mechanism entirely.

zakutin, 2022-11-21:
Hello @hughsie,

> Is that from fwupd?

Not sure what causes causes the message. When I execute `sudo fwupdmgr update` ("Upgrade UEFI dbx from 78 to 217") and reboot the system, the message "couldn't parse dbx signatures: -74" appears during the boot. Only after resetting the Secure Boot keys in BIOS to system default, the message disappears, and the `dbxtool -l` shows the old state without enrolled signatures, as it was before. I tried to perform it with both versions, Ubuntu apt (1.7.9) and snap stable fwupd (1.8.6).

Sorry for off topic. I'm a bit lost in mechanisms of updating the keys. In fact, the fwupdmgr dbx update went fine on ThinkPad T480, and on X230 not (probably because of outdated BIOS). Could you please describe how to perform the dbx upgrade in other ways?

hughsie, 2022-11-21:
> couldn't parse dbx signatures: -74" appears during the boot

That looks like the system BIOS doesn't know what to do with the new dbx contents. Maybe you could update the BIOS using an ISO file to see if that unlocks LVFS and dbx updates?

nixpulvis, 2026-04-26:
I ended up here while trying to check for updates and install them on my 2017 Thinkpad X1 Carbon. I swear I've used this tool in the past to update this machine. Has something changed that I would have been unaware of?