fwupd/fwupd #3591: T480: Embedded Controller - Document must begin with an element

shyukri, 2021-08-03:
**Describe the bug**
A clear and concise description of what the bug is.

**Steps to Reproduce**
Steps to reproduce the behavior.
```
# sudo fwupdmgr get-updates
....
20L5000BBM
│
└─Embedded Controller:
  │   Device ID:          c19580db344e418cf2b23297bcf46d75c06e432a
  │   Current version:    0.1.21
  │   Minimum Version:    0.1.21
  │   Vendor:             DMI:LENOVO
  │   GUIDs:              b87a926d-189c-49a1-b1ff-921099b1de89
  │                       646e8b00-c980-5327-b889-b518b0a1cc1c ← UEFI\RES_{B87A926D-189C-49A1-B1FF-921099B1DE89}
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T480 Embedded Controller Update:
        New version:      0.1.22
        Remote ID:        lvfs
        Summary:          Lenovo ThinkPad T480 Embedded Controller Firmware
        License:          Proprietary
        Size:             791.7 kB
        Created:          2016-07-08
        Urgency:          High
        Vendor:           Lenovo Ltd.
        Flags:            is-upgrade
        Description:      
        Lenovo ThinkPad T480 Embedded Controller Firmware.
        
        The computer will be restarted automatically after updating BIOS completely . Do NOT turn off your computer or remove the AC adaptor while update is in progress.
        
        Update item in this release are applicable to Windows OS only.
      
# sudo fwupdmgr update
Decompressing…           [-                                      ]Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
Upgrade available for Embedded Controller from 0.1.21 to 0.1.22
20L5000BBM must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
Downloading…             [***************************************]
failed to compile 3952e58e94a21e91cc960b52badd41d4eccb7b0f:prefix=components: Error on line 46 char 2: Document must begin with an element (e.g. <book>)
```
**Expected behavior**
A clear and concise description of what you expected to happen.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

**System UEFI configuration**
Please provide the output of the following commands:

```shell
sudo efibootmgr -v
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0017,0018,0019,001A,001B,001C,001D,001E,001F,0001
Boot0000* opensuse-secureboot   HD(1,GPT,29439c0f-02ed-47a2-a846-f75aba44780b,0x800,0xfa000)/File(\EFI\opensuse\shim.efi)
Boot0001* Linux-Firmware-Updater        HD(1,GPT,29439c0f-02ed-47a2-a846-f75aba44780b,0x800,0xfa000)/File(\EFI\opensuse\fwupdx64.efi)
Boot0010  Setup FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu     FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen      FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics    FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu        FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery   FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key  FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD        VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe1 VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a401)
Boot001A* NVMe0 VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001B* ATA HDD0      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f602)
Boot001C* USB HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001D* PCI LAN       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001E  Other CD      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001F  Other HDD     VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0020* USBR BOOT CDROM       PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0021* USBR BOOT Floppy      PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0022* ATA HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0023* ATAPI CD      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
```

```shell
efivar -l | grep fw
```

```shell
tree /boot
```

**Additional questions**

- Operating system and version:
- openSUSE Tumbleweed
- Have you tried rebooting?
- Is this a regression?
- yes
- Are you using an NVMe disk?
- yes
- Is secure boot enabled?
- yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
- No

hughsie, 2021-08-03:
What fwupd version do you have installed?

shyukri, 2021-08-03:
Just updated the comment. it's `client version: 1.5.8`

shyukri, 2021-08-03:
`        Update item in this release are applicable to Windows OS only. ` should this explain it?

hughsie, 2021-08-03:
Can you update to 1.5.11 please. That contains the fix to load compressed archives.

shyukri, 2021-08-03:
I'll have to fix my [OBS](https://build.opensuse.org/) setup and submit 1.5.11 in openSUSE and retry :).
Thanks a lot.

scaleoutsean, 2021-08-04:
I have 1.5.11 and can't see that it fixes this problem (on Ubuntu 20.04).

```sh
$ sudo fwupdmgr get-updates
...
$ sudo fwupdmgr update     
Devices with no available firmware updates: 
 • Thunderbolt host controller
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Upgrade available for Embedded Controller from 0.1.21 to 0.1.22
20L5CTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: Y
Downloading…             [***************************************]
Decompressing…           [***************************************]failed to compile 5b41632e511b0a4f42b665ea95f8dd8acb95669a:prefix=components: Error on line 46 char 2: Document must begin with an element (e.g. <book>)

$  sudo fwupdmgr --version  
client version:	1.5.11
compile-time dependency versions
	gusb:	0.3.4

Decompressing…           [-                                      ]daemon version:	1.5.11
```

hughsie, 2021-08-04:
It's fixed in 1.5.12

closed 2021-09-19 by stale[bot]