fwupd/fwupd #3593: Support for --allow-reinstall with update command

djcampello, 2021-08-03:
This would allow to reinstall the same update while using the `update` command and getting the most up-to-date version from a remote. This is particularly useful for multiple runs of the `test` plugin without having to restart the daemon to reset the plugin state to outdated.

cc @g2pop

hughsie, 2021-08-03:
isn't that just `fwupdmgr reinstall` by a different name?

djcampello, 2021-08-03:
:O Thank you Richard! I just completely ignored the existence of `reinstall`

djcampello, 2021-08-03:
Not quite the same... If firmware is out-dated `reinstall` would try to reinstall that version which might not be available.

djcampello, 2021-08-03:
But we can make our test work with it... So closing again...

g2pop, 2021-08-05:
It seems we need the "--allow-reinstall" flag in particular. We'll be running updates right after starting fwupd, but "fwupdmgr reinstall" won't work until we've run an update already. We would have to change the command and restart fwupd before every update command. If we have the flag, we only need to use "fwupdmgr update -v b585990a-003e-5270-89d5-3705a17f9a43 --allow-reinstall" the entire time.

superm1, 2021-08-10:
As an alternative to `--allow-reinstall` for the update command (which generally I don't think makes sense outside of the test plugin), can you instead "downgrade" the test device in between runs?

djcampello, 2021-08-10:
I think the main issue is that we would need to maintain another .cab for the other version.

superm1, 2021-08-10:
But wouldn't it also more closely model how a real device works?

I really don't think --allow-reinstall and update/upgrade should be going together.

djcampello, 2021-08-10:
What about having `install` use files on a remote? That should also solve the problem plus would be a nice feature to be able to pick particular releases without manually downloading them

superm1, 2021-08-10:
> What about having `install` use files on a remote? That should also solve the problem plus would be a nice feature to be able to pick particular releases without manually downloading them

That sounds like a good solution.

hughsie, 2023-02-20:
I think we've got this now:

	[hughsie@hughsie-work build (main %)]$ fwupdmgr install
	0.	Cancel
	1.	48b0e21e27b6bab9c8208a603e5b65d8a42344e5 (USB3.1 Hub)
	2.	26c4f1230928a9765029f8f2e5c82bef85f55d8e (USB3.1 Hub)
	3.	23cf6368c14a875f74c38a5a423518f38d8abbbc (ColorHug)
	4.	2292ae5236790b47884e37cf162dcf23bfcd1c60 (Embedded Controller)
	5.	6c0b61db31db54f515165deb7ddb40e4b777ce54 (HP 910 White BT Mouse)
	6.	349bb341230b1a86e5effe7dfe4337e1590227bd (Intel Management Engine)
	7.	a45df35ac0e948ee180fe216a5f703f32dda163f (System Firmware)
	8.	901515bb1e030acc17640c170ae1505f8e6d7766 (ThinkPad Thunderbolt 3 Dock USB Audio)
	9.	362301da643102b9f38477387e2193e57abaa590 (UEFI dbx)
	10.	07b9f6c3b0d723cac5ddb859b651b58fdfd1a4df (Unifying Receiver)
	11.	08e1798bf5d9cb56a0290b552cab6c1a371b5089 (WDC PC SN720 SDAQNTW-256G-1001)
	Choose device [0-11]: 3
	0.	Cancel
	1.	1.2.6
	2.	1.2.5
	3.	1.2.4
	4.	1.2.3
	5.	1.2.2
	6.	1.2.0

closed 2023-02-20 by hughsie