fwupd/fwupd #3596: fwupdmgr doesn't give a warning when updating a composite device with an older fw version

hardboprobot, 2021-08-04:
**Describe the bug**
`fwupdmgr` not warning about upgrading an older version when running `install` on a composite device where one of the involved firmwares has an older version than the one currently installed.

**Steps to Reproduce**
Using a .cab file for a composite device (parent and child devices with different firmwares), run `fwupdmgr install <fw>  --allow-reinstall` on the device so that the fw version of the child device is the same or higher than the one installed and the fw version of the parent device is older than the one currently installed.

**Expected behavior**
The client should stop with an error message (the version provided is older than the one installed on one of the devices). The daemon does show a warning message in the verbose output.

**fwupd version information**

```shell
client version:	1.6.1-281-gea563756
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.6.1-281-gea563756
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): source, private repo.

**fwupd log**

```
08:28:24:0192 FuMain               Called GetPlugins()
08:28:24:0193 FuMain               Called SetFeatureFlags(31)
08:28:24:0194 FuMain               Called Install(*,9)
08:28:24:0194 FuMain               got option reason
08:28:24:0194 FuMain               got option filename
08:28:24:0194 FuMain               got option allow-reinstall
08:28:24:0194 FuEngine             Emitting PropertyChanged('Status'='decompressing')
08:28:24:0194 FuMain               Emitting PropertyChanged('Status'='decompressing')
08:28:24:0194 FuCabinet            failed to verify MPR05.00_B0008_mezzy_ble_pro_prod.metainfo.xml: no JcatItem
08:28:24:0194 FuCabinet            processing file: MPR05.00_B0008_mezzy_ble_pro_prod.metainfo.xml
08:28:24:0194 FuCabinet            failed to verify MPR05.00_B0008_mezzy_ble_pro_s122v811.metainfo.xml: no JcatItem
08:28:24:0194 FuCabinet            processing file: MPR05.00_B0008_mezzy_ble_pro_s122v811.metainfo.xml
08:28:24:0194 XbSilo               compiling f12de1f1240dc87df984ebdef06beffbe4d7c19b:filename=MPR05.00_B0008_mezzy_ble_pro_prod.metainfo.xml:prefix=components…
08:28:24:0194 XbSilo               compiling 2f74bb9feb40fdc8abf6c0dc256e6c5dc5202239:filename=MPR05.00_B0008_mezzy_ble_pro_s122v811.metainfo.xml:prefix=components…
08:28:24:0195 FuCabinet            processing release: MPR05.00_B0008
08:28:24:0195 FuCabinet            processing release: 0x0112
08:28:24:0195 FuEngine             Emitting PropertyChanged('Status'='idle')
08:28:24:0195 FuMain               Emitting PropertyChanged('Status'='idle')
08:28:24:0195 FuMain               first pass requirement on 10aa9ef6f15c92170b40763d3e48800760483ba4:com.logitech.UnifyingReceiver.firmware failed: Specified firmware is older than installed 'MPR05.00_B0008 < MPR05.00_B0009'
08:28:24:0195 XbSilo               ignoring for OR statement: element name suggests is unknown in silo
08:28:24:0195 XbSilo               ignoring for OR statement: element name suggests is unknown in silo
08:28:24:0195 FuMain               Emitting PropertyChanged('Status'='waiting-for-auth')
08:28:24:0219 FuMain               Emitting PropertyChanged('Status'='idle')
08:28:24:0219 FuIdle               inhibiting: update
08:28:24:0219 FuEngine             composite update 1: 7ae617156b741d80c217fd6eb8ae15c89a9b9e6f
08:28:24:0219 FuPlugin             composite_prepare(dell_dock)
08:28:24:0220 FuHistory            modifying device SoftDevice [7ae617156b741d80c217fd6eb8ae15c89a9b9e6f]
08:28:24:0220 FuHistory            remove device SoftDevice [7ae617156b741d80c217fd6eb8ae15c89a9b9e6f]
08:28:24:0221 FuHistory            add device SoftDevice [7ae617156b741d80c217fd6eb8ae15c89a9b9e6f]
08:28:24:0223 FuDeviceList         no replug or re-enumerate required
08:28:24:0223 FuEngine             prepare -> FuLogitechHidPpRadio:
```

**fwupdmgr log**

```
sudo src/fwupdmgr install <MPR05.00_B0008_full.cab> --allow-reinstall
WARNING: This package has not been validated, it may not work properly.
Decompressing…           [***************************************]
Authenticating…          [-                                      ]
Authenticating…          [***************************************]
Installing on SoftDevice…[ -                                     ]
Action Required: Unifying Receiver needs to be manually restarted to complete the update. Please unplug it and plug it back again.
Installing on SoftDevice…[       \                               ]
Installing on Unifying Receiver (bootloader)…                    ]
Installing on SoftDevice…[************************************** ] Less than one minute remaining…
Installing on Unifying Receiver (bootloader)…******************* ]
Installing on SoftDevice…[***************************************]
Restarting device…       [***************************************]
Successfully installed firmware
```

hughsie, 2021-08-04:
so, IIUC,

**device** state.

    device: 0.1.2
    soft-device 3.4.5

then you try to install a **firmware** of:

    device: 0.1.1 (i.e. older)
    soft-device 3.4.5

...with `allow-reinstall` -- then I agree that should fail.  Did I understand correctly?

hardboprobot, 2021-08-05:
@hughsie that's correct.

hughsie, 2021-08-05:
I'm a bit confused:

    08:28:24:0195 FuMain               first pass requirement on 10aa9ef6f15c92170b40763d3e48800760483ba4:com.logitech.UnifyingReceiver.firmware failed: Specified firmware is older than installed 'MPR05.00_B0008 < MPR05.00_B0009'

That should have meant that we don't add a FuInstallTask, and sure enough:

    08:28:24:0219 FuEngine             composite update 1: 7ae617156b741d80c217fd6eb8ae15c89a9b9e6f

Is 7ae617156b741d80c217fd6eb8ae15c89a9b9e6f the radio device or the main device?

hughsie, 2021-08-05:
Could it be that the front-end is confusing you with the wrong device name and the engine is doing the right thing?

hardboprobot, 2021-08-05:
`10aa9ef6f15c92170b40763d3e48800760483ba4` is the main device and `7ae617156b741d80c217fd6eb8ae15c89a9b9e6f` is the radio.

I'm not sure the engine is doing the right thing. The daemon is definitely noticing that one of the firmwares is older than the installed one, but the end result is that:

1 - the radio software is the first one to get updated, as per the priority declared in the metadata
2 - after it's updated, the daemon doesn't proceed to install the main app fw but the client doesn't warn about this

If you need me to do more specific tests and capture logs, let me know.

hughsie, 2021-08-05:
> I'm not completely sure how to achieve that.

Maybe something like https://github.com/fwupd/fwupd/pull/3605 ??

hardboprobot, 2021-08-06:
@hughsie thanks for the PR. I tested #3605 and it doesn't have the effect that I was expecting. Actually, it prevents any kind of composite update on the devices that have that flag enabled:

```
sudo src/fwupdmgr install <RBM14_00_B0007_full.cab> --allow-reinstall
WARNING: This package has not been validated, it may not work properly.
Decompressing…           [***************************************]
device is required to be part of the composite update: No supported devices found
```

Anyway, this is not a hard requirement for me at all and that particular (and unlikely) scenario can be bypassed by installing with `--allow-older`. I just raised the issue in case it was a limitation that was oversighted.

closed 2021-09-19 by stale[bot]