fwupd/fwupd #3597: Ubuntu 18.04 to 20.04 upgrade, fwupd-refresh.service, Main process exited, code=exited, status=1/FAILURE

ajcollett, 2021-08-04:
**Describe the bug**
Hello!

I have a strange issue on 11 servers I run, all "desktop boxes" with Ubuntu server installed. 

All of these machines have been upgraded from 18.04, they all use LDAP via SSSD and are different in hardware, so it's not HW specific issues. However, it may have something to do with SSSD and LDAP, though I checked, and we don't have users in the range that the DynamicUser option in systemd service files makes use of. 

I have other servers (Virtual Machines, not baremetal) that have the same setup but work fine in this regard. 
However, they have fresh 20.04 installs, so I think there is something weird hanging around from the upgrade. 

Whenever `fwupd-refresh.service` runs, it fails with `Main process exited, code=exited, status=1/FAILURE` on those 11 servers. There is so little in the logs, that I ended up commenting out bits in the service files, and doing Strace. 

That's how I know it's related to the DynamicUser option. With that commented out, and when run manually on the cmd line, the program runs with no issue. Adding --verbose doesn't help either. 

I also did:

```shell
sudo apt autoremove --purge fwupd
sudo apt install fwupd
```

With no luck. I am pretty certain this has to do with the the DynamicUsers, but I cannot see what or why. The directories for the dynamic user exist and see to be similar to the directories on working machines. 

**Steps to Reproduce**
Not sure, but what I did:
1. Upgrade machine from 18.04 to 20.04
2. Use SSSD to setup LDAP (was already done in 18.04)
3. Use the system normally, observe failing service. 

**Expected behavior**
No error, the update should happen, and have a exit code of 0 or 2, not 1. 

**fwupd version information**
Installed with default Ubuntu setup.

```shell
client version: 1.5.11
compile-time dependency versions
        gusb:   0.3.4

daemon version: 1.5.11
```

**fwupd device information**
So, this is one system, as an example. As far as I know, there are no fwupdates here. 

I also have other systems that also fail, that use UEFI, so it shouldn't be related to that. 

```shell
WARNING: UEFI firmware can not be updated in legacy BIOS mode
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:legacy-bios for more information.
7033W2D
│
├─        Intel(R) Core™ i5-2400 CPU @ 3.10GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x0000002f
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         87b4a46c-74e8-5075-a2f0-8bb470d5cf18 ← CPUID\PRO_0&FAM_06&MOD_2A
│                         80fcd4ac-c2a2-55db-b4c9-973373b86677 ← CPUID\PRO_0&FAM_06&MOD_2A&STP_7
│     Device Flags:       • Internal device
│   
├─2nd Generation Core Processor Family Integrated Graphics Controller:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    09
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              2c4f7111-7383-5054-9a9d-3041a56062e8 ← PCI\VEN_8086&DEV_0102&SUBSYS_17AA3070&REV_09
│                         5c454891-4324-5e23-ab65-e1507fdb83dd ← PCI\VEN_8086&DEV_0102&SUBSYS_17AA3070
│                         ae4b97d5-ca2f-5770-9445-b5a78edd6f19 ← PCI\VEN_8086&DEV_0102&REV_09
[... 76 lines cut ...]
futex(0x7fe86d6faf38, FUTEX_WAKE_PRIVATE, 2147483647) = 0
futex(0x7fe86d6faf38, FUTEX_WAKE_PRIVATE, 2147483647) = 0
rt_sigaction(SIGPIPE, {sa_handler=SIG_IGN, sa_mask=[PIPE], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0x7fe86d1fc210}, {sa_handler=SIG_IGN, sa_mask=[], sa_flags=0}, 8) = 0
futex(0x7fe86d6faf38, FUTEX_WAKE_PRIVATE, 2147483647) = 0
futex(0x7fe86d6faf38, FUTEX_WAKE_PRIVATE, 2147483647) = 0
futex(0x7fe86d6faf38, FUTEX_WAKE_PRIVATE, 2147483647) = 0
futex(0x7fe86d6faf38, FUTEX_WAKE_PRIVATE, 2147483647) = 0
futex(0x7fe86d6faf38, FUTEX_WAKE_PRIVATE, 2147483647) = 0
futex(0x7fe86d6faf38, FUTEX_WAKE_PRIVATE, 2147483647) = 0
mmap(NULL, 8392704, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS|MAP_STACK, -1, 0) = 0x7fe868726000
mprotect(0x7fe868727000, 8388608, PROT_READ|PROT_WRITE) = 0
clone(child_stack=0x7fe868f25c70, flags=CLONE_VM|CLONE_FS|CLONE_FILES|CLONE_SIGHAND|CLONE_THREAD|CLONE_SYSVSEM|CLONE_SETTLS|CLONE_PARENT_SETTID|CLONE_CHILD_CLEARTID, parent_tid=[240570], tls=0x7fe868f26700, child_tidptr=0x7fe868f269d0) = 240570
poll([{fd=3, events=POLLIN}], 1, -1)    = 1 ([{fd=3, revents=POLLIN}])
read(3, "\1\0\0\0\0\0\0\0", 16)         = 8
poll([{fd=3, events=POLLIN}], 1, -1)    = 1 ([{fd=3, revents=POLLIN}])
read(3, "\1\0\0\0\0\0\0\0", 16)         = 8
futex(0x7fe86d6faf38, FUTEX_WAKE_PRIVATE, 2147483647) = 0
write(3, "\1\0\0\0\0\0\0\0", 8)         = 8
futex(0x5560f4bf1460, FUTEX_WAKE_PRIVATE, 2147483647) = 0
openat(AT_FDCWD, "/usr/lib/x86_64-linux-gnu/charset.alias", O_RDONLY) = -1 ENOENT (No such file or directory)
write(2, "Failed to connect to daemon: Exh"..., 119) = 119
close(3)                                = 0
exit_group(1)                           = ?
+++ exited with 1 +++
```

hughsie, 2021-08-04:
@superm1 anything you can think of? I think this is probably a downstream bug and probably belongs in launchpad...

superm1, 2021-08-05:
I would suspect something related to the directories normally created by systemd when DynamicUser is set. Can you compare the permissions or presence of fwupd related directories in /var on your two machines (working with dynamic users and not)?

Possibly related to them being made before upgrade to a version of fwupd that supported dynamic user in the unit.

If this hypothesis is correct then I think we need some packaging change on cleanup that is missing, but let's better characterize what's different between your machines.

ajcollett, 2021-08-06:
Hi, 

So... I don't have helpful news, I don't think.

I looked at another machine that had 18.04 still on it, and realised how differently fwupd is setup between 18.04 and 20.04. 

I got a few more directories to compare, but I can't see anything that might cause this. More over, when swapping Dynamic user to false, those Dynamic directories get removed and re-created. The private ones anyway. 

So I cannot see that being as issue. 

I looked all over for logs relating to SSSD and users, but nothing pops when the service fails. I stopped SSSD, but that made no difference either. 

I also don't know if this is a fault with the Dynamic users, or just this one Unit, since there are no other Unit files on my systems that also use Dynamic users... 

... <time passes> ...

Okay so I went and used some examples on the below website to test DynamicUser. I even used the same Cache directory as fwupd and I got the same UID. So there's nothing wrong with becoming that user with that ID. I was able to create a file in the cache directory too. 

http://0pointer.net/blog/dynamic-users-with-systemd.html

That makes me wonder though, at the /var/lib/fwupd directory. It's fairly empty on the affected nodes, and the service has no write perms to set that up. Where are those files supposed to come from?

If I run the service on the CMD, it works, and those directories are populated, but the service still fails if I run that afterwards.

So here's the output from `find /var/ -name *fwupd* | xargs ls -lhaR {}`
Node where service works:
```shell
   lrwxrwxrwx 1 root  root    16 Jul 21 23:53 /var/cache/fwupdmgr -> private/fwupdmgr
    -rw-r--r-- 1 root  root   277 Feb  1  2021 /var/lib/dpkg/info/fwupd-signed.list
    -rw-r--r-- 1 root  root   292 Jun 18  2020 /var/lib/dpkg/info/fwupd-signed.md5sums
    -rw-r--r-- 1 root  root   653 Jun 18  2020 /var/lib/dpkg/info/fwupd.conffiles
    -rw-r--r-- 1 root  root   19K Feb  1  2021 /var/lib/dpkg/info/fwupd.list
    -rw-r--r-- 1 root  root   25K Jun 18  2020 /var/lib/dpkg/info/fwupd.md5sums
    -rwxr-xr-x 1 root  root  2.5K Jun 18  2020 /var/lib/dpkg/info/fwupd.postinst
    -rwxr-xr-x 1 root  root  1.1K Jun 18  2020 /var/lib/dpkg/info/fwupd.postrm
    -rwxr-xr-x 1 root  root   634 Jun 18  2020 /var/lib/dpkg/info/fwupd.preinst
    -rwxr-xr-x 1 root  root   304 Jun 18  2020 /var/lib/dpkg/info/fwupd.prerm
    -rw-r--r-- 1 root  root   258 Feb  1  2021 /var/lib/dpkg/info/libfwupd2:amd64.list
    -rw-r--r-- 1 root  root   223 Jun 18  2020 /var/lib/dpkg/info/libfwupd2:amd64.md5sums
    -rw-r--r-- 1 root  root    33 Jun 18  2020 /var/lib/dpkg/info/libfwupd2:amd64.shlibs
    -rw-r--r-- 1 root  root   13K Jun 18  2020 /var/lib/dpkg/info/libfwupd2:amd64.symbols
    -rw-r--r-- 1 root  root    73 Jun 18  2020 /var/lib/dpkg/info/libfwupd2:amd64.triggers
    -rw-r--r-- 1 root  root   333 Feb  1  2021 /var/lib/dpkg/info/libfwupdplugin1:amd64.list
    -rw-r--r-- 1 root  root   235 Jun 18  2020 /var/lib/dpkg/info/libfwupdplugin1:amd64.md5sums
    -rw-r--r-- 1 root  root    45 Jun 18  2020 /var/lib/dpkg/info/libfwupdplugin1:amd64.shlibs
    -rw-r--r-- 1 root  root   20K Jun 18  2020 /var/lib/dpkg/info/libfwupdplugin1:amd64.symbols
    -rw-r--r-- 1 root  root    73 Jun 18  2020 /var/lib/dpkg/info/libfwupdplugin1:amd64.triggers
    -rw-r--r-- 1 root  root   135 Apr  1  2020 /var/lib/polkit-1/localauthority/10-vendor.d/fwupd.pkla
    -rw-r--r-- 1 root  root    60 Feb  1  2021 /var/lib/systemd/deb-systemd-helper-enabled/fwupd-refresh.timer.dsh-also
    -rw-r--r-- 1 root  root     0 Feb  1  2021 /var/lib/systemd/deb-systemd-helper-enabled/timers.target.wants/fwupd-refresh.timer
    -rw-r--r-- 1 root  root     0 Aug  6 03:44 /var/lib/systemd/timers/stamp-fwupd-refresh.timer
    
    /var/cache/fwupd:
    total 3.6M
    drwxr-xr-x  2 root root 4.0K Aug  6 03:44 .
    drwxr-xr-x 15 root root 4.0K Jul 21 23:53 ..
    -rw-r--r--  1 root root 3.6M Aug  6 03:44 metadata.xmlb
    -rw-r--r--  1 root root 1.6K Jul 21 23:53 metainfo.xmlb
    -rw-r--r--  1 root root  51K Jul 21 23:53 quirks.xmlb
    
    /var/cache/private/fwupdmgr:
    total 12K
    drwxr-xr-x 3 62803 62803 4.0K Jul 21 23:53 .
    drwx------ 3 root  root  4.0K Jul 21 23:53 ..
    drwxr-xr-x 2 62803 62803 4.0K Aug  6 03:44 fwupd
    
    /var/cache/private/fwupdmgr/fwupd:
    total 736K
    drwxr-xr-x 2 62803 62803 4.0K Aug  6 03:44 .
    drwxr-xr-x 3 62803 62803 4.0K Jul 21 23:53 ..
[... 217 lines cut ...]
    drwxr-xr-x 3 root root 4.0K Aug  6 11:04 .
    drwxr-xr-x 6 root root 4.0K Aug  6 11:04 ..
    drwxr-xr-x 2 root root 4.0K Aug  6 11:07 lvfs

    /var/lib/fwupd/remotes.d/lvfs:
    total 736K
    drwxr-xr-x 2 root root 4.0K Aug  6 11:07 .
    drwxr-xr-x 3 root root 4.0K Aug  6 11:04 ..
    -rw-r--r-- 1 root root 722K Aug  6 11:07 metadata.xml.gz
    -rw-r--r-- 1 root root  488 Aug  6 11:07 metadata.xml.gz.asc

    /var/tmp/systemd-private-fdf5178b012a41b799e676c05d514186-fwupd.service-1D1Njh:
    total 12K
    drwx------ 3 root root 4.0K Aug  6 11:04 .
    drwxrwxrwt 6 root root 4.0K Aug  6 11:06 ..
    drwxrwxrwt 2 root root 4.0K Aug  6 11:04 tmp

    /var/tmp/systemd-private-fdf5178b012a41b799e676c05d514186-fwupd.service-1D1Njh/tmp:
    total 8.0K
    drwxrwxrwt 2 root root 4.0K Aug  6 11:04 .
    drwx------ 3 root root 4.0K Aug  6 11:04 ..

```

Are there any other directories I should compare?

ajcollett, 2021-08-20:
So, I reinstalled one of the servers from Scratch, and it's happy now. So there is something being left over, maybe even in systemd, that makes this fail. I am not sure how to debug this further.

g2p, 2022-04-22:
I used the following to troubleshoot a similar issue:

    sudo systemd-run --wait -pEnvironment=LANG=C -pSuccessExitStatus=2 -pProtectSystem=strict -pProtectHome=read-only -pPrivateTmp=yes -pRemoveIPC=yes -pDynamicUser=yes --pipe fwupdmgr refresh

With this output:

    Failed to connect to daemon: Exhausted all available authentication mechanisms (tried: EXTERNAL) (available: EXTERNAL)

Which pointed to #3037 (and https://github.com/systemd/systemd/issues/22737) as a cause.

bernd-wechner, 2022-06-03:
For what it's worth, I have [cockpit](https://cockpit-project.org/) installed on my servers and noticed this on one of them:

[image]

which landed me here. No idea what that means. But seems @g2p things it has https://github.com/fwupd/fwupd/issues/3037 (and https://github.com/systemd/systemd/issues/22737) as a cause.