fwupd/fwupd #3608: fwupd stopped working after update to 1.6.2

iyanmv, 2021-08-05:
**Describe the bug**
`fwupd` stopped working for me after updating from 1.6.1 to 1.6.2. I am using Arch Linux in a Lenovo Thinkpad X1 Yoga (gen 6). I opened a bug [here](https://bugs.archlinux.org/task/71740?string=fwupd&project=5&search_name=&type%5B0%5D=&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=open&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=) but they recommended me opening upstream since it doesn't look like a packaging issue.

When I run, for example, `fwupdmgr refresh`, after a few seconds doing nothing, I get the following error:

```
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
```
I copy here the output from `dmesg` and `systemctl status`:

```
fwupd[2252]: segfault at 0 ip 0000000000000000 sp 00007ffe9c818aa8 error 14 in fwupd[55c66defa000+10000]
[  159.385998] Code: Unable to access opcode bytes at RIP 0xffffffffffffffd6.

fwupd.service - Firmware update daemon
     Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static)
     Active: failed (Result: core-dump) since Thu 2021-08-05 16:22:36 CEST; 4min 21s ago
       Docs: https://fwupd.org/
    Process: 2767 ExecStart=/usr/lib/fwupd/fwupd (code=dumped, signal=SEGV)
   Main PID: 2767 (code=dumped, signal=SEGV)
        CPU: 302ms

Aug 05 16:22:35 thinkpad fwupd[2767]: 14:22:35:0999 FuPluginDfu          unable to guess initial device state from interface 0
Aug 05 16:22:35 thinkpad fwupd[2767]: 14:22:35:0999 FuPluginDfu          unable to guess initial device state from interface 0
Aug 05 16:22:35 thinkpad fwupd[2767]: 14:22:35:0999 FuPluginDfu          unable to guess initial device state from interface 0
Aug 05 16:22:35 thinkpad fwupd[2767]: 14:22:35:0999 FuPluginDfu          DFU version 0xe1c0 invalid, v1.1 assumed
Aug 05 16:22:35 thinkpad fwupd[2767]: 14:22:35:0999 FuPluginDfu          DFU version 0x0c00 invalid, v1.1 assumed
Aug 05 16:22:35 thinkpad fwupd[2767]: 14:22:35:0999 FuPluginDfu          13d3:5411 is missing download capability
Aug 05 16:22:36 thinkpad fwupd[2767]: 14:22:36:0459 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:1f.5: failed to mmap SPIBAR: Operation not permitted
Aug 05 16:22:36 thinkpad systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
Aug 05 16:22:36 thinkpad systemd[1]: fwupd.service: Failed with result 'core-dump'.
Aug 05 16:22:36 thinkpad systemd[1]: Failed to start Firmware update daemon.
```

**Steps to Reproduce**
Install `fwupd` package from Arch Linux repos and try to run any command from `fwupdmgr`.

**Additional questions**

- Operating system and version: Arch Linux
- Have you tried rebooting? Yes
- Is this a regression? It looks like it is

Please let me know if I could provide any other logs or information that may be helpful. Everything was working fine with 1.6.1.

hughsie, 2021-08-05:
Is this fixed by https://github.com/fwupd/fwupd/pull/3604 ? if so, I should probably tag 1.6.3....

PHLAK, 2021-08-05:
I'm seeing what I believe is the same thing.

```
Aug 05 11:16:23 Blurr systemd[1]: Starting Firmware update daemon...
Aug 05 11:16:23 Blurr fwupd[51410]: 18:16:23:0657 FuPluginDfu          DFU version 0xe1c0 invalid, v1.1 assumed
Aug 05 11:16:23 Blurr fwupd[51410]: 18:16:23:0657 FuPluginDfu          unable to guess initial device state from interface 0
Aug 05 11:16:23 Blurr fwupd[51410]: 18:16:23:0657 FuPluginDfu          DFU version 0x0300 invalid, v1.1 assumed
Aug 05 11:16:23 Blurr fwupd[51410]: 18:16:23:0657 FuPluginDfu          unable to guess initial device state from interface 0
Aug 05 11:16:23 Blurr fwupd[51410]: 18:16:23:0657 FuPluginDfu          unable to guess initial device state from interface 0
Aug 05 11:16:23 Blurr fwupd[51410]: 18:16:23:0657 FuPluginDfu          unable to guess initial device state from interface 0
Aug 05 11:16:23 Blurr fwupd[51410]: 18:16:23:0657 FuPluginDfu          unable to guess initial device state from interface 0
Aug 05 11:16:23 Blurr fwupd[51410]: 18:16:23:0657 FuPluginDfu          unable to guess initial device state from interface 0
Aug 05 11:16:23 Blurr fwupd[51410]: 18:16:23:0657 FuPluginDfu          unable to guess initial device state from interface 0
Aug 05 11:16:23 Blurr fwupd[51410]: 18:16:23:0657 FuPluginDfu          unable to guess initial device state from interface 0
Aug 05 11:16:23 Blurr fwupd[51410]: 18:16:23:0657 FuPluginDfu          unable to guess initial device state from interface 0
Aug 05 11:16:23 Blurr fwupd[51410]: 18:16:23:0657 FuPluginDfu          DFU version 0xe1c0 invalid, v1.1 assumed
Aug 05 11:16:23 Blurr fwupd[51410]: 18:16:23:0657 FuPluginDfu          DFU version 0x0c00 invalid, v1.1 assumed
Aug 05 11:16:23 Blurr fwupd[51410]: 18:16:23:0657 FuPluginDfu          13d3:5411 is missing download capability
Aug 05 11:16:23 Blurr fwupd[51410]: 18:16:23:0845 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:1f.5: failed to mmap S>
Aug 05 11:16:24 Blurr systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
Aug 05 11:16:24 Blurr systemd[1]: fwupd.service: Failed with result 'core-dump'.
Aug 05 11:16:24 Blurr systemd[1]: Failed to start Firmware update daemon.
```

I'm currently unable to test #3604 at the moment but will try to get to that later.

outofink, 2021-08-05:
I had the same issue, #3604 fixes things for me.

iyanmv, 2021-08-09:
@hughsie yes, that patch fixes the issue for me.

superm1, 2021-08-09:
@hughsie I think it's worth branching 1_6_X  before the ABI change and spinning a 1.6.3 with that and moving the ABI change into 1.7.x (or 2.x.x?)

hughsie, 2021-08-09:
Good idea. Done!

mdedetrich, 2021-08-14:
I can also confirm that I am getting the same problem

superm1, 2021-08-18:
fixed in 1.6.3

closed 2021-08-18 by superm1

vishnumur777, 2023-07-22:
this issue raised me at the latest version 1.9.3-1 version 

failed to add device /sys/devices/pci0000:00/0000:00:1f.0: failed to mmap spibar: operation not permitted

hughsie, 2023-07-22:
> failed to mmap spibar: operation not permitted

This is a different issue -- you probably need to disable intel spi.

vishnumur777, 2023-07-22:
how to do that
is there any documentation or provide steps to disable intel spi.

hughsie, 2023-07-24:
You need to compile fwupd with `-Dplugin_intel_spi=false`