fwupd/fwupd #3616: sudo dpkg --configure -a

LocalHeroPro, 2021-08-08:
**Describe the bug**
Can't run command `apt update` or `apt dist-upgrade`

**Steps to Reproduce**
```shell
$ sudo apt upgrade
// ...
 Setting up fwupd (1.5.11-0ubuntu1~20.04.2) ...###################################################################################################################################################...........................................................................] 
fwupd-offline-update.service is a disabled or a static unit not running, not starting it.
fwupd-refresh.service is a disabled or a static unit not running, not starting it.
// here we have freeze
```

**Expected behavior**
Can update OS.

**fwupd version information**

```shell
client version: 1.5.11
compile-time dependency versions
        gusb:   0.3.4

daemon version: 1.5.11
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
Super Server
│
├─ASPEED Graphics Family:
│     Device ID:          0ec05f487924f76b6f4eee3433bdee489638de9d
│     Current version:    30
│     Vendor:             ASPEED Technology, Inc. (PCI:0x1A03)
│     GUIDs:              833ebd32-7454-5e3f-9e74-a3cd70196910
│                         35282f5b-9392-5d88-ac31-eff56e28e442
│                         1e32a628-412d-5613-bcbd-b86dd11a86ab
│                         237d7407-66ed-5408-852b-2d5d2647b131
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Intel(R) Xeon(R) CPU E3-1230 v6 @ 3.50GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000ea
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         809a0b93-8a12-5338-a571-ad5583acf896
│                         d0f754d5-1395-5573-bc83-85ba955da70a
│     Device Flags:       • Internal device
│   
├─SAS3008 PCI-Express Fusion-MPT SAS-3 (AOC-S3008L-L8e):
│     Device ID:          5987c136e0024e0f45943927a31c5c4c711f3eb0
│     Current version:    02
│     Vendor:             Broadcom / LSI (PCI:0x1000)
│     GUIDs:              89f86806-bbd8-5323-a920-fa03de53ff69
│                         371cde05-a3e3-544a-a682-992f475e97d3
│                         e81e38cf-0c3d-5116-bb56-9f0d7ae87bd6
│                         4c9845e9-44c3-5652-9da6-7b347f46ac41
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
└─System Firmware:
  │   Device ID:          1d70a31dd3ff0941761550a38436fc990c6c3ff0
  │   Current version:    5
  │   Minimum Version:    5
  │   Vendor:             Supermicro (DMI:American Megatrends Inc.)
  │   GUIDs:              7039436b-6acf-433b-86a1-368ec2ef7e1f
  │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
  │                       733edd7c-75ee-5db0-8a0a-3138c35f755c
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─UEFI dbx:
        Device ID:        362301da643102b9f38477387e2193e57abaa590
        Summary:          UEFI Revocation Database
        Current version:  238
        Minimum Version:  238
        Vendor:           UEFI:Linux Foundation
        Install Duration: 1 second
        GUIDs:            c6682ade-b5ec-57c4-b687-676351208742
                          f8ba2887-9411-5c36-9cee-88995bb39731
        Device Flags:     • Internal device
                          • Updatable
                          • Needs a reboot after installation
```

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?

LocalHeroPro, 2021-08-08:
```shell
$ sudo dpkg --configure -a
Setting up libzfs2linux (0.8.3-1ubuntu12.12) ...
Setting up xserver-xephyr (2:1.20.11-1ubuntu1~20.04.2) ...
Setting up libzpool2linux (0.8.3-1ubuntu12.12) ...
Setting up libdrm2:amd64 (2.4.105-3~20.04.1) ...
Setting up fwupd (1.5.11-0ubuntu1~20.04.2) ...
fwupd-offline-update.service is a disabled or a static unit not running, not starting it.
fwupd-refresh.service is a disabled or a static unit not running, not starting it.
fwupd.service is a disabled or a static unit not running, not starting it.
```

superm1, 2021-08-08:
This should be a downstream distro bug.

I would suggest to modify /var/lib/dpkg/info/fwupd.postinst and add a set -x near the top to see what command it really hangs last on.

Add that to the downstream report.

closed 2021-08-17 by superm1

LocalHeroPro, 2021-09-01:
I set:
```shell
#!/bin/sh
set -e -x
//
```
and run
```shell
$ sudo dpkg --configure -a
Setting up libdrm-nouveau2:amd64 (2.4.105-3~20.04.1) ...
Setting up libgbm1:amd64 (21.0.3-0ubuntu0.2~20.04.1) ...
Setting up libdrm-radeon1:amd64 (2.4.105-3~20.04.1) ...
Setting up libdrm-intel1:amd64 (2.4.105-3~20.04.1) ...
Setting up xserver-xorg-core (2:1.20.11-1ubuntu1~20.04.2) ...
Setting up libxatracker2:amd64 (21.0.3-0ubuntu0.2~20.04.1) ...
Setting up zfsutils-linux (0.8.3-1ubuntu12.12) ...
zfs-import-scan.service is a disabled or a static unit, not starting it.
zfs-import-scan.service is a disabled or a static unit, not starting it.
Setting up libegl-mesa0:amd64 (21.0.3-0ubuntu0.2~20.04.1) ...
Setting up xwayland (2:1.20.11-1ubuntu1~20.04.2) ...
Setting up fwupd (1.5.11-0ubuntu1~20.04.2) ...
+ dpkg-maintscript-helper rm_conffile /etc/fwupd.conf 1.0.0~ -- configure 1.3.11-1~focal1
+ dpkg-maintscript-helper rm_conffile /etc/fwupd/remotes.d/fwupd.conf 1.2.7~ -- configure 1.3.11-1~focal1
+ dpkg-maintscript-helper rm_conffile /etc/dbus-1/system.d/org.freedesktop.fwupd.conf 1.3.2~ -- configure 1.3.11-1~focal1
+ dpkg-maintscript-helper rm_conffile /etc/modules-load.d/fwupd-msr.conf 1.5.3~ -- configure 1.3.11-1~focal1
+ dpkg-maintscript-helper rm_conffile /etc/modules-load.d/fwupd-platform-integrity.conf 1.5.3~ -- configure 1.3.11-1~focal1
+ [ configure = configure ]
+ deb-systemd-helper unmask fwupd-refresh.timer
+ deb-systemd-helper --quiet was-enabled fwupd-refresh.timer
+ deb-systemd-helper enable fwupd-refresh.timer
+ [ configure = configure ]
+ [ -d /run/systemd/system ]
+ systemctl --system daemon-reload
+ [ -n 1.3.11-1~focal1 ]
+ _dh_action=restart
+ deb-systemd-invoke restart fwupd-offline-update.service fwupd-refresh.service fwupd-refresh.timer fwupd.service
fwupd-offline-update.service is a disabled or a static unit not running, not starting it.
fwupd-refresh.service is a disabled or a static unit not running, not starting it.
fwupd.service is a disabled or a static unit not running, not starting it.
```
and get frozen .............
ctrl + c
```shell
^Cdpkg: error processing package fwupd (--configure):
 installed fwupd package post-installation script subprocess was interrupted
Setting up libdbi-perl:amd64 (1.643-1ubuntu0.1) ...
Setting up libdrm-amdgpu1:amd64 (2.4.105-3~20.04.1) ...
Setting up mesa-vulkan-drivers:amd64 (21.0.3-0ubuntu0.2~20.04.1) ...
Setting up libgl1-mesa-dri:amd64 (21.0.3-0ubuntu0.2~20.04.1) ...
dpkg: dependency problems prevent configuration of fwupd-signed:
 fwupd-signed depends on fwupd (= 1.5.11-0ubuntu1~20.04.2); however:
  Package fwupd is not configured yet.

dpkg: error processing package fwupd-signed (--configure):
 dependency problems - leaving unconfigured
Setting up zfs-zed (0.8.3-1ubuntu12.12) ...
Setting up mariadb-server-10.4 (1:10.4.21+maria~focal) ...
Setting up libglx-mesa0:amd64 (21.0.3-0ubuntu0.2~20.04.1) ...
Setting up mariadb-server (1:10.4.21+maria~focal) ...
Processing triggers for systemd (245.4-4ubuntu3.11) ...
Processing triggers for man-db (2.9.1-1) ...
Processing triggers for libc-bin (2.31-0ubuntu9.2) ...
Errors were encountered while processing:
 fwupd
 fwupd-signed
```

superm1, 2021-09-01:
Okay, add that to your downstream report in Ubuntu.