fwupd/fwupd #3621: Fwupdmgr doesn't understand multiple ESPs (again)

c0deright, 2021-08-10:
**Describe the bug**

I have a RAID1 setup with a cron job synchronizing the non-RAID-ESP from the first SSD to the non-RAID-ESP on the second ESP. Both RAID-1 members have their own seperate ESP so the system survives a missing/damaged disk.

Having two ESPs breaks fwupd (again). Please see https://github.com/fwupd/fwupd/issues/2843 for an old report on exactly the same issue.

**Steps to Reproduce**

```
fwupdmgr
WARNING: UEFI ESP partition not detected or configured
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
Command not found

Use fwupdmgr --help for help
```

```
root@p15:~>/usr/bin/fwupdtool esp-list --verbose
11:34:12:0071 FuDebug              Verbose debugging enabled (on console 1)
11:34:12:0096 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme1n1p3, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
11:34:12:0100 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
11:34:12:0104 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme1n1p2, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
11:34:12:0107 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
11:34:12:0112 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
11:34:12:0113 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme1n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
Choose a volume:
0.	Cancel
1.	/org/freedesktop/UDisks2/block_devices/nvme0n1p1
2.	/org/freedesktop/UDisks2/block_devices/nvme1n1p1
1
/boot/efi/EFI/ubuntu/grubx64.efi
/boot/efi/EFI/ubuntu/shimx64.efi
/boot/efi/EFI/ubuntu/mmx64.efi
/boot/efi/EFI/ubuntu/BOOTX64.CSV
/boot/efi/EFI/ubuntu/grub.cfg
/boot/efi/EFI/ubuntu/fwupdx64.efi
/boot/efi/EFI/BOOT/BOOTX64.EFI
/boot/efi/EFI/BOOT/fbx64.efi
/boot/efi/EFI/BOOT/mmx64.efi
root@p15:~>/usr/bin/fwupdtool esp-list --verbose
11:34:14:0415 FuDebug              Verbose debugging enabled (on console 1)
11:34:14:0444 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme1n1p3, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
11:34:14:0448 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
11:34:14:0451 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme1n1p2, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
11:34:14:0454 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
11:34:14:0459 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
11:34:14:0461 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme1n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
Choose a volume:
0.	Cancel
1.	/org/freedesktop/UDisks2/block_devices/nvme0n1p1
2.	/org/freedesktop/UDisks2/block_devices/nvme1n1p1
2
/boot/efi-backup/EFI/ubuntu/grubx64.efi
/boot/efi-backup/EFI/ubuntu/shimx64.efi
/boot/efi-backup/EFI/ubuntu/mmx64.efi
/boot/efi-backup/EFI/ubuntu/BOOTX64.CSV
/boot/efi-backup/EFI/ubuntu/grub.cfg
/boot/efi-backup/EFI/ubuntu/fwupdx64.efi
/boot/efi-backup/EFI/BOOT/BOOTX64.EFI
/boot/efi-backup/EFI/BOOT/fbx64.efi
/boot/efi-backup/EFI/BOOT/mmx64.efi
```

**Expected behavior**

fwupdmgr should work without that error

**fwupd version information**

[... 266 lines cut ...]
/dev/nvme0n1p1 on /boot/efi type vfat (rw,relatime,fmask=0077,dmask=0077,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/nvme1n1p1 on /boot/efi-backup type vfat (rw,relatime,fmask=0077,dmask=0077,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
```

**/etc/fwupd/uefi_capsule.conf**

```
[uefi_capsule]

# the shim loader is required to chainload the fwupd EFI binary unless
# the fwupd.efi file has been self-signed manually
#DisableShimForSecureBoot=true

# the EFI system partition path used
# if this is is not /boot/efi, /boot, or /efi
#OverrideESPMountPoint=
OverrideESPMountPoint=/boot/efi

# amount of free space required on the ESP, for example using 0x2000000 for 32Mb
#RequireESPFreeSpace=

# with the UEFI removable path enabled, the default esp path is set to /EFI/boot
# the shim EFI binary and presumably this is $ESP/EFI/boot/bootx64.efi
#FallbacktoRemovablePath=false
```

hughsie, 2021-08-10:
I'm about to go on extended PTO, could you please debug the issue a little and try to find why the ESP is not being detected. It might also be worth testing 1.6.3 to see if the issue has already been fixed in a newer version.

c0deright, 2021-08-10:
Can you give me some hints? I could try strace or is there anyhing else that might reveal which ESP fwupd chooses?

hughsie, 2021-08-10:
https://github.com/fwupd/fwupd/blob/master/libfwupdplugin/fu-common.c#L2978 gets called when it's not specified manually. I suspect something isn't readying the config file and just blindly calling fu_common_get_esp_default instead.

superm1, 2021-08-10:
Can you try to check the daemon log for errors about `invalid OverrideESPMountPoint`?

c0deright, 2021-08-11:
> Can you try to check the **daemon** log for errors about `invalid OverrideESPMountPoint`?

Yikes, now as you said it... I didn't know there was a daemon involved. After restarting the _fwupd_ service the error is gone...

Sorry for all the confusion, I didn't expect there to be a daemon.

FYI:

* fwupdmgr v1.3.11 did work with the 2 ESPs without having to use `OverrideESPMountPoint`. It just used the `/boot/efi` mount point.
* after updating to v1.5.11 the 2 ESPs resulted in the mentioned error so I modified `/etc/fwupd/uefi_capsule.conf` but didn't think of restarting the daemon already running in the background. That's when I opened this issue.

My bad 😔

closed 2021-08-11 by c0deright

hughsie, 2021-08-11:
Well.... I think we ought to watch the /etc/fwupd/uefi_capsule.conf config files and restart/reload the daemon as required. Can you open a new issue please, and make the title something like "Watch /etc/fwupd/uefi_capsule.conf for changes and restart the daemon as required" please.