fwupd/fwupd #3631: Support

divyassh, 2021-08-12:
**Description**

Multiple retimers are needed to support TBT/USB4 functionality for longer trace length between the Type-C connector and the SoC mux. This is a new change which will require firmware upgrade for multiple retimers on a single Type-C port.

**Solution:**

fwupd should be able to get the number of retimers and be able to update the firmware for each.
The retimer closest to the Type-C port should be upgraded first. The retimer firmware for the cascaded retimers needs to be assumed as separate binaries.

**Describe alternatives you've considered**

Currently this can be done manually. Steps in details below:
1. Boot up system with port 0 NDA (no device attached)
ls /sys/bus/thunderbolt/devices/0-0/usb4_port1, sysfs entries like "offline" "rescan" should be there.

2. Set up TBT connection
localhost /sys/bus/thunderbolt/devices/0-0/usb4_port1 # echo 1 > ./usb4_port1/offline

3. Scan retimers
localhost /sys/bus/thunderbolt/devices/0-0/usb4_port1 # echo 1 > ./usb4_port1/rescan
After scanning, you should have retimer entries 0-0:1.1 and 0-0:1.2.

localhost /sys/bus/thunderbolt/devices/0-0 # ls usb4_port1
0-0:1.1 0-0:1.2 firmware_node link offline power rescan uevent

4. Copy binary to connector side non-active nvmem and authenticate

localhost /sys/bus/thunderbolt/devices/0-0/usb4_port1 # dd if=<retimer_firmware_facing_connector.bin> of=./0-0:1.2/nvm_non_active1/nvmem

echo 2 > ./0-0:1.2/nvm_authenticate

Copy binary to soc side non-active nvmem and authenticate

localhost /sys/bus/thunderbolt/devices/0-0/usb4_port1 #dd if=<retimer_firmware_facing_soc.bin> of=./0-0:1.1/nvm_non_active0/nvmem

echo 2 > ./0-0:1.1/nvm_authenticate

5. Disconnect TBT connection
localhost /sys/bus/thunderbolt/devices/0-0/usb4_port1 # echo 0 > ./usb4_port1/offline

hughsie, 2021-08-12:
This looks tricky to do in fwupd. We don't know what retimers are attached without a manual scan? Do the retimers give any vid/PID/guid we can use for firmware stream matching?

superm1, 2021-08-12:
Is there not a way to toggle force power on these retimers to make them show up without a connection?

closed 2021-08-15 by superm1