fwupd/fwupd #3633: fwupdmgr won't update if sha1 hex digest in cabinet metainfo is uppercase

fdevibe, 2021-08-13:
**Description**
The checksum comparison is case sensitive, so if the checksum hex digest in the metainfo xml for a cabinet file is written in uppercase, it won't match the digest returned by `g_compute_checksum_for_bytes()` (as this returns a digest in lowercase, see https://docs.gtk.org/glib/func.compute_checksum_for_bytes.html).

**Steps to Reproduce**
`fwupdmgr install N2HET64W.cab` on a Lenovo X1 Carbon, type 20QE, with the cabinet file fetched from https://pcsupport.lenovo.com/no/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-carbon-7th-gen-type-20qd-20qe/downloads/driver-list/component?name=BIOS%2FUEFI

**Expected behavior**
The checksum should match and the BIOS should be updated.

**fwupd version information**
```shell
$ fwupdmgr --version
client version:	1.6.1
compile-time dependency versions
	gusb:	0.3.7

daemon version:	1.6.1
```

fwupd was nstalled on Arch by pacman, but the current master branch also reflects this behavior, see https://github.com/fwupd/fwupd/blob/1c25453aa80ba00694ad868a298b09f319439af9/libfwupdplugin/fu-cabinet.c#L310 (and line 309). `g_strcmp0()` is case sensitive and `g_compute_checksum_for_bytes()` returns lowercase digests.

**Additional questions**

- Operating system and version:
Arch Linux.
- Have you tried rebooting?
Yes.
- Is this a regression?
Not over the last few years, as far as I can tell.

hughsie, 2021-08-13:
The guid needs to be lowercase as per the specification. Archives from the LVFS have been checked and verified and should never have this issue.

fdevibe, 2021-08-13:
@hughsie I see. I was trying to figure out whether the error was on the part of Lenovo, who made these cabinet files, but didn't know where to look. Which specification is this?

superm1, 2021-08-13:
This is the spec here: https://lvfs.readthedocs.io/en/latest/metainfo.html

However to be fair, I don't see anywhere that mentions it needs to be lowercase...  Anything uploaded to LVFS has the metainfo rewritten, so it would be made lower case at that time if it was wrong.