fwupd/fwupd #3635: Dell precision 5510 (XPS 15 9550), system firmware update to 1.17.0 - no supported devices found

doman18, 2021-08-13:
**Describe the bug**
Kubuntu 20.04. All previouse firmware updates worked flawlesly. The latest is seen in update manager (Discover) but when i try to install it, it throws error like 'no supported devices found'

**Steps to Reproduce**
Its just there? No additional steps needed beside having KubuntuLTS@XPS 15 9550

**Expected behavior**
It should just restart laptop and perform update as usuall

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
client version: 1.5.11
compile-time dependency versions
        gusb:   0.3.4

daemon version: 1.5.11

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Apt?

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
Precision 5510
│
├─Dell TB16:
│ │   Device ID:          ed9027c86c336ba5d3312f83c06f78b2ea6ef915
│ │   Summary:            A Thunderbolt™ 3 docking station
│ │   Current version:    0.0.0
│ │   Vendor:             Dell Inc. (PCI:0x1028)
│ │   GUID:               e7ca1f36-bf73-4574-afe6-a4ccacabf479
│ │   Device Flags:       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │ 
│ ├─Dell TB16 Passive Cable:
│ │     Device ID:        29bca9f02e33e23330ae801ae7d6c22eb2b758e0
│ │     Summary:          A Thunderbolt™ 3 docking station
│ │     Current version:  0.3.9
│ │     Vendor:           Dell Inc. (PCI:0x1028)
│ │     GUID:             fece1537-d683-4ea8-b968-154530bb6f73
│ │     Device Flags:     • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │   
│ ├─Dell TB16 Port Controller 1:
│ │     Device ID:        8d1214d30a0630abf1d775a5ad856bf5e2c6a539
│ │     Summary:          A Thunderbolt™ 3 docking station
│ │     Current version:  1.2.17
│ │     Vendor:           Dell Inc. (PCI:0x1028)
│ │     GUID:             8fe183da-c94e-4804-b319-0f1ba5457a69
│ │     Device Flags:     • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │   
│ ├─Dell TB16 Port Controller 2:
│ │     Device ID:        8cd00798d53fae75c23c757b4497e158fe9f8135
│ │     Summary:          A Thunderbolt™ 3 docking station
│ │     Current version:  1.2.18
│ │     Vendor:           Dell Inc. (PCI:0x1028)
│ │     GUID:             1b52c630-86f6-4aee-9f0c-474dc6be49b6
│ │     Device Flags:     • Updatable
│ │                       • System requires external power source
[... 511 lines cut ...]
├── vmlinuz-5.3.0-62-generic
├── vmlinuz-5.4.0-70-generic
├── vmlinuz-5.4.0-77-generic
├── vmlinuz-5.4.0-80-generic
└── vmlinuz.old -> vmlinuz-5.4.0-77-generic

14 directories, 326 files
```

**Additional questions**

- Operating system and version:
- Kubuntu 20.04 

- Have you tried rebooting?
- Yes, many times

- Is this a regression?
- I dont know.

- Are you using an NVMe disk?
Yes. But previous updates went fine.

- Is secure boot enabled?
No.

superm1, 2021-08-13:
>Kubuntu 20.04. All previouse firmware updates worked flawlesly. The latest is seen in update manager (Discover) but when i try to install it, it throws error like 'no supported devices found'

So are you saying that it works properly from the command line tool (fwupdmgr) but fails in KDE Discover only?  

My suspicion here is this is related to the SRU to move from fwupd 1.3.x to 1.5.x in Ubuntu 20.04.  If my suspicion is correct this is actually a downstream bug report that should be opened here: https://launchpad.net/ubuntu/+source/plasma-discover.

superm1, 2021-08-13:
Please try `fwupdmgr update` and see if the same error reproduces.
Is so - giving this more thought it might be https://github.com/fwupd/firmware-dell/issues/69.  I'm going to move it into the dell issues project, and @dell-client-linux will need to investigate further.
If it doesn't, then it's a downstream issue for KDE discover and you should report it downstream.