fwupd/fwupd #3649: flashrom plugin tries to read ME region when attempting to update/downgrade

Sean-StarLabs, 2021-08-17:
The exact error is `failed to back up original firmware` if the CSME is enabled. fwupd is basically doing:
```
flashrom -p internal -w bios.rom -i bios --ifd
```
Which will attempt to read the whole flash, including the ME region (and fail). To make it work, you have to give it the `-N` or `--noverify-all` switch, so it'll only read the region it's actually flashing. i.e.
```
flashrom -p internal -w coreboot.rom -i bios --ifd --noverify-all
```
So question is, what would be the best way to fix it? Changing the backup method to only do the BIOS region would be straightforward but then, I guess, it should only ever write to the BIOS... if there's ever a case where it shouldn't?

superm1, 2021-08-17:
Maybe this is a good case for a private flag in the plugin to control the behavior.

Sean-StarLabs, 2021-08-17:
Flag for flashing something other than the BIOS, right? That'll be 99.9% of use cases

hughsie, 2021-08-18:
What's the libflashrom flag for -N out of interest?

Sean-StarLabs, 2021-08-18:
(--noverify-all) Don't check all regions. Default behaviour is to read them all before and after, to check if they have/haven't changed accordingly.

hughsie, 2021-08-18:
No, the thing we have to set in the code when using libflashrom.h

Sean-StarLabs, 2021-08-18:
I read that very wrong...

FLASHROM_FLAG_VERIFY_WHOLE_CHIP = 0

closed 2021-09-15 by Sean-StarLabs