fwupd/fwupd #3656: lvfs: checksum failure seen intermittently after running lubuntu 20.04 discover update tool

hrmmmph, 2021-08-21:
Condensed output seen intermittently after running discover update tool on lubuntu 20.04.  The discover team said this is a downstream problem with fwupd:

Fwupd Info: saving new firmware metadata to: "/home/tom/.cache/fwupd/remotes.d/lvfs/metadata.xml.gz"
kf5.kio.kio_http: Can't communicate with kded_kcookiejar!
Fwupd Error 0 Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 98261db7124c8026fb88f77768787efc2f8355c3
qml: message: Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 98261db7124c8026fb88f77768787efc2f8355c3

********************************************************************************

Expanded output:

plasma-discover 
(0x7fff61803d40) Warning: Icon theme "elementary" not found.
qrc:/qml/Feedback.qml:2:1: module "org.kde.userfeedback" is not installed
adding empty sources model QStandardItemModel(0x55a7c79c8110)
kdeinit5: preparing to launch '/usr/lib/x86_64-linux-gnu/libexec/kf5/klauncher'
kdeinit5: Launched KLauncher, pid = 1254, result = 0
Connecting to deprecated signal QDBusConnectionInterface::serviceOwnerChanged(QString,QString,QString)
kdeinit5: opened connection to :0
kdeinit5: Got EXEC_NEW '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/http.so' from launcher.
kdeinit5: preparing to launch '/usr/lib/x86_64-linux-gnu/qt5/plugins/kf5/kio/http.so'
org.kde.plasma.libdiscover: Couldn't find a category for  "fwupd-backend"
Fwupd Info: saving new firmware metadata to: "/home/tom/.cache/fwupd/remotes.d/lvfs/metadata.xml.gz"
kf5.kio.kio_http: Can't communicate with kded_kcookiejar!
Fwupd Error 0 Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 98261db7124c8026fb88f77768787efc2f8355c3
qml: message: Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 98261db7124c8026fb88f77768787efc2f8355c3
took really long to fetch KNSBackend(0x55a7c76bedf0, name = "/usr/share/knsrcfiles/k3btheme.knsrc")
took really long to fetch KNSBackend(0x55a7c7756880, name = "/usr/share/knsrcfiles/discover_ktexteditor_codesnippets_core.knsrc")
took really long to fetch PackageKitBackend(0x55a7c7798f90)
(0x7fffd6854a30) Warning: QXcbConnection: XCB error: 3 (BadWindow), sequence: 1061, resource id: 31457302, major code: 18 (ChangeProperty), minor code: 0
kdeinit5: PID 1257 terminated.
Properties constructor called
TRANSLATIONS_DIR: Loading translation file "qterminal_en_US.qm" from dir /usr/share/qterminal/translations
load success: false
Icon theme "elementary" not found.
QDir::exists: Empty or null file name
Canot open file "/home/tom/.config/qterminal.org/qterminal_bookmarks.xml"
Trying to load translation file from dir "/usr/share/Lubuntu"
Trying to load translation file from dir "/usr/local/share"
Trying to load translation file from dir "/usr/share"
Trying to load translation file from dir "/var/lib/snapd/desktop"
Trying to load translation file from dir "/usr/share/qtermwidget5/translations"
default KB_LAYOUT_DIR:  "/usr/share/qtermwidget5/kb-layouts"
loadAllColorSchemes
Shell program: "/bin/bash"
("/bin/bash")
[1893:1893:0813/164521.398440:ERROR:sandbox_linux.cc(374)] InitializeSandbox() called with multiple threads in process gpu-process.
dbus-daemon[1040]: Activating service name='org.a11y.atspi.Registry' requested by ':1.12' (uid=1000 pid=1848 comm="/opt/google/chrome/chrome --incognito --enable-cra" label="unconfined")
dbus-daemon[1040]: Successfully activated service 'org.a11y.atspi.Registry'
SpiRegistry daemon is running with well-known name - org.a11y.atspi.Registry

**************************************************************************

I'm wondering what the security ramifications of these messages are.

****************************************************************************8

hughsie, 2021-08-28:
You didn't provide any of the information we requested in the issue template.

hrmmmph, 2021-08-29:
I update lubuntu 20.04 and related software every time I boot up (daily) and am still seeing this issue.  I have been using the "discover" gui to do the updates.  In other words, the issue is still occurring in all the latest software revisions.  If you need other specific information, let me know what this is.  I'm not able determine how to get to an issue template from here.

superm1, 2021-09-02:
Follow https://github.com/fwupd/fwupd/issues/new?assignees=&labels=bug&template=bug-report-general.md&title=

superm1, 2022-01-14:
duplicate of #3652

closed 2022-01-14 by superm1