fwupd/fwupd #3657: build failure: fu-hash.h: No such file or directory

dkwo, 2021-08-22:
If I try to build fwupd 1.6.3 on void linux (x86_64-musl), it fails complaining that `fatal error: fu-hash.h: No such file or directory`.
I attach the full log.
[f.log](https://github.com/fwupd/fwupd/files/7027912/f.log)

dkwo, 2021-08-22:
this is the template I'm using to build it https://github.com/void-linux/void-packages/pull/32636

superm1, 2021-08-22:
FYI, your packaging needs updating for 1.6.x release.
```
WARNING: Unknown options: "efi-includedir, efi-ld, efi-ldsdir, efi-libdir, efi-objcopy, gtkdoc, pkcs7"
```
*   the efi-* options are now used in the subproject `fwupd-efi` (which may be packaged separately as well). 
*  Docs are in a new option for replacing gtkdoc
* pkcs7 isn't a configure option anymore

Regarding the failure it seems to me like it's likely trying to run a parallel build and `fu_hash` even thought it's a dependency for the tpm-eventlog plugin (among others) isn't getting run first.  Can you share the exact command that gets executed to kick off the configure and build?  This hasn't reproduced in CI at all (including on void!).  It doesn't seem to be in your full log.

superm1, 2021-08-22:
For me, trying to explicitly run it on the tpm-eventlog plugin it doesn't repro:
```
rm -rf build; meson build; ninja -C build plugins/tpm-eventlog/libfu_plugin_tpm_eventlog.so
```

superm1, 2021-08-22:
I came up with a way to reproduce and a fix, you can try to apply that to your build.

closed 2021-08-22 by superm1

dkwo, 2021-08-23:
Thanks, this solves my problem.

commit 3b4b0556 (2021-08-22): trivial: fix a compilation dependency error (Fixes: #3657)

Reproducible by manually calling failing build targets first
```
ninja -C build plugins/tpm-eventlog/fwupdtpmevlog.p/fu-tpm-eventlog-common.c.o
ninja: Entering directory `build'
[1/1] Compiling C object plugins/tpm-eventlog/fwupdtpmevlog.p/fu-tpm-eventlog-common.c.o
FAILED: plugins/tpm-eventlog/fwupdtpmevlog.p/fu-tpm-eventlog-common.c.o
cc -Iplugins/tpm-eventlog/fwupdtpmevlog.p -Iplugins/tpm-eventlog -I../plugins/tpm-eventlog -I. -I.. -Ilibfwupd -I../libfwupd -Ilibfwupdplugin -I../libfwupdplugin -I/usr/include/libxmlb-1 -I/usr/include [... cut]
commit 2d4aecf2 (2021-08-23): trivial: fix a compilation dependency error (Fixes: #3657)

Reproducible by manually calling failing build targets first
```
ninja -C build plugins/tpm-eventlog/fwupdtpmevlog.p/fu-tpm-eventlog-common.c.o
ninja: Entering directory `build'
[1/1] Compiling C object plugins/tpm-eventlog/fwupdtpmevlog.p/fu-tpm-eventlog-common.c.o
FAILED: plugins/tpm-eventlog/fwupdtpmevlog.p/fu-tpm-eventlog-common.c.o
cc -Iplugins/tpm-eventlog/fwupdtpmevlog.p -Iplugins/tpm-eventlog -I../plugins/tpm-eventlog -I. -I.. -Ilibfwupd -I../libfwupd -Ilibfwupdplugin -I../libfwupdplugin -I/usr/include/libxmlb-1 -I/usr/include [... cut]