fwupd/fwupd #3679: Can't update firmware : fwupdx64.efi.signed missing

gbdomubpkm, 2021-08-28:
Device 

LENOVO ThinkPad X280 with Ubuntu 20.04.3 LTS installed

[image]

Hello
First i'm not a developer, only a user with a basic knowledge, so forgive me if i'm not precise enough..

Two weeks ago i think, i received these update in SNAP store and i tried to upgrade firmware with snap store but not possible to upgrade.
Then by going on forums i tried the command line   **sudo fwupdmgr update**   if i remember correctly : not possible too because of the signed file missing. I also had to send a bug report (is it to Ubuntu or Lenovo ?) at the end of the unsuccessful command.
Still unable to update to date. Should I wait for the snap store to fix the problem?
I would like to point out this problem if you can solve it (easily) or if you can pass on the information to the right hands.

With kind regards

superm1, 2021-08-31:
It sounds like you're missing the package `fwupd-signed` on your system.  Can you install it?

gbdomubpkm, 2021-08-31:
Yes indeed, i'm missing this signed package (issue title).
How to install this signed package ? I don't know how to do that.
Is this the right approach? If so, do you have a complete procedure? Thank you in advance.

superm1, 2021-08-31:
`apt install fwupd-signed`

closed 2021-08-31 by superm1