fwupd/fwupd #3680: Version comparison error: "Specified firmware is older than installed '0x15449bc < 22301115'"

leogama, 2021-08-28:
**Describe the bug**
This is similar to Similar to #3217. `fwupdmgr` compares the firmware update version (22301116) to the installed version (22301115), concludes that the update is older than the current firmware and refuses to install it. The informative message (both in GNOME Software and in `fwupdmgr`) says that 22301115 would be upgraded to 22301116, showing both in decimal form. But the error message shows just the higher version number in hexadecimal form and the other in decimal.

GNOME Software capture:
[image]

Command-line outputs:
```
$ fwupdmgr get-updates
Devices with no available firmware updates: 
 • DELL08C8:00 04F3:30C3
 • System Firmware
 • UEFI dbx
 • WDC WD10SPZX-21Z10T0
Vostro 3480
│
└─CL1-3D128-Q11 NVMe SSSTC 128GB:
  │   Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
  │   Summary:            NVM Express Solid State Drive
  │   Current version:    22301115
  │   Vendor:             Intel Corporation (NVME:0x1E95)
  │   GUIDs:              c37ff45c-9789-5d14-a8a6-cb1d664cca9c
  │                       1cd7457a-27a9-4af9-912b-73af953d94b4
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─NVM Express Solid State Drive for CL1 Device Update:
        New version:      22301116
        Remote ID:        lvfs
        Summary:          Non SED NVME
        License:          Proprietary
        Size:             1,1 MB
        Created:          2021-01-14
        Urgency:          High
        Vendor:           SSSTC
        Flags:            is-upgrade
        Description:      
        Fixed the issue where the system does not reboot after enabling verifier mode.
        
        Improved the SSD performance and enhanced the battery life by reducing the power consumption.
```

```
$ fwupdmgr update
Devices with no available firmware updates: 
 • DELL08C8:00 04F3:30C3
 • System Firmware
 • UEFI dbx
 • WDC WD10SPZX-21Z10T0
Upgrade available for CL1-3D128-Q11 NVMe SSSTC 128GB from 22301115 to 22301116
Vostro 3480 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
Downloading…             [***************************************]
Decompressing…           [***************************************]
Specified firmware is older than installed '0x15449bc < 22301115'
```

I suspect one of the numbers is in the wrong format internally, and/or is being silently coerced to an inappropriate type instead of raising an error.

**Steps to Reproduce**
Get a Dell notebook with an Intel NVMe SSD, wait for a firmware update for it, try to install the update either via GNOME Software or via `fwupdmgr update` command.

**Expected behavior**
`fwupd` could either recognize that `22301116` and `0x15449bc` are numbers and the same number, or identify the problem and show a meaningful error message. Note: I don't know how is the firmware version specification or the software implementation, so don't know if it's possible.

**fwupd version information**
Please provide the version of the daemon and client.
[... 114 lines cut ...]
│   
└─WDC WD10SPZX-21Z10T0:
      Device ID:          5dbeb140337f610d54913c0f43d15fdaa0eafb51
      Summary:            ATA Drive
      Current version:    02.01A02
      Vendor:             Western Digital (ATA:0x101C, OUI:0014ee)
      GUIDs:              fa8b4ca0-786c-54ea-9bd6-d025389b0590
                          807552eb-11ba-5e03-8894-84dad0e329a6
                          18d6d543-e351-5d90-a1c4-f0d3429463a2
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

**Additional questions**

- Operating system and version: Ubuntu Linux 20.04 (up-to-date)
- Have you tried rebooting? No.
- Is this a regression? I think it was the first firmware update attempt in this machine:
```
$ fwupdmgr get-history                   
No history
```

hughsie, 2021-08-30:
Does using fwupd 1.6.x help here? If so, we should add that requirement on the firmware.

leogama, 2021-08-30:
Installed the latest version (edge channel, the stable version crashed) available via Snap but... the update disappeared. I think they rolled it back.

leogama, 2021-08-31:
Is the previously downloaded cab(?) file somewhere in my system, or is it thrown away and then redownloaded in the next try? I can find files in `/var/cache/fwupd`, `/var/cache/fwupdmgr` and `/var/lib/fwupd`, most metadata, but no cab file.

hughsie, 2021-08-31:
It's not cached in most cases unfortunately -- otherwise we'd have a multimegabyte cache folder just growing in size and then have to define a cache eviction policy, and tests, etc. Try `fwupdmgr refresh --force` then `fwupdmgr get-updates`

leogama, 2021-08-31:
I did that yesterday and no update showed up. Today it appeared again, but:
```
fwupdmgr update 
cp: cannot stat '/snap/fwupd/3358/var': No such file or directory
WARNING: This package has not been validated, it may not work properly.
Devices with no available firmware updates: 
 • DELL08C8:00 04F3:30C3
 • System Firmware
 • UEFI dbx
 • WDC WD10SPZX-21Z10T0
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade CL1-3D128-Q11 NVMe SSSTC 128GB from 22301115 to 22301116?            ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Fixed the issue where the system does not reboot after enabling verifier     ║
║ mode.                                                                        ║
║ Improved the SSD performance and enhanced the battery life by reducing the   ║
║ power consumption.                                                           ║
║ Vostro 3480 must remain plugged into a power source for the duration of the  ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
Downloading…             [***************************************]
Decompressing…           [***************************************]
Specified firmware is older than installed '0x15449bc < 22301115'

```

leogama, 2021-08-31:
Here is the metadata from the offending CAB file (https://fwupd.org/downloads/45eea86ce2a61581d3ebb621a4f2c9c6717b193355e23b5dc55b13c6b5434625-22301116.cab):
```xml
<?xml version="1.0" encoding="UTF-8"?>
<component type="firmware">
  <id>com.dell.guid1cd7457a.firmware</id>
  <name>NVM Express Solid State Drive for CL1</name>
  <summary>Non SED NVME</summary>
  <description>
    <p>improves performance and stability.</p>
  </description>
  <provides>
    <firmware type="flashed">1cd7457a-27a9-4af9-912b-73af953d94b4</firmware>
  </provides>
  <url type="homepage">http://support.dell.com</url>
  <metadata_license>CC0-1.0</metadata_license>
  <project_license>LicenseRef-proprietary</project_license>
  <developer_name>SSSTC</developer_name>
  <categories>
    <category>X-Device</category>
  </categories>
  <custom>
    <value key="LVFS::VersionFormat">number</value>
    <value key="LVFS::UpdateProtocol">org.nvmexpress</value>
  </custom>
  <releases>
    <release version="0x15449bc" date="2021-01-14" urgency="high">
      <checksum type="sha1" filename="22301116.bin" target="content">02943e29b4aef04eb2fc23d9a930285bf51016f9</checksum>
      <checksum type="sha256" filename="22301116.bin" target="content">70b7110b6a14876d4b46004f7c81a178e2d95efb6676a58254fc2d70835141cf</checksum>
      <description>
        <p>Fixed the issue where the system does not reboot after enabling verifier mode.</p>
        <p>Improved the SSD performance and enhanced the battery life by reducing the power consumption.</p>
      </description>
      <size type="installed">1130496</size>
      <artifacts>
        <artifact type="binary">
          <filename>22301116.bin</filename>
          <checksum type="sha1">02943e29b4aef04eb2fc23d9a930285bf51016f9</checksum>
          <checksum type="sha256">70b7110b6a14876d4b46004f7c81a178e2d95efb6676a58254fc2d70835141cf</checksum>
        </artifact>
      </artifacts>
    </release>
  </releases>
</component>
```
Could the release version be in hexadecimal form? It seems that `fwupd` did convert it to decimal at some point to show the informative message `Upgrade CL1-3D128-Q11 NVMe SSSTC 128GB from 22301115 to 22301116?`.

Oh, and I'm using the Snap `edge` version now —ironically, the `stable` version didn't work.
```
$ fwupdmgr --version             
cp: cannot stat '/snap/fwupd/3358/var': No such file or directory
client version: 1.6.2-81-g566f8e27
compile-time dependency versions
        gusb:   0.3.7

daemon version: 1.6.2-81-g566f8e27
```

superm1, 2021-08-31:
To me this looks like a firmware CAB metainfo error.  The CAB has the key:
```
    <value key="LVFS::VersionFormat">number</value>
```
But the version is formatted as hex according to https://fwupd.org/lvfs/docs/metainfo/version

@dell-client-linux I think this update should be pulled and that version format key being fixed to hex or version switched to "22301116" instead of "0x15449bc".