fwupd/fwupd #3681: Potentially incorrect information being sent to newer wacom tablets

Davidy22, 2021-08-29:
**Describe the bug**
There is a [known bug](https://github.com/linuxwacom/input-wacom/issues/174) [in linux wacom support](https://github.com/linuxwacom/input-wacom/issues/266) that occurs with newer tablets where they connect in android mode when plugged in. Curiously, this bug only occurs when plugging the device in, and not when the system starts up. The current accepted solution is to [work around it](https://github.com/linuxwacom/libwacom/wiki/Troubleshooting#special-cases), but this isn't ideal for the creative types that buy these tablets.

Recently I've found that

```
sudo systemctl stop fwupd.service
sudo systemctl disable fwupd.service
sudo systemctl mask fwupd.service
```

Makes plugging in the tablet detect the computers correctly as a desktop machine, and not as android. I'm not entirely familiar with fwupd, but I gather from descriptions of how the bug happens that it's due to something being sent to the tablet that makes it incorrectly determine the computer as an android device. Which I suspect is coming from fwupd given that disabling fwupd seems to make the issue stop entirely.

**Steps to Reproduce**
Plug a wacom bamboo 4100 into your computer and try to use it. Other affected models [listed here](https://github.com/linuxwacom/libwacom/wiki/Troubleshooting#special-cases)

**Expected behavior**
Wacom tablet uses the full drawing region.

**fwupd version information**

```shell
fwupdmgr --version
```
client version:	1.5.11
compile-time dependency versions
	gusb:	0.3.4

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): I don't consciously remember installing it, so it came with the distro or was automatically installed as a dependency to something else probably.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```
WARNING: UEFI capsule updates not available or enabled in firmware setup
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
HP ENVY 15 Notebook PC
│
├─4th Gen Core Processor Integrated Graphics Controller:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    06
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              8d8280eb-f57e-553a-b6b5-7cbf9de992ea
│                         29954491-cfbf-560f-a80a-a787ba945f88
│                         5570d9e8-f98d-54d6-977f-d4a02487bb96
│                         630813c1-8104-5749-a082-1de4ff3f1497
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─GK107M [GeForce GT 750M]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a1
│     Vendor:             NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:              2b1b4077-3e1f-5556-8afe-102af0650aa4
│                         2eaf8108-0e97-52b6-a649-09715b329629
│                         1e7d7574-d5fc-581c-bd2c-0100f6c6686e
│                         8dd444ee-b473-5617-ac88-056a91262038
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Intel(R) Core™ i7-4702MQ CPU @ 2.20GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000028
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         dd2cdaeb-2b08-599d-aaec-5a8374add8d6
│                         112c1ba3-2c29-5ab2-bd72-ea65bd10dd67
│     Device Flags:       • Internal device
│   
├─Intuos S:
│ │   Device ID:          983c3cffc6fd36d32b00b62928d30721eaeb93db
│ │   Current version:    1.11
│ │   Vendor:             Wacom Co.,Ltd. (USB:0x056A)
│ │   Install Duration:   10 seconds
│ │   GUIDs:              8b156275-075c-5806-bc3a-9e8ac6ea6e6c
│ │                       facf5428-3e37-58f5-9283-10c1e512f943
│ │   Device Flags:       • Updatable
│ │ 
│ └─Intuos S [Legacy Bluetooth Module]:
│       Device ID:        aee3401aa6df389d741b322e18a554ecfeada4bb
│       Current version:  0.0
│       Vendor:           Wacom Co.,Ltd. (USB:0x056A)
│       Install Duration: 30 seconds
│       GUID:             2ea38282-7c4c-59b3-b8ab-9a30a5a215e3
│       Device Flags:     • Updatable
│     
├─TOSHIBA MQ01ABD100:
│     Device ID:          602b0a6cc821d155208724f0e22f8d111542b74c
│     Summary:            ATA Drive
│     Current version:    AX001C
│     Vendor:             Toshiba (ATA:0x1179, OUI:000039)
│     GUIDs:              295a7d5e-1a3c-564a-ae09-82002a244da5
│                         17628727-4832-5b2c-af2b-f408e0d99ff7
│                         539bd28f-704f-5906-81b1-08eef8cdd97f
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI dbx:
      Device ID:          362301da643102b9f38477387e2193e57abaa590
      Summary:            UEFI Revocation Database
      Current version:    78
      Minimum Version:    78
      Vendor:             UEFI:Linux Foundation
      Install Duration:   1 second
      GUIDs:              c6682ade-b5ec-57c4-b687-676351208742
                          f8ba2887-9411-5c36-9cee-88995bb39731
                          6c34d1d9-4282-5127-8a4f-86049022a915
                          8f327278-3121-5d0a-b083-7cac3cd17313
      Device Flags:       • Internal device
                          • Updatable
                          • Needs a reboot after installation

**Additional questions**

- Operating system and version: Linux mint 20.2
- Have you tried rebooting? Yes
- Is this a regression? I haven't used a version where this doesn't occur.

hughsie, 2021-08-29:
Iirc this was fixed in 1.6.3

Davidy22, 2021-08-30:
I went and installed by snap because my distro's version is very behind. I notice that the latest snaps are 1.6.2, although I wasn't able to test that version either because trying to
```
sudo systemctl enable fwupd.service
```
Gives a unit file does not exist error. Not sure how to resolve this, took a quick venture at compiling but as the readme promised, there's some hairy dependency stuff going on. In short, can't quite check that it does work yet, but I can update on this when 1.6.3 comes down the pipe to me eventually.

superm1, 2021-09-14:
Yeah 1.6.3 didn't publish since we had the job turned off when we branched for 1.7.x.  If we have a 1.6.4 before 1.7.x can try to get it published.

>sudo systemctl enable fwupd.service

With the snap it's `snap.fwupd.fwupd.service`

>Not sure how to resolve this, took a quick venture at compiling but as the readme promised, there's some hairy dependency stuff going on. 

Would be interested in your experience here.  With the stuff in master running `./contrib/setup` will install all the dependencies for you on the big distros.  I notice we haven't updated the wiki page for this though, so i'll do that.

Davidy22, 2021-09-16:
Alright, went and ran setup and `meson build` got stopped by gudev not found. I can wait, the bug is solved for me by just uninstalling fwupd. I'll come back when a built version of fwupd with the fix comes down the pipeline so I can test and report back to the wacom driver devs.

superm1, 2021-09-16:
Can you run ./contrib/setup ?

Davidy22, 2021-09-16:
Yeah, running setup installed some things, but apparently gudev wasn't one of those things. I ran it again and it didn't install anything extra but it would have been weird if there were things that only installed on the second run anyways. meson build still fails at the same place, gudev isn't in my package manager, will just wait for when you have 1.7 or 1.6.4 built.

superm1, 2021-09-16:
I wonder if this is because you're on mint it's detecting the wrong OS to resolve dependencies.

Can up share your /etc/lsb-release? Maybe we can tell them if it's using the wrong keys for dependencies on mint.

Davidy22, 2021-09-16:
yeah sure
\
DISTRIB_ID=LinuxMint
DISTRIB_RELEASE=20.2
DISTRIB_CODENAME=uma
DISTRIB_DESCRIPTION="Linux Mint 20.2 Uma"

superm1, 2021-09-16:
OK, I wrote up https://github.com/fwupd/fwupd/pull/3756 which I think should help improve the situation for you.

superm1, 2021-09-16:
That stuff was merged, so if you update to master and try `contrib/setup` again hopefully it installs a lot more for you.

Davidy22, 2021-09-16:
Did a little git pull and reran setup, nothing new got installed. Ran meson build anyway just in case the setup output was lying, it wasn't

superm1, 2021-09-16:
Can you share output from running `./contrib/ci/generate_dependencies.py`?

Let's make sure it's falling back right

Davidy22, 2021-09-16:
05:42:47❯ ./contrib/ci/generate_dependencies.py
abigail-tools
bash-completion
bubblewrap
ca-certificates
clang
clang-tools
clang-format
libcairo-dev
libcairo-gobject2
libjson-glib-dev
libftdi1-dev
libpci-dev
fonts-noto
devscripts
debhelper
dbus-x11
dh-strip-nondeterminism
libelf-dev
fakeroot
libfreetype6-dev
fontconfig
libflashrom-dev
gcab
gettext
gnu-efi
git
libglib2.0-doc
libglib2.0-dev
gobject-introspection
gnome-desktop-testing
gnutls-dev
gnutls-bin
gtk-doc-tools
libxmlb-dev
libjcat-dev
libarchive-dev
libefivar-dev
libefiboot-dev
libgcab-dev
libgirepository1.0-dev
libgudev-1.0-dev
libgusb-dev
libsmbios-dev
libsoup2.4-dev
libcurl4-gnutls-dev
libtool-bin
libumockdev-dev
lintian
locales
meson
mingw-w64-tools
gir1.2-pango-1.0
pkg-config
policykit-1
libmm-glib-dev
libqmi-glib-dev
libmbim-glib-dev
libpolkit-gobject-1-dev
python3-gi-cairo
python3-pygments
python3-typogrify
python3-toml
python3-markdown
python3-jinja2
python3-requests
libsqlite3-dev
systemd
libsystemd-dev
shared-mime-info
texlive-fonts-recommended
udev
umockdev
valac
valgrind
libtss2-dev
shellcheck
libprotobuf-c-dev
protobuf-c-compiler
sudo

superm1, 2021-09-17:
OK great, the fallback code is working now for linuxmint, but there is another problem gluing it to contrib/setup.  I'll do another PR.

superm1, 2021-09-17:
OK that PR is merged, can you try `./contrib/setup` one more time?  If it's not working still, please include `./contrib/setup --debug` output (which is new!)

Davidy22, 2021-09-17:
Meson build still stuck on the same step. Debug: https://gist.github.com/Davidy22/d2290590c6e69965244adce14a09bdf4

superm1, 2021-09-20:
Ah ha, you have something wrong with the packages on your system.  So the helper script isn't able to recover.  That's why not everything is getting installed.

```
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 clang : Depends: clang-10 (>= 10~) but it is not going to be installed
 clang-tools : Depends: clang-tools-10 (>= 10~) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

You'll probably want to run `apt -f install` and similar tools to figure out what is wrong with your packages (unless you intentionally caused that).

Davidy22, 2021-09-20:
Eh it's some lengthy dependency conflict probably, I'm just going to wait for the prebuilt.

superm1, 2021-09-20:
Well the thing you're missing to build is at a minimum `libgudev-1.0-dev`.  You can try to install that to proceed for now.

Davidy22, 2021-09-20:
yeah there's a chain of deps after that, got to gpgme that I couldn't immediately guess the name in my repo

superm1, 2021-09-21:
It seems like `apt policy clang-tools-10` would be a good starting point to fix whatever you've got wrong.  Maybe you picked up a PPA at some point that has some cruft that is bad packaging.

superm1, 2021-09-25:
FWIW the 1.6.4 snap is in the snap store now.

Davidy22, 2021-09-26:
Ran 

```
sudo systemctl enable snap.fwupd.fwupd.service
sudo systemctl start snap.fwupd.fwupd.service
```

And plugging the tablet in seems to not do the thing anymore, although given that this bug is also solved by just uninstalling fwupd, accidentally not enabling fwupd fully would be an incorrect test. Checked systemctl list-units and I see snap.fwupd.fwupd.service enabled and snap-fwupd-3443.mount, so I think I have it installed and running? Think this is resolved now, closing.

closed 2021-09-26 by Davidy22

superm1, 2021-09-26:
You could confirm it's running by a `fwupdmgr get-devices` call and that would confirm it's working too.

Davidy22, 2021-09-27:
I get output from it, so looks like it's installed and running