fwupd/fwupd #3685: grub config file prevents removal of kernel

Sean-StarLabs, 2021-08-31:
The `/etc/grub.d/35_fwupd` config file seems to prevent removing kernels on Debian based distro's, based on building master from 7ae233b7cfb8ccf964e131f5dde2f1073374cf8e.

```
/etc/grub.d/35_fwupd: line 5: /usr/share/grub: Is a directory
run-parts: /etc/kernel/postrm.d/zz-update-grub exited with return code 126
dpkg: error processing package linux-image-5.11.0-27-generic (--remove):
installed linux-image-5.11.0-27-generic package post-removal script subprocess returned error exit status 1
dpkg: too many errors, stopping
```

superm1, 2021-08-31:
Can you try reverting 684bc038132977ff28e54550f95941cff5e4ee84 to see if it helps?

Sean-StarLabs, 2021-09-01:
Unfortunately, it doesn't make a difference

superm1, 2021-09-01:
Can you add `set -x` into the file?

Sean-StarLabs, 2021-09-01:
That does the trick

superm1, 2021-09-01:
It's supposed to add more debugging not fix it.....

Sean-StarLabs, 2021-09-01:
Okay, ignore me - I'd removed the file to get past it, assumed updating would put it back.

Take 2...

Sean-StarLabs, 2021-09-01:
```
sudo update-grub
Sourcing file `/etc/default/grub'
Sourcing file `/etc/default/grub.d/init-select.cfg'
Generating grub configuration file ...
Found background image: wallpaper.png
Found linux image: /boot/vmlinuz-5.11.0-31-generic
Found initrd image: /boot/initrd.img-5.11.0-31-generic
Adding boot menu entry for UEFI Firmware Settings
+ /usr/share/grub
/etc/grub.d/35_fwupd: line 6: /usr/share/grub: Is a directory
```

Line 6 is:
```
${pkgdatadir:?}
```

If I just comment that out, it runs happily:
```
Sourcing file `/etc/default/grub'
Sourcing file `/etc/default/grub.d/init-select.cfg'
Generating grub configuration file ...
Found background image: wallpaper.png
Found linux image: /boot/vmlinuz-5.11.0-31-generic
Found initrd image: /boot/initrd.img-5.11.0-31-generic
Adding boot menu entry for UEFI Firmware Settings
+ . /usr/share/grub/grub-mkconfig_lib
++ prefix=/usr
++ exec_prefix=/usr
++ datarootdir=/usr/share
++ datadir=/usr/share
++ bindir=/usr/bin
++ sbindir=/usr/sbin
++ '[' x/usr/share/grub = x ']'
++ test x = x
++ grub_probe=/usr/sbin/grub-probe
++ test x = x
++ grub_file=/usr/bin/grub-file
++ test x = x
++ grub_mkrelpath=/usr/bin/grub-mkrelpath
++ which gettext
++ :
++ grub_tab='	'
+ '[' -f /var/lib/fwupd/uefi_capsule.conf ']'
done
```

superm1, 2021-09-01:
can you change that line to:
`[ -d ${pkgdatadir:?} ]`

I think that should fix it.

Sean-StarLabs, 2021-09-01:
Yup, that works

closed 2021-09-01 by superm1

Rhahkeem, 2021-09-12:
Is there any way to add this fix for focal ?

superm1, 2021-09-14:
Fwupd 1.7.x is not released.  So you must have a locally built package.  Just apply the patch to your package.

PR #3693 (2021-09-01): trivial: fix the grub configuration from 684bc03
fixes #3685

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 92c8f0e7 (2021-09-01): trivial: fix the grub configuration from 684bc03
fixes #3685