fwupd/fwupd #3690: Add the capability to reset cmos if layout has changed in coreboot

Sean-StarLabs, 2021-09-01:
This will reset the CMOS (at least in our coreboot port).

Not really sure on the how or the where to put this, I was thinking in flashrom based of a quirk in the metadata.

Any suggestions?

```
#include <stdio.h>
#include <stdbool.h>
#include <stdint.h>
#include <unistd.h>
#include <sys/io.h>

/* From coreboot's src/include/pc80/mc146818rtc.h file */
#define RTC_BASE_PORT 0x70

/* 
 * This is the offset of the first of the two checksum bytes
 * we may want to figure out how we can determine this dynamically
 * during execution.
 */
#define CMOS_CHECKSUM_OFFSET 123

void cmos_checksum_write(uint16_t checksum);
void cmos_write_byte(uint8_t addr, uint8_t val);

int main(void)
{
	/* Call ioperm() to grant us access to ports 0x70 and 0x71 */
	ioperm(RTC_BASE_PORT, 2, true);

	/* Write a default value to the CMOS checksum */
	cmos_checksum_write(0xffff);
}

void cmos_checksum_write(uint16_t checksum)
{
	unsigned char lo, hi;

	/* The checksum is stored in a big-endian format. */
	hi = (unsigned char)(checksum >> 8);
	lo = (unsigned char)(checksum & 0x00ff);
	cmos_write_byte(CMOS_CHECKSUM_OFFSET, hi);
	cmos_write_byte(CMOS_CHECKSUM_OFFSET + 1, lo);
}

void cmos_write_byte(uint8_t addr, uint8_t val)
{
	/* Reject addresses in the second bank */
	if (addr >= 128)
		return;

	outb(addr, RTC_BASE_PORT);
	outb(val, RTC_BASE_PORT + 1);
}
```

hughsie, 2021-09-02:
I don't mind if it goes into the flashrom plugin, but it certainly does need a quirk. It seems a shame we can't do this in the firmware payload on first boot.

superm1, 2021-09-15:
fixed in:
f959b198d223c95f1983256ee2e8e801eb1ac052
8cc158bda42e358ce6e398ab46cac47d1e1ab919
af9d9b3ffd299eef0478f7c68a74aad7eda17763

closed 2021-09-15 by superm1