fwupd/fwupd #3697: fwupdmgr refresh: signatures were required, but none supplied

andyhhp, 2021-09-02:
**Describe the bug**
Attempting to refresh metadata fails due to signatures

**Steps to Reproduce**

```
$ fwupd.fwupdmgr refresh
WARNING: This package has not been validated, it may not work properly.
Updating lvfs
Downloading…             [***************************************]
Failed to update metadata for lvfs: signatures were required, but none supplied
```

Using `-v` no useful extra information was provided other than the CDN path (https://cdn.fwupd.org/downloads/) which offer these two files:
```
$ shasum *
55291ea01c0c3611788bc298212a7812df398c56  firmware.xml.gz
8e937681e3a6c8025c25acd561bb318aebfa6eca  firmware.xml.gz.asc
```

and GPG does appear to be somewhat unhappy with the result
```
$ gpg --verify firmware.xml.gz.asc 
gpg: assuming signed data in 'firmware.xml.gz'
gpg: Signature made Thu 02 Sep 2021 08:17:37 BST
gpg:                using RSA key 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
gpg: Can't check signature: No public key
```

**Expected behaviour**
No signature problems.  At a minimum the error message is misleading, claiming that no signature was provided, when in fact one was.

**fwupd version information**

```shell
 fwupd.fwupdmgr --version
client version:	1.6.2
compile-time dependency versions
	gusb:	0.3.7

daemon version:	1.6.2
```

This is Ubuntu 20.04 using the snap because the non-snap version (1.5.11) is too old to update my main system firmware.

**Additional questions**

- Is this a regression?
It has worked in the past, so something has changed, but I can't pinpoint it more than that.

andyhhp, 2021-09-02:
There is no `fwupd.fwupd`.  Available commands are:
```
$ fwupd.
fwupd.dbxtool        fwupd.dfu-tool       fwupd.fwupdagent     fwupd.fwupdmgr       fwupd.fwupdtool      fwupd.fwupdtpmevlog  
```

superm1, 2021-09-02:
I think that's a good idea to take a look at them, sure.

superm1, 2021-09-02:
> I suspect the snap is broken -- maybe `fwupd.fwupd --verbose` would give a clue?

1.6.3 didn't promote since we disabled the job when branching to 1.7.x, but 1.6.2 should be OK.  There might be a real bug here, let's look at the files.

andyhhp, 2021-09-02:
[files.tar.gz](https://github.com/fwupd/fwupd/files/7101981/files.tar.gz)
Here they are

superm1, 2021-09-03:
thanks.  @hughsie something peculiar on LVFS signing side it seems to me.  The data looks good and valid and decompresses fine.  Do we have a timing problem again?

```
$ gpg --verify bad/firmware.xml.gz.asc 
gpg: assuming signed data in 'bad/firmware.xml.gz'
gpg: Signature made Thu 02 Sep 2021 02:17:37 AM CDT
gpg:                using RSA key 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
gpg: BAD signature from "Linux Vendor Firmware Service <sign@fwupd.org>" [unknown]
$ gpg --verify good/firmware.xml.gz.asc 
gpg: assuming signed data in 'good/firmware.xml.gz'
gpg: Signature made Thu 02 Sep 2021 02:44:18 AM CDT
gpg:                using RSA key 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
gpg: Good signature from "Linux Vendor Firmware Service <sign@fwupd.org>" [unknown]
gpg: WARNING: This key is not certified with a trusted signature!
gpg:          There is no indication that the signature belongs to the owner.
Primary key fingerprint: 3FC6 B804 410E D084 0D8F  2F97 48A6 D80E 4538 BAC2
```

andyhhp, 2021-09-03:
Thanks for the help so far.  I have a few observations.

It has been surprisingly hard to find your GPG key.  https://fwupd.org/ has nothing on it, and while https://lvfs.readthedocs.io/en/latest/search.html?q=gpg+key&check_keywords=yes&area=default makes it clear that there are GPG keys in use, it doesn't provide them.  In the end I gave up and tried https://github.com/fwupd/fwupd/search?q=gpg+key which did reveal them, even if only via the build scripts.  IMO, one of the pieces of official documentation ought to identify all the keys currently in use, and what they sign.

The error message `signatures were required, but none supplied` presented to the end user is a long way away from `BAD signature` which appears to be the root of the issue.  Not even with `fwupd.fwupdmgr -v refresh` did I get given any hint that the signature did exist and had been rejected.

Are the snaps going to get updated in due course?  From your comments thus far, latest/stable isn't 1.6.2 any more.

hughsie, 2021-09-03:
> It has been surprisingly hard to find your GPG key

In Fedora they're installed in /etc/pki/fwupd-metadata/

hughsie, 2021-09-03:
So my guess here was that fwupd was compiled against (i.e. build time, not run time) a newer version of libjcat than is being shipped. I'm pretty sure updating to libjcat 0.1.4 (released 2020-10-23) will fix this, although, you should really update to 0.1.8 for other fixes too.

superm1, 2021-09-03:
> So my guess here was that fwupd was compiled against (i.e. build time, not run time) a newer version of libjcat than is being shipped. I'm pretty sure updating to libjcat 0.1.4 (released 2020-10-23) will fix this, although, you should really update to 0.1.8 for other fixes too.

At least for snap that seems quite unlikely snap downloads using subproject which as of d9f4b89f1b9b47460536bfc33daf2b90f359d2d8 should be 0.1.8.

>In Fedora they're installed in /etc/pki/fwupd-metadata/

And in the source tree you can find them in `data/pki`.  @hughsie any reason not to upload to a keyserver?  It might make them more accessible.

hughsie, 2021-09-03:
> any reason not to upload to a keyserver

No reason at all; does that have to be me or can anybody do it?

closed 2022-01-21 by hughsie