fwupd/fwupd #3701: Dell WD19: fwupd uses synaptics_mst plugin to probe Dell WD19's VMM5331 mst device

CragW, 2021-09-03:
**Describe the bug**
I'm seeing a reported issue that has error prints `failed to set quirk key` from the console output, it seems `fwupd` wrongly used `synaptics_mst` plugin to probe dock's `VMM5331` device. 

I was told Issue is reproducible on Chrome OS R96 14192 (fwupd 1.6.3) as well as Ubuntu 20.04 (snap variant 1.6.2), still trying to recreate by myself.

```
01:36:36:0239 FuDevice             failed to set quirk key DellDockUnlockTarget=9: quirk key not supported
01:36:36:0239 FuDevice             failed to set quirk key DellDockInstallDurationI2C=360: quirk key not supported
01:36:36:0239 FuDevice             failed to set quirk key DellDockBlobMajorOffset=0x18400: quirk key not supported
01:36:36:0239 FuDevice             failed to set quirk key DellDockBlobMinorOffset=0x18401: quirk key not supported
01:36:36:0239 FuDevice             failed to set quirk key DellDockBlobBuildOffset=0x18402: quirk key not supported
```
[image]

**Steps to Reproduce**
fwupdtool get-devices

**Expected behavior**
In `dell-dock` plugin already set the priority for `dell-dock` over `synaptics_mst` on `VMM5331` mst device specifically, device shouldn't be added by `synaptics_msc` plugin.

**fwupd version information**
`fwupdtool --version`
client version: 1.6.3

**Dock SKU**
Please mention which module is installed in your WD19.

- [x] WD19 (Single-C)
- [x] WD19TB (Thunderbolt)
- [ ] WD19DC (Dual-C)

**Verbose daemon logs**
[8h4n0g3_all_ng.log](https://github.com/fwupd/fwupd/files/7103251/8h4n0g3_all_ng.log)

```
09:24:23:0480 FuMain               ADDED:
FuSynapticsMstDevice:
VMM5331 in Dell dock
  DeviceId:             585c346c009da3cc19a400fb59a4ed8d0cbd0151
  CompositeId:          b5556eda4eb2315f490b0f72884f8c891c055274
  Guid:                 b3908108-9d2a-5f8e-83d7-729707550d27 <- MST-259
  Guid:                 89fec0b6-6b76-5008-b82c-5e5c6c164007 <- MST-panamera-vmm5331-259
  Guid:                 a26f73f2-3738-59a4-aa9c-5f45109eed21 <- MST-panamera-259
  Guid:                 f15aa55c-9cd5-5942-85ae-a6bf8740b96c <- MST-panamera
  Summary:              Multi Stream Transport controller
  Plugin:               synaptics_mst
  Protocol:             com.synaptics.mst
```

CragW, 2021-09-03:
@superm1 do you know any recent changes might be causing this issue?

hughsie, 2021-09-03:
@CragW can you upload the complete --verbose log showing enumeration please? Thanks.

CragW, 2021-09-03:
I attached [8h4n0g3_all_ng.log](https://github.com/fwupd/fwupd/files/7103251/8h4n0g3_all_ng.log) to the bug description, it's produced by `fwupdtool get-devices -v` command, please let me know if something else needed.

superm1, 2021-09-03:
It seems like in addition to the synaptics mst probing now, there is also two MST devices as a result and the de-duplication stopped working..
```
09:24:24:0070 FuMain               VMM5331 in Dell dock
  DeviceId:             e340b01c4bc406b3c7368ab138e8f7374b0e40fe
  CompositeId:          b5556eda4eb2315f490b0f72884f8c891c055274
  Guid:                 89fec0b6-6b76-5008-b82c-5e5c6c164007 <- MST-panamera-vmm5331-259
  Summary:              Multi Stream Transport controller
  Plugin:               dell_dock
  Protocol:             com.synaptics.mst
  Flags:                updatable|require-ac|supported|registered|dual-image|usable-during-update|skips-restart
  Vendor:               Dell Inc.
  VendorId:             USB:0x413C
  Version:              05.04.06
  VersionFormat:        triplet
  Icon:                 dock-usb,video-display
  InstallDuration:      360
  Created:              2021-08-17

09:24:24:0070 FuMain               VMM5331 in Dell dock
  DeviceId:             585c346c009da3cc19a400fb59a4ed8d0cbd0151
  CompositeId:          b5556eda4eb2315f490b0f72884f8c891c055274
  Guid:                 b3908108-9d2a-5f8e-83d7-729707550d27 <- MST-259
  Guid:                 89fec0b6-6b76-5008-b82c-5e5c6c164007 <- MST-panamera-vmm5331-259
  Guid:                 a26f73f2-3738-59a4-aa9c-5f45109eed21 <- MST-panamera-259
  Guid:                 f15aa55c-9cd5-5942-85ae-a6bf8740b96c <- MST-panamera
  Summary:              Multi Stream Transport controller
  Plugin:               synaptics_mst
  Protocol:             com.synaptics.mst
  Flags:                updatable|require-ac|supported|registered|dual-image|usable-during-update|skips-restart
  Vendor:               Dell Inc.
  VendorId:             DRM_DP_AUX_DEV:0x06CB
  Version:              5.04.06
  VersionFormat:        triplet
  Icon:                 video-display
  InstallDuration:      95
  Created:              2021-08-17
```

hughsie, 2021-09-03:
So we're adding it okay:

    FuDellDockMst:
    VMM5331 in Dell dock
      DeviceId:             e340b01c4bc406b3c7368ab138e8f7374b0e40fe
      CompositeId:          b5556eda4eb2315f490b0f72884f8c891c055274
      Guid:                 89fec0b6-6b76-5008-b82c-5e5c6c164007 <- MST-panamera-vmm5331-259
      Summary:              Multi Stream Transport controller
      Plugin:               dell_dock
      Protocol:             com.synaptics.mst

but we're also adding it again:

    09:24:23:0480 FuMain               ADDED:
    FuSynapticsMstDevice:
    VMM5331 in Dell dock
      DeviceId:             585c346c009da3cc19a400fb59a4ed8d0cbd0151
      CompositeId:          b5556eda4eb2315f490b0f72884f8c891c055274
      Guid:                 b3908108-9d2a-5f8e-83d7-729707550d27 <- MST-259
      Guid:                 89fec0b6-6b76-5008-b82c-5e5c6c164007 <- MST-panamera-vmm5331-259
      Guid:                 a26f73f2-3738-59a4-aa9c-5f45109eed21 <- MST-panamera-259
      Guid:                 f15aa55c-9cd5-5942-85ae-a6bf8740b96c <- MST-panamera
      Summary:              Multi Stream Transport controller
      Plugin:               synaptics_mst
      Protocol:             com.synaptics.mst

So the `quirk key not supported` comes from:

	# MST Hub
	[MST-panamera-vmm5331-259]
	Plugin = dell_dock
	...
	DellDockInstallDurationI2C = 360
	DellDockBlobMajorOffset = 0x18400
	DellDockBlobMinorOffset = 0x18401
	DellDockBlobBuildOffset = 0x18402

which matches both the Dell dock and the Synaptics plugin. So lets dig a bit further. We're supposed to dedupe and throw away the native one, as we have:

    fu_plugin_add_rule(plugin, FU_PLUGIN_RULE_BETTER_THAN, "synaptics_mst");

all that does is set the plugin `priority` a little higher, as we can see:

    09:24:22:0734 FuPluginList         dell_dock [0] better than synaptics_mst [0] so bumping to [1]

This has the effect of making the dell_dock device a little higher in priority:

    VMM5331 in Dell dock
      DeviceId:             e340b01c4bc406b3c7368ab138e8f7374b0e40fe
      Plugin:               dell_dock
    ...
      Priority:             1

...but we don't seem to actually call `fu_device_get_priority` anywhere useful. I suspect this is a regression.

hughsie, 2021-09-03:
Maybe a stupid question, but could we just do:

    [MST-panamera-259]
    Inhibit = Handled by Dell Dock

and make everything *much* simpler?

CragW, 2021-09-03:
Out of curiosity if device has been probed by `dell-dock` plugin why `synaptics_mst` probing it again? Is it because `MST-panamera-259` Guid gets extended/added from `dell-dock` plugin?

superm1, 2021-09-03:
> ...but we don't seem to actually call fu_device_get_priority anywhere useful. I suspect this is a regression.

It certainly is a regression - this used to work properly.  I distinctly remember messages from the lines of:
```
ignoring device %s [%s] as better device %s [%s] already exists
```
And so that means this is probably an effect of 078beafb2d2817749870505f5eb8ad0553136d86.

superm1, 2021-09-03:
> Maybe a stupid question, but could we just do:
> 
> ```
> [MST-panamera-259]
> Inhibit = Handled by Dell Dock
> ```
> 
> and make everything _much_ simpler?

Well I feel like the MST plugin shouldn't have probed in the first place though.  Shouldn't it have been looking at the `Plugin` key and said oh I don't match, skip this one.

hughsie, 2021-09-03:
> why synaptics_mst probing it again

Both plugins are probing the hardware and adding the same instance ID (but with different physical and logical IDs), which gives two devices with different device IDs, but with one common GUID.

https://github.com/fwupd/fwupd/commit/078beafb2d2817749870505f5eb8ad0553136d86#diff-0ca8e2440f2a61dd519d0d54169d640924b38a9455e2bbf9a24f25e7de348b0cL716 looks like to be the root cause.

hughsie, 2021-09-03:
>  Shouldn't it have been looking at the Plugin key

Ohh hang on you're right. The synaptics MST quirk is *opt-in* not opt-out. Why did synaptics MST enumerate the device at all?

superm1, 2021-09-03:
It seems to me it probably gets backend device added for `PCI_SLOT_NAME=0000:00:02.0:drm_dp_aux2` which at that time hasn't matched any quirks so `fu_device_get_possible_plugins` won't have anything.  The GUID and quirk match is then set "within" the plugin when the instance IDs are added

hughsie, 2021-09-03:
Nah, look closer at the dump:

	FuSynapticsMstDevice:
	VMM5331 in Dell dock
	  DeviceId:             585c346c009da3cc19a400fb59a4ed8d0cbd0151
	  ...
	  Plugin:               synaptics_mst
	  ...
	  DeviceFile:           /dev/drm_dp_aux2
	  DeviceKind:
	  Mode:                 DIRECT

We're missing an **abort!!** if `DeviceKind==NULL`

hughsie, 2021-09-03:
test PR coming up.

closed 2021-09-03 by hughsie