fwupd/fwupd #3718: gi-docgen errror

dkwo, 2021-09-07:
After a recent update of gi-docgen on void linux, when building fwupd I get the error
```
[51/57] Generating doc-fwupd with a custom command
FAILED: docs/libfwupd
/usr/bin/gi-docgen generate --quiet --add-include-path=/builddir/fwupd-1.6.3/build/docs/../libfwupd --config=docs/fwupd.toml --output-dir=docs/libfwupd --no-namespace-dir --content-dir=/builddir/fwupd-1.6.3/docs libfwupd/Fwupd-2.0.gir
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/gidocgen/gidocmain.py", line 74, in run
    res = options.run_func(options)
  File "/usr/lib/python3.9/site-packages/gidocgen/gdgenerate.py", line 2837, in run
    gen_reference(conf, options, parser.get_repository(), templates_dir, theme_conf, content_dirs, output_dir)
  File "/usr/lib/python3.9/site-packages/gidocgen/gdgenerate.py", line 2767, in gen_reference
    src = utils.find_extra_content_file(content_dirs, config.urlmap_file)
  File "/usr/lib/python3.9/site-packages/gidocgen/utils.py", line 880, in find_extra_content_file
    raise FileNotFoundError(f"Content file {file} not found in any content directory")
FileNotFoundError: Content file ../urlmap.js not found in any content directory
```
It was working fine before the update.
Is the file urlmap.js supposed to be provided by fwupd?

Thanks.

dkwo, 2021-09-07:
This is the build attempt and error log https://github.com/void-linux/void-packages/pull/32636

hughsie, 2021-09-07:
You can use `-Ddocs=gtkdoc` for no if that helps, while we sort out the root cause.

dkwo, 2021-09-07:
That helps, thanks.

superm1, 2021-09-07:
>This is the build attempt and error log void-linux/void-packages#32636

Just FYI, once this is all up and working for y'all, please contribute the packaging stuff you guys are doing that's different from what the void CI does back into fwupd CI so we can keep track of the things done differently better from upstream.

>FileNotFoundError: Content file ../urlmap.js not found in any content directory

So `urlmap.js` is in `data/urlmap.js`, but in the `*.toml.in` we try to refer to it as `../urlmap.js`.  This is intentional so that we share the same file between two gi-docgen instances we generate but maybe it's breaking for you?
See if this helps: https://github.com/fwupd/fwupd/pull/3721

paper42, 2021-09-07:
> So `urlmap.js` is in `data/urlmap.js`, but in the `*.toml.in` we try to refer to it as `../urlmap.js`. This is intentional so that we share the same file between two gi-docgen instances we generate [...]

This fix seems to work, thank you.

superm1, 2021-09-08:
fixed with https://github.com/fwupd/fwupd/pull/3721

closed 2021-09-08 by superm1

dkwo, 2021-09-08:
That fixes it, thank you.

dkwo, 2021-09-08:
@superm1 Once this is merged into void repos, what should be done to update fwupd ci?

superm1, 2021-09-08:
Compare what your packaging does and if anything significantly different send a PR