fwupd/fwupd #3719: Is there any way to recover a device from an interrupted firmware update?

ivanmikh, 2021-09-07:
I am looking for a way to recover a modem after an interrupted firmware update. I am working with Quectel modems, ModemManager and modem-manager plugin in fwupd.

Currently, if the firmware upgrade process is interrupted (e.g. a power failure on the host), the modem boots into recovery mode. In this mode, it will not be recognized by the ModemManager and will not be exposed to fwupd. For now, we just use a custom service that will recover the modem in case firmware update fails.

Maybe there is a way to implement this kind of functionality in fwupd?

superm1, 2021-09-07:
When it is in this mode how is it recognized and how to recover specifically?

I think you need something that can recognize say the USB device in bootloader mode and create enough of a fwupd device that it can flash back.

It might need to be a recovery plugin for both steps or maybe just a plugin that creates the device and sets GUID so regular modem manager plugin and recover it without using modem manager.

ivanmikh, 2021-09-07:
> When it is in this mode how is it recognized and how to recover specifically?

It boots up with a different vendor and device id (generic Qualcomm ids). Recovery process is very similar to the update process: we need to setup a firmware loader directory with a specific binary and reload the modem into download mode, then start flashing new firmware.

> I think you need something that can recognize say the USB device in bootloader mode and create enough of a fwupd device that it can flash back.
> It might need to be a recovery plugin for both steps [...]

I think the second option sounds good. Is there any example on how to pass a device from one plugin to another?

superm1, 2021-09-07:
>It boots up with a different vendor and device id (generic Qualcomm ids). Recovery process is very similar to the update process: we need to setup a firmware loader directory with a specific binary and reload the modem into download mode, then start flashing new firmware.

When it's got the generic IDs, how do you tell it's your device?  Do you have some sort of NVRAM offset you can look at, or somewhere for some board strapping ID?  You'll need something to properly identify the device to be able to properly build the GUID for it.

>I think the second option sounds good. Is there any example on how to pass a device from one plugin to another?

Look at plugins that use `fu_plugin_device_registered` to act on a device provided by a plugin.  Particularly a good example for your case is the interaction between the `dell` and `uefi` plugins.  The `dell` plugin makes a virtual device and the `uefi` plugin picks up the virtual device and makes a device from it.

Other examples are more that one plugin makes a device and others add or take attributes from it.
@hughsie any other comments?

closed 2021-09-07 by superm1