fwupd/fwupd #3759: get-devices returns different GUIDs for TPM

therealjuanmartinez, 2021-09-16:
**Describe the bug**
fwupdtool defaults to 'tpm' plugin over 'dell' plugin 

**Steps to Reproduce**
```
#Observe GUIDs generated for TPM by non-dell plugin
sudo fwupdtool get-devices --show-all
```

**Expected behavior**
Output for TPM should instead match output from the below command
`sudo fwupdtool get-devices --plugins dell --plugins uefi_capsule`

**fwupd version information**
```
client version: 1.6.3
compile-time dependency versions
        gusb:   0.3.7

daemon version: 1.6.3
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):  SOURCE

**fwupd device information**
```WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
WARNING: This package has not been validated, it may not work properly.
Latitude 5490
│
├─Core™ i5-8350U CPU @ 1.70GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000ea
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         561403e8-143a-5071-ab09-bf5e4c146983
│                         54bdc429-0179-5019-9cb6-8a343324c97c
│     Device Flags:       • Internal device
│
├─SC311 SATA 256GB:
│     Device ID:          64ec4cd1e1c9565e79b00f4e7221b1c689b33e96
│     Summary:            ATA drive
│     Current version:    70000P10
│     Vendor:             SK hynix (OUI:ace42e, ATA:0x1C5C)
│     Update Error:       Cannot install update when not on AC power
│     GUIDs:              cdd8cb3a-3604-5af9-a7db-f502a05214ed
│                         64d370ac-e8e2-51c6-a419-3d02aacfe39b
│                         db0a94bf-8c1a-587f-ae81-d43aab70f084
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    66816
│ │   Minimum Version:    66816
│ │   Vendor:             Dell Inc. (DMI:Dell Inc.)
│ │   Update State:       Success
│ │   Update Error:       Cannot install update when not on AC power
│ │   GUIDs:              cdcae5ae-413a-4198-b866-8028e994dd53
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       a380f883-0986-57e2-9a84-abffe0993aca
│ │   Device Flags:       • Internal device
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │                       • Updatable
│ │
│ ├─Intel AMT [unprovisioned]:
│ │     Device ID:        e2623122c99d58220498aacbfcfdb1baebbae3c5
│ │     Summary:          Hardware and firmware technology for remote out-of-band management
│ │     Current version:  11.8.55.3510
│ │     Bootloader Version:11.8.55.3510
│ │     Vendor:           Intel Corporation
│ │     GUID:             2800f812-b7b4-2d4b-aca8-46e0ff65814c
│ │     Device Flags:     • Internal device
│ │
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  77
│       Minimum Version:  77
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            00fe3755-a4d8-5ef7-ba5f-47979fbb3423
│                         4a6cd2cb-8741-5257-9d1f-89a275dacca7
│                         c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Only version upgrades are allowed
│
├─TPM:
│ │   Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Current version:    7.2.0.1
│ │   Vendor:             Nuvoton Technology (TPM:NTC)
│ │   Update State:       Success
│ │   GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6
│ │                       fac1c8f3-73c8-5cd6-8330-07a3690b5140
│ │                       4521fefa-4a2f-537f-a7f9-d353841eb18e
│ │                       e9ccc1dc-960a-5e09-afe9-e59a904b776d
│ │                       c05f797a-8d97-547f-a7a3-c5c0339748e9
│ │   Device Flags:       • Internal device
│ │
│ └─Event Log:
│       Device ID:        58bd405f31c48e6eca290b425f530a94c91e955c
│       GUID:             a25657fe-b5dc-5be0-8b78-8b9dfec678ff
│       Device Flags:     • Internal device
│
└─UHD Graphics 620:
      Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
      Current version:    07
      Vendor:             Intel Corporation (PCI:0x8086)
      GUIDs:              d5895039-1e9c-5163-a122-49b4093e981b
                          c5acff38-33cd-5363-bf21-40ad14cd7601
                          fed2efa4-7045-55a1-b4fc-b29283d59fe5
                          8b72a10c-1279-5f8e-a28a-34fb11a58240
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available
```

**Additional questions**

- Operating system and version: 20.04.1-Ubuntu  5.11.0-34-generic  
- Have you tried rebooting? Yes
- Is this a regression? Unclear

Would also add that the following line suggests the intent for dell plugin to override tpm:
https://github.com/fwupd/fwupd/blob/c5dd4d4d4a7050891cacdd2f4b011b5a99b14f3b/plugins/dell/fu-plugin-dell.c#L889

superm1, 2021-09-16:
Can you please add a verbose log for both:

`sudo fwupdtool get-devices --plugins dell --plugins uefi_capsule --plugins tpm`
and 
`sudo fwupdtool get-devices --plugins dell --plugins uefi_capsule`

therealjuanmartinez, 2021-09-16:
`sudo fwupdtool get-devices --plugins dell --plugins uefi_capsule --plugins tpm --verbose`

```
14:11:26:0009 FuDebug              Verbose debugging enabled (on console 0)
14:11:26:0010 FuMain               locked /run/lock/fwupdtool
14:11:26:0016 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
14:11:26:0016 FuConfig             loading config values from /usr/local/etc/fwupd/daemon.conf
14:11:26:0016 FuConfig             using autodetected max archive size 18.4 EB
14:11:26:0017 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for 'gio-vfs'
14:11:26:0017 FuRemoteList         loading /usr/local/share/fwupd/metainfo
14:11:26:0017 XbSilo               attempting to load /var/local/cache/fwupd/metainfo.xmlb
14:11:26:0017 XbSilo               file: b3b81aee-efd0-4c10-14d6-121740735045, current:b3b81aee-efd0-4c10-14d6-121740735045, cached: (null)
14:11:26:0017 XbSilo               loading silo with file contents
14:11:26:0018 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/fwupd-tests.conf
14:11:26:0018 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/lvfs-testing.conf
14:11:26:0018 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/vendor-directory.conf
14:11:26:0018 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/lvfs.conf
14:11:26:0018 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/vendor.conf
14:11:26:0018 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/dell-esrt.conf
14:11:26:0018 FuRemoteList         ordering lvfs-testing=lvfs+1
14:11:26:0018 FuRemoteList         ignoring unfound remote fwupd
14:11:26:0018 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
14:11:26:0018 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
14:11:26:0018 Jcat                 trying to load certificate from /usr/local/etc/pki/fwupd/LVFS-CA.pem
14:11:26:0018 Jcat                 reading /usr/local/etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
14:11:26:0018 Jcat                 loaded 1 certificates
14:11:26:0018 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
14:11:26:0018 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
14:11:26:0019 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
14:11:26:0019 Jcat                 trying to load certificate from /usr/local/etc/pki/fwupd-metadata/LVFS-CA.pem
14:11:26:0019 Jcat                 reading /usr/local/etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
14:11:26:0019 Jcat                 loaded 1 certificates
14:11:26:0019 Jcat                 reading /var/local/lib/fwupd/pki/secret.key with 2459 bytes
14:11:26:0019 Jcat                 reading /var/local/lib/fwupd/pki/client.pem with 1383 bytes
14:11:26:0026 FuEngine             client certificate exists and working
14:11:26:0026 FuHistory            trying to open database '/var/local/lib/fwupd/pending.db'
14:11:26:0026 FuHistory            got schema version of 7
14:11:26:0026 FuHistory            migrating v7 database by altering
14:11:26:0026 FuHistory            schema version 7 is unknown
14:11:26:0026 FuSmbios             trying to read /sys/class/dmi/id
14:11:26:0026 FuHwids              smbios property Manufacturer=Dell Inc.
14:11:26:0026 FuHwids              smbios property EnclosureKind=a
14:11:26:0026 FuHwids              smbios property Family=Latitude
14:11:26:0026 FuHwids              smbios property ProductName=Latitude 5490
14:11:26:0026 FuHwids              smbios property ProductSku=0816
14:11:26:0026 FuHwids              smbios property BiosVendor=Dell Inc.
14:11:26:0026 FuHwids              smbios property BiosVersion=1.5.0
14:11:26:0026 FuHwids              smbios property BiosMajorRelease=01
14:11:26:0026 FuHwids              smbios property BiosMinorRelease=05
14:11:26:0026 FuHwids              smbios property FirmwareMajorRelease=ff
14:11:26:0026 FuHwids              smbios property FirmwareMinorRelease=ff
14:11:26:0026 FuHwids              smbios property BaseboardManufacturer=Dell Inc.
14:11:26:0026 FuHwids              smbios property BaseboardProduct=0DH2HV
14:11:26:0027 XbSilo               attempting to load /var/local/cache/fwupd/metadata.xmlb
14:11:26:0027 XbSilo               file: c7d12a9f-592b-2c1d-4c8d-a8b9743b1436, current:c7d12a9f-592b-2c1d-4c8d-a8b9743b1436, cached: (null)
14:11:26:0027 XbSilo               loading silo with file contents
14:11:26:0027 FuEngine             1 components now in silo
14:11:26:0030 FuPlugin             init(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_uefi_capsule.so)
14:11:26:0031 FuPlugin             init(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_tpm.so)
14:11:26:0031 FuContext            added udev subsystem watch of tpm
14:11:26:0031 FuPlugin             init(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_dell.so)
14:11:26:0031 FuPluginDell         Using libsmbios 2.4
14:11:26:0031 GLib                 setenv()/putenv() are not thread-safe and should not be used after threads are created
[... cut ...]
  Checksum:             SHA256(2aa125d1d08f0864ec60d02c6ffa61cdfaf099a6e1f111b6a63a3019c8429ab6)
  Vendor:               Dell Inc.
  VendorId:             DMI:Dell Inc.
  Version:              66816
  VersionLowest:        66816
  VersionFormat:        number
  VersionRaw:           0x00010500
  VersionLowestRaw:     0x00010500
  Icon:                 computer
  Created:              2021-09-16
  UpdateState:          success

14:11:26:0224 FuPlugin             destroy(uefi_capsule)
14:11:26:0225 FuPlugin             destroy(dell)

```

therealjuanmartinez, 2021-09-16:
`sudo fwupdtool get-devices --plugins dell --plugins uefi_capsule --verbose`

```
14:11:47:0079 FuDebug              Verbose debugging enabled (on console 0)
14:11:47:0080 FuMain               locked /run/lock/fwupdtool
14:11:47:0085 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
14:11:47:0086 FuConfig             loading config values from /usr/local/etc/fwupd/daemon.conf
14:11:47:0086 FuConfig             using autodetected max archive size 18.4 EB
14:11:47:0087 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for 'gio-vfs'
14:11:47:0087 FuRemoteList         loading /usr/local/share/fwupd/metainfo
14:11:47:0087 XbSilo               attempting to load /var/local/cache/fwupd/metainfo.xmlb
14:11:47:0087 XbSilo               file: b3b81aee-efd0-4c10-14d6-121740735045, current:b3b81aee-efd0-4c10-14d6-121740735045, cached: (null)
14:11:47:0087 XbSilo               loading silo with file contents
14:11:47:0087 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/fwupd-tests.conf
14:11:47:0088 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/lvfs-testing.conf
14:11:47:0088 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/vendor-directory.conf
14:11:47:0088 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/lvfs.conf
14:11:47:0088 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/vendor.conf
14:11:47:0088 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/dell-esrt.conf
14:11:47:0088 FuRemoteList         ordering lvfs-testing=lvfs+1
14:11:47:0088 FuRemoteList         ignoring unfound remote fwupd
14:11:47:0088 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
14:11:47:0088 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
14:11:47:0088 Jcat                 trying to load certificate from /usr/local/etc/pki/fwupd/LVFS-CA.pem
14:11:47:0088 Jcat                 reading /usr/local/etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
14:11:47:0089 Jcat                 loaded 1 certificates
14:11:47:0089 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
14:11:47:0089 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
14:11:47:0089 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
14:11:47:0089 Jcat                 trying to load certificate from /usr/local/etc/pki/fwupd-metadata/LVFS-CA.pem
14:11:47:0089 Jcat                 reading /usr/local/etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
14:11:47:0089 Jcat                 loaded 1 certificates
14:11:47:0089 Jcat                 reading /var/local/lib/fwupd/pki/secret.key with 2459 bytes
14:11:47:0089 Jcat                 reading /var/local/lib/fwupd/pki/client.pem with 1383 bytes
14:11:47:0098 FuEngine             client certificate exists and working
14:11:47:0098 FuHistory            trying to open database '/var/local/lib/fwupd/pending.db'
14:11:47:0098 FuHistory            got schema version of 7
14:11:47:0098 FuHistory            migrating v7 database by altering
14:11:47:0098 FuHistory            schema version 7 is unknown
14:11:47:0099 FuSmbios             trying to read /sys/class/dmi/id
14:11:47:0099 FuHwids              smbios property Manufacturer=Dell Inc.
14:11:47:0099 FuHwids              smbios property EnclosureKind=a
14:11:47:0099 FuHwids              smbios property Family=Latitude
14:11:47:0099 FuHwids              smbios property ProductName=Latitude 5490
14:11:47:0099 FuHwids              smbios property ProductSku=0816
14:11:47:0099 FuHwids              smbios property BiosVendor=Dell Inc.
14:11:47:0099 FuHwids              smbios property BiosVersion=1.5.0
14:11:47:0099 FuHwids              smbios property BiosMajorRelease=01
14:11:47:0099 FuHwids              smbios property BiosMinorRelease=05
14:11:47:0099 FuHwids              smbios property FirmwareMajorRelease=ff
14:11:47:0099 FuHwids              smbios property FirmwareMinorRelease=ff
14:11:47:0099 FuHwids              smbios property BaseboardManufacturer=Dell Inc.
14:11:47:0099 FuHwids              smbios property BaseboardProduct=0DH2HV
14:11:47:0099 XbSilo               attempting to load /var/local/cache/fwupd/metadata.xmlb
14:11:47:0099 XbSilo               file: c7d12a9f-592b-2c1d-4c8d-a8b9743b1436, current:c7d12a9f-592b-2c1d-4c8d-a8b9743b1436, cached: (null)
14:11:47:0099 XbSilo               loading silo with file contents
14:11:47:0099 FuEngine             1 components now in silo
14:11:47:0103 FuPlugin             init(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_uefi_capsule.so)
14:11:47:0104 FuPlugin             init(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_dell.so)
14:11:47:0104 FuPluginDell         Using libsmbios 2.4
14:11:47:0104 GLib                 setenv()/putenv() are not thread-safe and should not be used after threads are created
14:11:47:0104 FuEngine             plugins disabled: realtek_mst, ep963x, steelseries, synaptics_mst, acpi_phat, solokey, acpi_dmar, thelio_io, linux_sleep, uefi_pk, bios, vli, analogix, fresco_pd, logitech_hidpp, ccgx, fastboot, amt, platform_integrity, dell_dock, altos, pci_bcr, goodixmoc, elantp, dfu_csr, csr, uefi_recovery, wac [... cut ...]
  Protocol:             org.uefi.capsule
  Flags:                internal|require-ac|registered
  Vendor:               Dell Inc.
  VendorId:             PCI:0x1028
  Version:              7.2.0.1
  VersionFormat:        quad
  Icon:                 computer
  Created:              2021-09-16
  UpdateState:          success
  UpdateError:          Updating disabled due to TPM ownership

14:11:47:0265 FuPlugin             destroy(dell)
14:11:47:0265 FuPlugin             destroy(uefi_capsule)

```

superm1, 2021-09-16:
So both the plugins are using the same physical device ID so it gets reused and problems ensue:
```
14:11:26:0219 FuDeviceList         found existing device c6a80ac3a22083423992a3cb15018989f37834d6
14:11:26:0219 FuDeviceList         copying update error Updating disabled due to TPM ownership to new device
14:11:26:0219 FuDeviceList         copying update state success to new device
14:11:26:0219 FuDeviceList         ::changed c6a80ac3a22083423992a3cb15018989f37834d6
```

Can you please see if this patch helps?  Please share updated logs as well with it.
```
$ git diff
diff --git a/plugins/dell/fu-plugin-dell.c b/plugins/dell/fu-plugin-dell.c
index 88d7c4de..1055dcb2 100644
--- a/plugins/dell/fu-plugin-dell.c
+++ b/plugins/dell/fu-plugin-dell.c
@@ -761,6 +761,7 @@ fu_plugin_dell_detect_tpm(FuPlugin *plugin, GError **error)
        /* build Standard device nodes */
        dev = fu_device_new_with_context(ctx);
        fu_device_set_physical_id(dev, "DEVNAME=/dev/tpm0");
+       fu_device_set_logical_id(dev, "UEFI");
        fu_device_add_instance_id(dev, tpm_guid_raw);
        fu_device_add_instance_id(dev, "system-tpm");
        fu_device_set_vendor(dev, "Dell Inc.");
```

therealjuanmartinez, 2021-09-16:
Applied patch, results for both commands below:

`sudo fwupdtool get-devices --plugins dell --plugins uefi_capsule --plugins tpm --verbose`

```
15:43:30:0721 FuDebug              Verbose debugging enabled (on console 0)
15:43:30:0722 FuMain               locked /run/lock/fwupdtool
15:43:30:0731 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
15:43:30:0732 FuConfig             loading config values from /usr/local/etc/fwupd/daemon.conf
15:43:30:0733 FuConfig             using autodetected max archive size 18.4 EB
15:43:30:0735 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for 'gio-vfs'
15:43:30:0736 FuRemoteList         loading /usr/local/share/fwupd/metainfo
15:43:30:0737 XbSilo               attempting to load /var/local/cache/fwupd/metainfo.xmlb
15:43:30:0737 XbSilo               file: f2e20b98-9117-1a40-bcda-84b23d4659f7, current:f2e20b98-9117-1a40-bcda-84b23d4659f7, cached: (null)
15:43:30:0738 XbSilo               loading silo with file contents
15:43:30:0738 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/fwupd-tests.conf
15:43:30:0739 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/lvfs-testing.conf
15:43:30:0740 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/vendor-directory.conf
15:43:30:0740 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/lvfs.conf
15:43:30:0741 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/vendor.conf
15:43:30:0741 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/dell-esrt.conf
15:43:30:0742 FuRemoteList         ordering lvfs-testing=lvfs+1
15:43:30:0743 FuRemoteList         ignoring unfound remote fwupd
15:43:30:0743 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
15:43:30:0743 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
15:43:30:0743 Jcat                 trying to load certificate from /usr/local/etc/pki/fwupd/LVFS-CA.pem
15:43:30:0744 Jcat                 reading /usr/local/etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
15:43:30:0744 Jcat                 loaded 1 certificates
15:43:30:0744 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
15:43:30:0744 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
15:43:30:0745 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
15:43:30:0745 Jcat                 trying to load certificate from /usr/local/etc/pki/fwupd-metadata/LVFS-CA.pem
15:43:30:0745 Jcat                 reading /usr/local/etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
15:43:30:0745 Jcat                 loaded 1 certificates
15:43:30:0745 Jcat                 reading /var/local/lib/fwupd/pki/secret.key with 2459 bytes
15:43:30:0745 Jcat                 reading /var/local/lib/fwupd/pki/client.pem with 1383 bytes
15:43:30:0761 FuEngine             client certificate exists and working
15:43:30:0762 FuHistory            trying to open database '/var/local/lib/fwupd/pending.db'
15:43:30:0763 FuHistory            got schema version of 7
15:43:30:0763 FuHistory            migrating v7 database by altering
15:43:30:0764 FuHistory            schema version 7 is unknown
15:43:30:0765 FuSmbios             trying to read /sys/class/dmi/id
15:43:30:0766 FuHwids              smbios property Manufacturer=Dell Inc.
15:43:30:0766 FuHwids              smbios property EnclosureKind=a
15:43:30:0766 FuHwids              smbios property Family=Latitude
15:43:30:0766 FuHwids              smbios property ProductName=Latitude 5490
15:43:30:0767 FuHwids              smbios property ProductSku=0816
15:43:30:0767 FuHwids              smbios property BiosVendor=Dell Inc.
15:43:30:0767 FuHwids              smbios property BiosVersion=1.5.0
15:43:30:0767 FuHwids              smbios property BiosMajorRelease=01
15:43:30:0767 FuHwids              smbios property BiosMinorRelease=05
15:43:30:0767 FuHwids              smbios property FirmwareMajorRelease=ff
15:43:30:0767 FuHwids              smbios property FirmwareMinorRelease=ff
15:43:30:0767 FuHwids              smbios property BaseboardManufacturer=Dell Inc.
15:43:30:0767 FuHwids              smbios property BaseboardProduct=0DH2HV
15:43:30:0768 XbSilo               attempting to load /var/local/cache/fwupd/metadata.xmlb
15:43:30:0768 XbSilo               file: 70cf277c-af4d-2ec6-b485-7f14d9f05253, current:70cf277c-af4d-2ec6-b485-7f14d9f05253, cached: (null)
15:43:30:0768 XbSilo               loading silo with file contents
15:43:30:0768 FuEngine             1 components now in silo
15:43:30:0780 FuPlugin             init(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_uefi_capsule.so)
15:43:30:0782 FuPlugin             init(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_tpm.so)
15:43:30:0782 FuContext            added udev subsystem watch of tpm
15:43:30:0784 FuPlugin             init(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_dell.so)
15:43:30:0784 FuPluginDell         Using libsmbios 2.4
15:43:30:0784 GLib                 setenv()/putenv() are not thread-safe and should not be used after threads are created
[... cut ...]
  Protocol:             org.uefi.capsule
  Flags:                internal|require-ac|registered
  Vendor:               Dell Inc.
  VendorId:             PCI:0x1028
  Version:              7.2.0.1
  VersionFormat:        quad
  Icon:                 computer
  Created:              2021-09-16
  UpdateState:          success
  UpdateError:          Updating disabled due to TPM ownership

15:44:12:0619 FuPlugin             destroy(dell)
15:44:12:0619 FuPlugin             destroy(uefi_capsule)
```

superm1, 2021-09-16:
@hughsie This is further fallout from 078beafb2d2817749870505f5eb8ad0553136d86
Do you think we need to do anything to inhibit the TPM from the TPM plugin?  Since it's not updatable (and thus shouldn't show up in get-devices output) I'm leaning on no.

closed 2021-09-16 by superm1

superm1, 2021-09-16:
This has been backported to 1.6.x and 1.5.x branches as well, so next time point releases are spun it will be picked up for them too.

commit fb667fec (2021-09-16): Fix probing the Dell TPM (Fixes: #3759)

Since commit 078beaf, device de-duplication between plugins doesn't
work anymore.  Both the TPM and Dell plugin create a device with an
identical physical ID and no logical ID.  This makes the device list
think whichever plugin probes second is just updating the first.

To avoid this problem, set a logical ID in the Dell plugin, ensuring
that the device ID built from physical + logical is different from the
TPM plugin.

The side effect of this is that both devices will export into the device
list, but the TPM plugin's devices will not be upgradable.
commit c0f24225 (2021-09-16): Fix probing the Dell TPM (Fixes: #3759)

Since commit 078beaf, device de-duplication between plugins doesn't
work anymore.  Both the TPM and Dell plugin create a device with an
identical physical ID and no logical ID.  This makes the device list
think whichever plugin probes second is just updating the first.

To avoid this problem, set a logical ID in the Dell plugin, ensuring
that the device ID built from physical + logical is different from the
TPM plugin.

The side effect of this is that both devices will export into the device
list, but the TPM plugin's devices will not be upgradable.
commit a63d3150 (2021-09-16): Fix probing the Dell TPM (Fixes: #3759)

Since commit 078beaf, device de-duplication between plugins doesn't
work anymore.  Both the TPM and Dell plugin create a device with an
identical physical ID and no logical ID.  This makes the device list
think whichever plugin probes second is just updating the first.

To avoid this problem, set a logical ID in the Dell plugin, ensuring
that the device ID built from physical + logical is different from the
TPM plugin.

The side effect of this is that both devices will export into the device
list, but the TPM plugin's devices will not be upgradable.

ch [... cut]
commit 015eb88e (2021-09-16): Fix probing the Dell TPM (Fixes: #3759)

Since commit 078beaf, device de-duplication between plugins doesn't
work anymore.  Both the TPM and Dell plugin create a device with an
identical physical ID and no logical ID.  This makes the device list
think whichever plugin probes second is just updating the first.

To avoid this problem, set a logical ID in the Dell plugin, ensuring
that the device ID built from physical + logical is different from the
TPM plugin.

The side effect of this is that both devices will export into the device
list, but the TPM plugin's devices will not be upgradable.

ch [... cut]