fwupd/fwupd #3762: Incorrect shim detection on Arch Linux

xuanruiqi, 2021-09-16:
**Describe the bug**
Despite shim is installed on ESP, `fwupdmgr` reports that "Secure boot is enabled, but shim isn't installed to the EFI system partition".

**Steps to Reproduce**
Try to update the system firmware. I can only update by temporarily disabling secure boot.

**Expected behavior**
The system firmware update successfully, without having to disable secure boot.

**fwupd version information**

```shell
$ fwupdmgr` --version
client version:	1.6.3
compile-time dependency versions
	gusb:	0.3.7

daemon version:	1.6.3
```

Installed using pacman, from official Arch repository.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
XPS 15 9510
│
├─DLL0945:00 04F3:311C:
│     Device ID:          0eeea9b449795ed307fcba6fa46f596d9679b87b
│     Summary:            Touchpad
│     Current version:    0x000a
│     Bootloader Version: 0x0001
│     Vendor:             HIDRAW:0x04F3
│     GUIDs:              9b6fef1a-77aa-565c-9983-9da2dd2c932b
│                         46598f8d-5777-50c4-8da9-cda0cc8bbc26
│                         d00322f0-ff1f-5458-9c64-8a771fb0b72e
│                         4f9fe0ae-f26e-5c1d-8a4e-a7506fb02f96
│                         6cf7ae6a-8b98-5b63-9af8-b23b932f65de
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─11th Gen Intel Core™ i7-11800H @ 2.30GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000034
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         a6bd4ca5-75a6-5796-b564-66b5cab1b11b
│                         d9dd5e77-df17-5bab-b5ec-22827598bfed
│     Device Flags:       • Internal device
│   
├─Fingerprint Sensor:
│     Device ID:          c84f8d7b73aa5faedcd3f118733f91799c1a78ec
│     Summary:            Match-On-Chip fingerprint sensor
│     Current version:    01000264
│     Vendor:             Goodix (USB:0x27C6)
│     Install Duration:   10 seconds
│     GUIDs:              267adb9d-b0eb-5ad0-9a94-db1e15359ebe
│                         6f754a46-4a1d-590e-9c92-2ddafd352253
│     Device Flags:       • Updatable
│                         • Device can recover flash failures
│   
├─GA107M [GeForce RTX 3050 Ti Mobile]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a1
│     Vendor:             NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:              8ebe2a3f-46b6-5ca4-9ac5-b72d18a47b92
│                         bd688f31-3893-5ca8-9c81-061f455cd61e
[... 138 lines cut ...]
│   └── systemd
│       └── systemd-bootx64.efi
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── intel-ucode.img
├── loader
│   ├── entries
│   │   ├── arch.conf
│   │   └── arch.conf~
│   ├── loader.conf
│   └── random-seed
├── MOK.cer
└── vmlinuz-linux

12 directories, 17 files
```

**Additional questions**

- Operating system and version: Arch Linux
- Have you tried rebooting? Yes, many times
- Is this a regression? I don't know for sure, but it could be.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled? Yes.
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No.

xuanruiqi, 2021-09-16:
Actually, I just discovered that I have no `shimx64.efi`, so this is my fault...
Nevertheless, `bootx64.efi` should probably added to the shim search, since on some systems this is where shim is installed to. For example, this is what the Arch Wiki [tells us to do](https://wiki.archlinux.org/title/Unified_Extensible_Firmware_Interface/Secure_Boot#shim).

superm1, 2021-09-16:
> Actually, I just discovered that I have no `shimx64.efi`, so this is my fault...
> Nevertheless, `bootx64.efi` should probably added to the shim search, since on some systems this is where shim is [...]

This behavior you want (using bootx64.efi for shim) isn't the default off of removable partitions. Nonetheless fwupd does support it.
See https://github.com/fwupd/fwupd/blob/master/plugins/uefi-capsule/uefi_capsule.conf specifically the key regarding fallback to removable path.
You should probably update the wiki page to suggest if that path is used them fwupd needs to be configured to recognize it.

xuanruiqi, 2021-11-30:
I reopened the issue since it seems to be beyond a naming issue.

I renamed my shim to `shimx64.efi` (i.e., it is at `/boot/EFI/BOOT/shimx64.efi`), but I still get the error. I suppose there's a problem with the detection logic. Setting `FallbacktoRemovablePath=true` did not fix the issue. Maybe there's a detection path issue on Arch Linux.

Or is there an option that lets me hard code shim location or something like that?

superm1, 2021-11-30:
I think it would be good to run the daemon in verbose mode and share your daemon logs.

xuanruiqi, 2021-12-03:
[Here](https://gist.github.com/xuanruiqi/7f0233b8cdb7a8df14cfa0b3bd7018ed) they are! Unfortunately, as far as I can understand, they are not too informative.

xuanruiqi, 2021-12-03:
Also reposted UEFI system information:
```
$ tree /boot
/boot
├── 52a2eacc384c4dc3ba8296096c1f4390
├── EFI
│   ├── arch
│   │   ├── fw
│   │   │   ├── fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
│   │   │   └── fwupd-6e106831-11dd-4c1b-b22b-526d6f78683c.cap
│   │   └── fwupdx64.efi
│   ├── BOOT
│   │   ├── grubx64.efi
│   │   ├── mmx64.efi
│   │   └── shimx64.efi
│   ├── dell
│   │   └── bios
│   │       └── recovery
│   │           └── BIOS_CUR.rcv
│   ├── Linux
│   └── systemd
│       └── systemd-bootx64.efi
├── initramfs-linux-clear-fallback.img
├── initramfs-linux-clear.img
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── intel-ucode.img
├── loader
│   ├── entries
│   │   ├── arch-clear.conf
│   │   └── arch.conf
│   ├── loader.conf
│   └── random-seed
├── MOK.cer
├── vmlinuz-linux
└── vmlinuz-linux-clear
```

```
$ efibootmgr -v      
BootCurrent: 0003
Timeout: 0 seconds
BootOrder: 0003,0004,0000,0002
Boot0000* UEFI HTTPs Boot	PciRoot(0x0)/Pci(0x1f,0x6)/MAC(000000000000,0)/IPv4(0.0.0.00.0.0.0,0,0)/Uri()N.....YM....R,Y.
Boot0002* Linux Firmware Updater	HD(1,GPT,978b9e16-120a-41e2-baaa-c235ec501575,0x800,0x1ff800)/File(\EFI\arch\fwupdx64.efi)
Boot0003* Linux Boot Mangaer, shim chain-loaded	HD(1,GPT,978b9e16-120a-41e2-baaa-c235ec501575,0x800,0x1ff800)/File(\EFI\BOOT\shimx64.efi)
Boot0004* Linux Boot Manager	HD(1,GPT,978b9e16-120a-41e2-baaa-c235ec501575,0x800,0x1ff800)/File(\EFI\systemd\systemd-bootx64.efi)
```

```
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-6e106831-11dd-4c1b-b22b-526d6f78683c-0
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-ux-capsule
````

xuanruiqi, 2022-01-14:
Hi! I wonder if anyone is following up this?

superm1, 2022-01-14:
Thanks for the ping.  Have a try with this: https://github.com/fwupd/fwupd/pull/4172

xuanruiqi, 2022-01-22:
Hmmm, now I get past the error, but I still can't get the update to run.

xuanruiqi, 2022-01-22:
But I kind of understand what's going on now:

```
$ tree /boot                   
/boot
├── 52a2eacc384c4dc3ba8296096c1f4390
├── EFI
│   ├── arch
│   │   ├── fw
│   │   │   ├── fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
│   │   │   └── fwupd-6e106831-11dd-4c1b-b22b-526d6f78683c.cap
│   │   └── fwupdx64.efi
│   ├── BOOT
│   │   ├── bootx64.efi
│   │   ├── grubx64.efi
│   │   ├── mmx64.efi
│   │   └── shimx64.efi
│   ├── dell
│   │   ├── bios
│   │   │   └── recovery
│   │   │       ├── BIOS_CUR.rcv
│   │   │       └── BIOS_PRE.rcv
│   │   └── logs
│   │       ├── diags_current.xml
│   │       └── diags_previous.xml
│   ├── Linux
│   └── systemd
│       └── systemd-bootx64.efi
├── initramfs-linux-clear-fallback.img
├── initramfs-linux-clear.img
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── intel-ucode.img
├── loader
│   ├── entries
│   │   ├── arch-clear.conf
│   │   └── arch.conf
│   ├── loader.conf
│   └── random-seed
├── MOK.cer
├── vmlinuz-linux
└── vmlinuz-linux-clear
```

```
$ efibootmgr --verbose
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0002,0001,0004,0000,0005
Boot0000* UEFI HTTPs Boot	PciRoot(0x0)/Pci(0x1f,0x6)/MAC(000000000000,0)/IPv4(0.0.0.00.0.0.0,0,0)/Uri()N.....YM....R,Y.
Boot0001* Linux Boot Manager (shim signed)	HD(1,GPT,978b9e16-120a-41e2-baaa-c235ec501575,0x800,0x1ff800)/File(\EFI\BOOT\bootx64.efi)
Boot0002* Linux Firmware Updater	HD(1,GPT,978b9e16-120a-41e2-baaa-c235ec501575,0x800,0x1ff800)/File(\EFI\boot\bootx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0004* Linux Boot Manager	HD(1,GPT,978b9e16-120a-41e2-baaa-c235ec501575,0x800,0x1ff800)/File(\EFI\systemd\systemd-bootx64.efi)
Boot0005* UEFI RST PM9A1 NVMe Samsung 1024GB S65VNF0R634875 	HD(1,GPT,978b9e16-120a-41e2-baaa-c235ec501575,0x800,0x1ff800)/File(\EFI\Boot\BootX64.efi)N.....YM....R,Y.
```

Seems to be something wacky going on here.  It's not getting the updater efi path right?

xuanruiqi, 2022-01-22:
By copying shim to /boot/EFI/arch/shimx64.efi, I could get the update to run. But this is more of a hack than of a solution to the problem.

superm1, 2022-01-22:
Hmm, so really the path appended should be `..\arch\fwupdx64.efi` for fallback path _or_ we should put fwupdx64.efi into the removable path.

xuanruiqi, 2022-01-23:
Yes, either choice seems reasonable

xuanruiqi, 2022-02-22:
Pinging again!

superm1, 2022-02-22:
64e7ea25b18949941cac50729fa192c3dd9e4880 went in, does that help you enough?

superm1, 2022-02-22:
I haven't gotten a chance to look further at this, if you want to take a stab, help is welcome.

xuanruiqi, 2022-02-22:
I have no familiarity with the codebase, but maybe I could take a stab.

solsticedhiver, 2022-03-17:
I was able to do some firmware updates on a Lenovo x270, running archlinux, but this was not straight-forward.

Even though, I have `/usr/lib/fwupd/efi/fwupdx64.efi.signed` and `/EFI/arch/shimx64.efi`, updates would not run because ...
shim is looking for grubx64.efi not fwupdx64.efi.

And I had to manually install shimx64.efi in `/EFI/arch` as reported in this issue.

Moreover I had to rename fwupdx64.efi to grubx64.efi in `/EFI/arch`

xuanruiqi, 2022-04-07:
Latest git version seems to be working me just fine. Just did an update and it went quite smoothly.

closed 2022-04-07 by hughsie

ret2src, 2023-06-30:
Ran into the same problem today and opposed to what OP said above, `fwupd-git` did _not_ immediately work for me. However, I suspect my issue was different from OPs and had to do with the missing `shim` installation on my system. So, effectively, it's probably my own fault for not paying closer attention to the Arch wiki when installing `fwupd` in combination with Secure Boot enabled.

Nevertheless, I still want to share my solution here in case somebody else made the same mistake I made.
So, here's what I did to fix the problem:

1. Install `shim`:

    ```bash
    $ sudo pacman -S shim
    ```

2. Copy `shimx64.efi` to `/EFI/arch` (make sure the path is correct, depending on where your `EFI` folder is):

    ```bash
    $ sudo cp /usr/share/shim/shimx64.efi /boot/EFI/arch/
    ```

3. Copy `fwupd64.efi` to `grubx64.efi` (not sure if this is required, but as others suggested it, I did it _just in case_):

    ```bash
    $ sudo cp /boot/EFI/fwupdx64.efi /boot/EFI/grubx64.efi
    ```

4. **! IMPORTANT !** Make sure that all binaries are signed and sign them if necessary:

    ```bash
    $ sudo sbctl verify
    Verifying file database and EFI images in /boot...
    ✓ /boot/vmlinuz-linux is signed
    ✓ /boot/EFI/BOOT/BOOTX64.EFI is signed
    ✓ /boot/EFI/systemd/systemd-bootx64.efi is signed
    ✓ /boot/EFI/arch/fwupdx64.efi is signed
    ✓ /boot/EFI/arch/grubx64.efi is signed
    ✗ /boot/EFI/arch/shimx64.efi is not signed
    ```

    ```bash
    $ sudo sbctl sign /boot/EFI/arch/shimx64.efi
    ✓ Signed /boot/EFI/arch/shimx64.efi
    ```

    ```bash
    $ sudo sbctl verify                         
    Verifying file database and EFI images in /boot...
    ✓ /boot/EFI/BOOT/BOOTX64.EFI is signed
    ✓ /boot/EFI/systemd/systemd-bootx64.efi is signed
    ✓ /boot/vmlinuz-linux is signed
    ✓ /boot/EFI/arch/fwupdx64.efi is signed
    ✓ /boot/EFI/arch/grubx64.efi is signed
    ✓ /boot/EFI/arch/shimx64.efi is signed
    ```

5. Go ahead and do your `fwupd`'ing ;-)