fwupd/fwupd #3776: cannot update as CustomerID is unset

bulletmark, 2021-09-19:
I updated today and now see two new error messages but I have no idea what they mean or how I can fix them:
```
│   
├─VMM3200 inside VMM3200 inside VMM3200:
│     Device ID:          47a0754f6a2c9a7eb5f2883e4b78834a64451577
│     Summary:            Multi-stream transport device
│     Current version:    0.00.00
│     Vendor:             Synaptics (DRM_DP_AUX_DEV:0x06CB)
│     Update Error:       cannot update as CustomerID is unset
│     GUIDs:              f3518933-f152-50fc-8be8-0256181994ef ← MST-0
│                         d6072785-6fc0-5f83-9d49-11376e7f48b1 ← MST-leaf
│                         1beb64bd-6dc8-5db2-84b9-f32964abc876 ← MST-leaf-vmm3200-0
│                         b57f0a20-424c-51e0-b91b-575f3d74ed3a ← MST-leaf-0
│   
└─VMM3200 inside VMM3200 inside VMM3200:
      Device ID:          d7bfd19a68de57bb7de8a0cc8f7b97680677abc4
      Summary:            Multi-stream transport device
      Current version:    0.00.00
      Vendor:             Synaptics (DRM_DP_AUX_DEV:0x06CB)
      Update Error:       cannot update as CustomerID is unset
      GUIDs:              f3518933-f152-50fc-8be8-0256181994ef ← MST-0
                          d6072785-6fc0-5f83-9d49-11376e7f48b1 ← MST-leaf
                          1beb64bd-6dc8-5db2-84b9-f32964abc876 ← MST-leaf-vmm3200-0
                          b57f0a20-424c-51e0-b91b-575f3d74ed3a ← MST-leaf-0
```

The two "Update Error" messages above are new and shown in RED so presumably they are a problem which needs addressing. Everything else is show in standard terminal text.

Also, is that subtitle _VMM3200 inside VMM3200 inside VMM3200_ correct? Looks odd, and I don't recall seeing that before.

**Steps to Reproduce**

`sudo fwupdmgr get-devices`

**fwupd version information**
Arch Linux current package version 1.6.3-1 installed by pacman.

```
sudo fwupdmgr --version
client version:	1.6.3
compile-time dependency versions
	gusb:	0.3.7

daemon version:	1.6.3
```

**fwupd device information**

```
lt:~ sudo fwupdmgr get-devices --show-all-devices
XPS 13 9300
│
├─WD19TB:
│ │   Device ID:          570867f4ddda9915445da59bd85cd0e0e507270e
│ │   Summary:            High performance dock
│ │   Current version:    01.00.01.01
│ │   Vendor:             Dell Inc. (USB:0x413C)
│ │   Install Duration:   1 minute
│ │   Serial Number:      4DN5523/3159128927644594
│ │   GUID:               cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
│ │   Device Flags:       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │ 
│ ├─Package level of Dell dock:
│ │     Device ID:        073624c16dd99abe01ba1da223a70321e4f29beb
│ │     Summary:          A representation of dock update status
[... 184 lines cut ...]
│     Current version:    0.00.00
│     Vendor:             Synaptics (DRM_DP_AUX_DEV:0x06CB)
│     Update Error:       cannot update as CustomerID is unset
│     GUIDs:              f3518933-f152-50fc-8be8-0256181994ef ← MST-0
│                         d6072785-6fc0-5f83-9d49-11376e7f48b1 ← MST-leaf
│                         1beb64bd-6dc8-5db2-84b9-f32964abc876 ← MST-leaf-vmm3200-0
│                         b57f0a20-424c-51e0-b91b-575f3d74ed3a ← MST-leaf-0
│   
└─VMM3200 inside VMM3200 inside VMM3200:
      Device ID:          d7bfd19a68de57bb7de8a0cc8f7b97680677abc4
      Summary:            Multi-stream transport device
      Current version:    0.00.00
      Vendor:             Synaptics (DRM_DP_AUX_DEV:0x06CB)
      Update Error:       cannot update as CustomerID is unset
      GUIDs:              f3518933-f152-50fc-8be8-0256181994ef ← MST-0
                          d6072785-6fc0-5f83-9d49-11376e7f48b1 ← MST-leaf
                          1beb64bd-6dc8-5db2-84b9-f32964abc876 ← MST-leaf-vmm3200-0
                          b57f0a20-424c-51e0-b91b-575f3d74ed3a ← MST-leaf-0
```

**Additional questions**

- Operating system and version: Arch Linux fully updated daily.
- Have you tried rebooting? Yes
- Is this a regression? Don't know.

superm1, 2021-09-20:
A few questions:
1) Do you have anything besides the WD19TB connected, or just that device?  
2) It would be really useful if you could provide output for:

`sudo fwupdtool get-devices --plugins synaptics_mst --verbose`

3) Do you have a monitor connected to the dock?  If so what, monitor and what connection?
4) If you have a monitor connected and disconnect it does error go away?

bulletmark, 2021-09-20:
1. Only WDTBTB connected. Attached to the dock are two DP monitors, wired keyboard, usb RF mouse, ethernet, webcam, and 3.5 mm audio.
2. Output below.
3. The two monitors are DP connected Dell U2719D.
4. If I disconnect one monitor then one of those go away, disconnect the other and the second one also goes.

```
lt:~ sudo fwupdtool get-devices --plugins synaptics_mst --verbose
02:46:08:0865 FuDebug              Verbose debugging enabled (on console 1)
02:46:08:0866 FuMain               locked /run/lock/fwupdtool
02:46:08:0873 FuConfig             loading config values from /etc/fwupd/daemon.conf
02:46:08:0873 FuConfig             using autodetected max archive size 18.4 EB
02:46:08:0874 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
02:46:08:0874 FuRemoteList         loading /usr/share/fwupd/metainfo
02:46:08:0874 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
02:46:08:0874 XbSilo               file: 9bb1f137-1405-701f-9e9c-3eb41159f2d7, current:9bb1f137-1405-701f-9e9c-3eb41159f2d7, cached: (null)
02:46:08:0874 XbSilo               loading silo with file contents
02:46:08:0874 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
02:46:08:0874 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/lvfs-testing.conf.pacsave
02:46:08:0874 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
02:46:08:0875 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
02:46:08:0875 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
02:46:08:0875 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
02:46:08:0875 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/lvfs-testing.conf.pacnew
02:46:08:0875 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
02:46:08:0875 FuRemoteList         ignoring unfound remote fwupd
02:46:08:0875 FuRemoteList         ordering lvfs-testing=lvfs+1
02:46:08:0875 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
02:46:08:0875 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
02:46:08:0875 Jcat                 loaded 1 certificates
02:46:08:0875 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
02:46:08:0875 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
02:46:08:0875 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
02:46:08:0875 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
02:46:08:0875 Jcat                 loaded 1 certificates
02:46:08:0875 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
02:46:08:0875 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
02:46:08:0875 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
02:46:08:0875 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
02:46:08:0881 FuEngine             client certificate exists and working
02:46:08:0881 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
02:46:08:0882 FuHistory            got schema version of 6
02:46:08:0882 FuSmbios             trying to read /sys/class/dmi/id
02:46:08:0882 FuHwids              smbios property Manufacturer=Dell Inc.
02:46:08:0882 FuHwids              smbios property EnclosureKind=a
02:46:08:0882 FuHwids              smbios property Family=XPS
02:46:08:0882 FuHwids              smbios property ProductName=XPS 13 9300
02:46:08:0882 FuHwids              smbios property ProductSku=096D
02:46:08:0882 FuHwids              smbios property BiosVendor=Dell Inc.
02:46:08:0882 FuHwids              smbios property BiosVersion=1.7.0
02:46:08:0882 FuHwids              smbios property BiosMajorRelease=01
02:46:08:0882 FuHwids              smbios property BiosMinorRelease=07
02:46:08:0882 FuHwids              smbios property FirmwareMajorRelease=ff
02:46:08:0882 FuHwids              smbios property FirmwareMinorRelease=ff
02:46:08:0882 FuHwids              smbios property BaseboardManufacturer=Dell Inc.
02:46:08:0882 FuHwids              smbios property BaseboardProduct=077Y9N
02:46:08:0882 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
02:46:08:0882 XbSilo               file: 29ab557a-cd00-5daa-bb7a-a3a34a12c41f, current:29ab557a-cd00-5daa-bb7a-a3a34a12c41f, cached: (null)
02:46:08:0882 XbSilo               loading silo with file contents
02:46:08:0883 FuEngine             1331 components now in silo
02:46:08:0892 FuPlugin             init(/usr/lib/fwupd-plugins-3/libfu_plugin_synaptics_mst.so)
02:46:08:0892 FuContext            added udev subsystem watch of drm
02:46:08:0892 FuContext            added udev subsystem watch of drm_dp_aux_dev
02:46:08:0892 FuEngine             plugins disabled: modem_manager, lenovo_thinklmi, linux_sleep, nitrokey, jabra, uefi_recovery, vli, acpi_phat, redfish, altos, hailuck, uefi_capsule, linux_swap, cros_ec, fastboot, thelio_io, flashrom, dell, emmc, linux_lockdown, uefi_pk, synaptics_rmi, synaptics_cxaudio, bcm57xx, elantp, optionrom, intel_spi, powerd, pixart_rf, system76_launch, ccgx, ebitdo, analogix, dell_esrt, realtek_mst, parade_lspcon, tpm_eventlog, bios, cpu, superio, tpm, iommu, acpi_facp, fresco_pd, pci_mei, amt, wac [... cut ...]
  Protocol:             com.synaptics.mst
  Flags:                updatable|require-ac|supported|registered|dual-image|usable-during-update|skips-restart
  Vendor:               Dell Inc.
  VendorId:             DRM_DP_AUX_DEV:0x06CB
  Version:              5.04.06
  VersionFormat:        triplet
  Icon:                 video-display
  InstallDuration:      95
  Created:              2021-09-20
  UpdateMessage:        There is no change.

02:46:11:0567 FuPlugin             destroy(synaptics_mst)
```

superm1, 2021-09-20:
OK, glad it's root caused where they're coming from.  There are a variety of things going on here.
1. The names aren't built properly.  This "inside VMM..." is wrong.  I think it's a mistake that happened during a plugin rewrite a while back (as far as f878a7bcba1cbac53af3f265f1afd694d4649fe4) but didn't get noticed until your issue was reported.
2. The device is missing a customer ID.  This means you can't target firmware at it safely.  It might be Dell's intention, it might be a mistake, but whatever the case is - it's not usable by fwupd in this circumstance like normal firmware updates.
3. The device shows up at all when CustomerID isn't set.  The error is entirely accurate, but it's not usable to regular people.  What we usually do is set that flag to let someone know an error they might be able to do something about, but they really can't do anything here.  So we have two options.  Either reject the device probe, or don't show the error.  The thing is if the customer ID isn't set potentially `fwupdtool install-blob` might be usable to recover, but if you rejected it during probe that goes away.

I've filed #3777 to fix what I could.

commit 24351dd9 (2021-09-20): synaptics-mst: fix naming error when device has a parent

commit f878a7bcba1cbac53af3f265f1afd694d4649fe4 made a mistake to
take the name of the device not the parent.

Fixes #3776
commit 472bfb91 (2021-09-20): synaptics-mst: don't set update error when customer ID not set

Users can't do anything about this.  This will prevent it showing
up in fwupdtool/fwupdmgr output by default.

Fixes: #3776
commit 24c4b6d8 (2021-09-20): synaptics-mst: inhibit updates when customer ID not set

Users can't do anything about this.  This will prevent it showing
up in fwupdtool/fwupdmgr output by default.

Fixes: #3776
commit cc2b9a32 (2021-09-20): synaptics-mst: fix naming error when device has a parent

commit f878a7bcba1cbac53af3f265f1afd694d4649fe4 made a mistake to
take the name of the device not the parent.

Fixes #3776
closed 2021-09-20 by superm1