fwupd/fwupd #3778: --ignore-power does not work for inhibited devices

djcampello, 2021-09-20:
The flag is not processed as the update is inhibited in the first place: https://github.com/fwupd/fwupd/blob/HEAD/src/fu-engine.c#L282

hughsie, 2021-09-21:
@superm1 would you be against adding this as a daemon.conf option? I'm not 100% happy about users disabling this at runtime (as the UEFI update will typically *fail* anyway, as the updater binary will verify the AC state with the EC) but I agree we need some kind of escape hatch for development.

superm1, 2021-09-21:
If it's for development only can't you just use the plugin allowlist to prevent that plugin causing inhibition to run?

superm1, 2021-09-21:
Although I see the point - what's the point of `--ignore-power` if it doesn't actually do anything since we inhibit devices for updates.  So this should probably be either:
1. Remove `--ignore-power` from frontend.  If you need to do this for development then disable `upower`/`powerd` from `daemon.conf` or `--plugins=`.
2. Change how `--ignore-power` works to let it work for inhibited devices.

My inclination is 1.

hughsie, 2021-09-21:
> My inclination is 1.

Me also.

superm1, 2021-09-21:
The problem with tearing it out though - we have exported it to the client.  So is it worth breaking `libfwupd` ABI over this?  granted I know ABI is breaking for `libfwupdplugin` in 1.7.0.  I am leaning on if we tear it out we need to mark it deprecated instead and have the daemon return something about it being an unsupported command now or something.

commit 02b28807 (2021-09-21): Remove support for `--ignore-power` by frontends

This functionality broke a number of releases ago as part of
implementing device inhibition and was just noticed now.

Instead of fixing it, the preference seems to be to remove the
functionality as it exists today as inhibitions can happen for
a number of reasons.

To still allow people to override these power warnings (such as during
development) add a new daemon configuration item that can be used.

Fixes: #3778
commit bd7c72d9 (2021-09-21): Remove support for `--ignore-power` by frontends

This functionality broke a number of releases ago as part of
implementing device inhibition and was just noticed now.

Instead of fixing it, the preference seems to be to remove the
functionality as it exists today as inhibitions can happen for
a number of reasons.

To still allow people to override these power warnings (such as during
development) add a new daemon configuration item that can be used.

Fixes: #3778
commit 3c2faa71 (2021-09-21): Remove support for `--ignore-power` by frontends

This functionality broke a number of releases ago as part of
implementing device inhibition and was just noticed now.

Instead of fixing it, the preference seems to be to remove the
functionality as it exists today as inhibitions can happen for
a number of reasons.

To still allow people to override these power warnings (such as during
development) add a new daemon configuration item that can be used.

Fixes: #3778
PR #3781 (2021-09-21): Remove support for `--ignore-power` by frontends
This functionality broke a number of releases ago as part of
implementing device inhibition and was just noticed now.

Instead of fixing it, the preference seems to be to remove the
functionality as it exists today as inhibitions can happen for
a number of reasons.

To still allow people to override these power warnings (such as during
development) add a new daemon configuration item that can be used.

Fixes: #3778

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-check [... cut]
closed 2021-09-21 by superm1