fwupd/fwupd #3785: fwupdmgr --get-releases prints results when it should not

djcampello, 2021-09-23:
fwupdmgr get-releases <wrong_id> prints error like `Invalid Arguments` or `failed to find any device providing <wrong_id>` but then proceeds to show releases for other devices in the system...

This causes issues if you do something like `fwupdmgr get-releases | grep Uri` and use the url with `fwupdmgr install <uri>`

The root cause is in https://github.com/fwupd/fwupd/blob/master/src/fu-util.c#L1336 by falling through instead of returning NULL

hughsie, 2021-09-23:
That looks kinda deliberate, although I agree the logic looks wonky. @superm1 would you agree that returning the `error` is the appropriate thing to do here?

superm1, 2021-09-24:
I'm pretty sure the previous logic is intentional - you put in a wrong device ID, and it's like "oh that's a bad device" so it shows you valid ones to pick from instead.  But I can certainly see it the other way too making sense.  No strong opinion that it should stay this way.

djcampello, 2021-09-24:
I think is just odd:

```
# fwupdmgr get-releases -v b585990a-003e-5270-89d5-3705a17f9a4
(fwupdmgr:17152): GLib-DEBUG: 08:20:34.781: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:17152): GLib-GIO-DEBUG: 08:20:34.781: _g_io_module_get_default: Found default implementation dummy (GDummyProxyResolver) for ?gio-proxy-resolver?
(fwupdmgr:17152): Fwupd-DEBUG: 08:20:34.783: Emitting ::status-changed() [idle]
WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
Invalid arguments
Selected device: Integrated Webcam?
  AppstreamId:          org.fwupd.fakedevice.firmware
  RemoteId:             fwupd-tests
  Summary:              Firmware for the ACME Corp Integrated Webcam
  Description:          <p>Fixes another bug with the flux capacitor to prevent time going backwards.</p>
  Version:              1.2.4
  Filename:             fakedevice124.cab
  Checksum:             SHA1(cffec02f8094d10552b68967425613d72d72ac02)
  Checksum:             SHA256(3fab34cfa1ef97238fb24c5e40a979bc544bb2b0967b863e43e7d58e0d9a923f)
  License:              GPL-2.0+
  Size:                 10 bytes
  Created:              2017-06-15
  Uri:                  https://storage.googleapis.com/chromeos-localmirror/lvfs/test/3fab34cfa1ef97238fb24c5e40a979bc544bb2b0967b863e43e7d58e0d9a923f-fakedevice124.cab
  Homepage:             http://www.acme.com/
  Urgency:              medium
  Vendor:               ACME Corp
  Flags:                is-upgrade

```

Is not consistent with other operations like `update`:

```
# fwupdmgr update -v b585990a-003e-5270-89d5-3705a17f9a4
(fwupdmgr:17340): GLib-DEBUG: 08:22:08.572: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:17340): GLib-GIO-DEBUG: 08:22:08.573: _g_io_module_get_default: Found default implementation dummy (GDummyProxyResolver) for ?gio-proxy-resolver?
(fwupdmgr:17340): Fwupd-DEBUG: 08:22:08.574: Emitting ::status-changed() [idle]
WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
Invalid arguments

Use fwupdmgr --help for help
```

djcampello, 2021-09-24:
Also, since the GUID is not even part of the output makes it very prone to errors to the unsuspecting user that does not notice the `Invalid arguments` line.

hughsie, 2021-09-24:
I agree it's inconsistent. Any chance you could do a PR to fix it up to behave like the other commands please?

djcampello, 2021-09-24:
Sure, I'll send it soon

superm1, 2021-09-24:
fixed by https://github.com/fwupd/fwupd/pull/3793

closed 2021-09-24 by superm1