fwupd/fwupd #3786: Release branch detected incorrectly

Sean-StarLabs, 2021-09-23:
fwupd detects AMI firmware as coreboot branch, so when you try and switch you get `Device StarBook is already on branch coreboot`.

I'm not sure where that's coming from, so not sure how to fix!

```
├─StarBook:
│ │   Device ID:          dbee8bd3b1ae0316ad143336155651eedb495a0e
│ │   Current version:    1.0.00
│ │   Vendor:             Star Labs (DMI:American Megatrends International, LLC.)
│ │   Release Branch:     coreboot
│ │   GUIDs:              230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       7c13a5b2-22f8-5564-8b48-4bbf20da8ed6
│ │                       e327ada1-c062-536b-a1bd-3dcfca48e390
│ │                       b7a2bed5-186f-5643-a96a-0ac51de16c41
│ │                       372a7618-7117-5c3d-b0ea-a67d4e351893
│ │                       612d0af2-6b4b-5d1a-bd63-433022fdaefe
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Device supports switching to a different branch of firmware

```

[coreboot](https://fwupd.org/lvfs/firmware/8422)
[AMI](https://fwupd.org/lvfs/firmware/8427)

hughsie, 2021-09-24:
I think the only way we set the branch is using a quirk match. e.g. 

    # StarBook Mk V (coreboot GUID)
    [7c13a5b2-22f8-5564-8b48-4bbf20da8ed6]
    Branch = coreboot
    VersionFormat = number
    Flags = reset-cmos

I don't know what `7c13a5b2-22f8-5564-8b48-4bbf20da8ed6` is, but it seems to be set when running your AMI firmware...

Sean-StarLabs, 2021-09-27:
Needed to use the one that has the firmware vendor in!

closed 2021-09-27 by Sean-StarLabs