fwupd/fwupd #3790: 1.6.4: test suite is failing

kloczek, 2021-09-24:
**Describe the bug**
Test suite is failing in two units.

**Steps to Reproduce**
I'm building rom package so normal steps:
- meson with below setup:
```spec
%meson \
        -D agent=true \
        -D build=all \
        -D consolekit=true \
        -D firmware-packager=true \
        -D gpg=true \
        -D gtkdoc=true \
        -D gudev=true \
        -D introspection=true \
        -D lvfs=true \
        -D man=true \
        -D pkcs7=true \
        -D plugin_altos=true \
        -D plugin_amt=true \
        -D plugin_coreboot=true \
        -D plugin_dell=true \
        -D plugin_dummy=true \
        -D plugin_emmc=true \
        -D plugin_flashrom=true \
        -D plugin_modem_manager=true \
        -D plugin_msr=true \
        -D plugin_nvme=false \
        -D plugin_nvme=true \
        -D plugin_redfish=true \
        -D plugin_synaptics=true \
        -D plugin_thunderbolt=true \
        -D plugin_uefi_capsule=true \
        -D plugin_uefi_pk=true \
        -D supported_build=true \
        -D systemd=true \
        -D tests=true \
        -D tpm=true \
        %{nil}
```
- meson build
- meson install
- meson test

**Expected behavior**
test suite should not fail

**fwupd version information**
1.6.4

**Additional questions**

- Operating system and version:
Linux/x86_64
- Have you tried rebooting?
N/A
- Is this a regression?
In the same env 1.6.3 test suyite passes without issues.

Test suite output:
```console
+ cd fwupd-1.6.4
+ /usr/bin/meson test -C x86_64-redhat-linux-gnu --num-processes 48 --print-errorlogs
ninja: Entering directory `/home/tkloczko/rpmbuild/BUILD/fwupd-1.6.4/x86_64-redhat-linux-gnu'
ninja: no work to do.
 1/32 fwupd-exported-api                     OK               0.05s
 2/32 fwupdplugin-exported-api               OK               0.02s
 3/32 fwupd-context-test                     OK               2.92s
 4/32 fwupd-thread-test                      OK               2.99s
[... 225 lines cut ...]
#   <vendor_id>0x8086</vendor_id>
#   <model_id>0x1234</model_id>
#   <generation>0x3</generation>
#   <ports>0x2</ports>
#   <has_pd>true</has_pd>
#   <section>1</section>
#   <section>6</section>
#   <section>1</section>
#   <section>2c</section>
# </firmware>
#
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――

Summary of Failures:

 7/32 fu-self-test                   FAIL             1.62s   killed by signal 6 SIGABRT
32/32 thunderbolt-self-test          TIMEOUT        120.09s   killed by signal 15 SIGTERM

Ok:                 30
Expected Fail:      0
Fail:               1
Unexpected Pass:    0
Skipped:            0
Timeout:            1
```

hughsie, 2021-09-24:
> Unable to find default local file monitor type

Are you somehow compiling glib without something like inotify?

kloczek, 2021-09-24:
Yep I;m building it with `fam=false`.
After detailed scan all nobarirsd ABIs I dould that none of the projests is using that part.
After +4 years using such configuration looks like fwupd wouls be first one.
I've enev started talking with glib maintainers about remove of that part from glib showing them that nothing is using that :)

hughsie, 2021-09-24:
Sure, we don't need `fam` -- but you should have fallen back to the inotify support. Can you share your glib meson configure output?

kloczek, 2021-09-24:
Sure of course
```spec
%bcond_with     asserts
%bcond_without  checks
%bcond_with     debug

[..]

%build
%meson \
        -D dtrace=false \
        -D fam=false \
        -D glib_assert=%{?with_asserts:true}%{!?with_asserts:false} \
        -D glib_checks=%{?with_checks:true}%{!?with_checks:false} \
        -D glib_debug=%{?with_debug:en}%{!?with_debug:dis}abled \
        -D gtk_doc=true \
        -D installed_tests=true \
        -D internal_pcre=false \
        -D libmount=enabled \
        -D man=true \
        -D nls=enabled \
        -D selinux=enabled \
        -D systemtap=true \
        -D xattr=true \
        %{nil}
```
So I'm using glib without debug and without assertions which deactivates all `g_assert()`  code.
There is separated set of assertions which is for test suites https://docs.gtk.org/glib/func.assert.html
Just checked your code ..
```console
[tkloczko@barrel fwupd-1.6.3]$ grep -r g_assert -w | wc -l
293
[tkloczko@barrel fwupd-1.6.3]$ grep -r g_assert -wl| grep test
libfwupd/fwupd-context-test.c
libfwupd/fwupd-self-test.c
libfwupd/fwupd-thread-test.c
libfwupdplugin/fu-self-test.c
plugins/ata/fu-self-test.c
plugins/dell/fu-self-test.c
plugins/dfu/fu-dfu-self-test.c
plugins/lenovo-thinklmi/fu-self-test.c
plugins/nvme/fu-self-test.c
plugins/optionrom/fu-self-test.c
plugins/redfish/fu-self-test.c
plugins/synaptics-mst/fu-self-test.c
src/fu-self-test.c
```
so more likely it is about wrong using asserttions if any of those files contains test suite code.

From: https://docs.gtk.org/glib/testing.html
In test suites definitelly as you see in glib documentation is that `g_assert()` should not be used and there are other dedicated `g_assert_*()` macros for that.
```
In addition to the traditional g_assert_true(), the test framework provides
an extended set of assertions for comparisons: g_assert_cmpfloat(),
g_assert_cmpfloat_with_epsilon(), g_assert_cmpint(), g_assert_cmpuint(),
g_assert_cmphex(), g_assert_cmpstr(), g_assert_cmpmem() and
g_assert_cmpvariant().  The advantage of these variants over plain
g_assert_true() is that the assertion messages can be more elaborate, and
include the values of the compared entities.

Note that g_assert() should not be used in unit tests, since it is a no-op
when compiling with G_DISABLE_ASSERT.  Use g_assert() in production code,
and g_assert_true() in unit tests.
```

Just in case .. I'm preparing to start building all glib using programs with `-DG_DISABLE_ASSERT`

hughsie, 2021-09-24:
No, I meant the output, of meson configure on a real system, not the arguments. You're correct about the fact I shouldn't be using g_assert() in self test code, but that's not the cause of the failure you're seeing here.

kloczek, 2021-09-24:
Just checked logs of the glib build
```console
+ /usr/bin/meson --buildtype=plain --prefix=/usr --libdir=/usr/lib64 --libexecdir=/usr/libexec --bindir=/usr/bin --sbindir=/usr/sbin --includedir=/usr/include --datadir=/usr/share --mandir=/usr/share/man --infodir=/usr/share/info --localedir=/usr/share/locale --sysconfdir=/etc --localstatedir=/var --sharedstatedir=/var/lib --wrap-mode=nodownload --auto-features=enabled . x86_64-redhat-linux-gnu -D dtrace=false -D fam=false -D glib_assert=false -D glib_checks=true -D glib_debug=disabled -D gtk_doc=true -D installed_tests=true -D internal_pcre=false -D libmount=enabled -D man=true -D nls=enabled -D selinux=enabled -D systemtap=true -D xattr=true
The Meson build system
Version: 0.59.1
Source dir: /home/tkloczko/rpmbuild/BUILD/glib-2.70.0
Build dir: /home/tkloczko/rpmbuild/BUILD/glib-2.70.0/x86_64-redhat-linux-gnu
Build type: native build
WARNING: Unknown options: "internal_pcre"
The value of new options can be set with:
meson setup <builddir> --reconfigure -Dnew_option=new_value ...
Project name: glib
Project version: 2.70.0
C compiler for the host machine: /usr/bin/gcc (gcc 11.2.1 "gcc (GCC) 11.2.1 20210728 (Red Hat 11.2.1-1)")
C linker for the host machine: /usr/bin/gcc ld.bfd 2.37-10
C++ compiler for the host machine: /usr/bin/g++ (gcc 11.2.1 "g++ (GCC) 11.2.1 20210728 (Red Hat 11.2.1-1)")
C++ linker for the host machine: /usr/bin/g++ ld.bfd 2.37-10
Host machine cpu family: x86_64
Host machine cpu: x86_64
Compiler for C supports arguments -fno-strict-aliasing: YES
Checking if "GNU C visibility attributes test" compiles: YES
Compiler for C supports arguments -fvisibility=hidden: YES
Message: Disabling GLib asserts
Has header "alloca.h" : YES
Has header "crt_externs.h" : NO
Has header "dirent.h" : YES
Has header "float.h" : YES
Has header "fstab.h" : YES
Has header "grp.h" : YES
Has header "inttypes.h" : YES
Has header "limits.h" : YES
Has header "linux/magic.h" : YES
Has header "locale.h" : YES
Has header "mach/mach_time.h" : NO
Has header "memory.h" : YES
Has header "mntent.h" : YES
Has header "poll.h" : YES
Has header "pwd.h" : YES
Has header "sched.h" : YES
Has header "spawn.h" : YES
Has header "stdatomic.h" : YES
Has header "stdint.h" : YES
Has header "stdlib.h" : YES
Has header "string.h" : YES
Has header "strings.h" : YES
Has header "sys/auxv.h" : YES
Has header "sys/event.h" : NO
Has header "sys/filio.h" : NO
Has header "sys/inotify.h" : YES                  <<<< HERE
Has header "sys/mkdev.h" : NO
Has header "sys/mntctl.h" : NO
Has header "sys/mnttab.h" : NO
Has header "sys/mount.h" : YES
Has header "sys/param.h" : YES
Has header "sys/resource.h" : YES
Has header "sys/select.h" : YES
Has header "sys/statfs.h" : YES
Has header "sys/stat.h" : YES
Has header "sys/statvfs.h" : YES
Has header "sys/sysctl.h" : NO
Has header "sys/time.h" : YES
Has header "sys/times.h" : YES
Has header "sys/types.h" : YES
Has header "sys/uio.h" : YES
Has header "sys/vfs.h" : YES
Has header "sys/vfstab.h" : NO
Has header "sys/vmount.h" : NO
Has header "sys/wait.h" : YES
Has header "termios.h" : YES
Has header "unistd.h" : YES
[... 565 lines cut ...]
Configuring version.xml using configuration
Program gtkdoc-scan found: YES (/usr/bin/gtkdoc-scan)
Program gtkdoc-scangobj found: YES (/usr/bin/gtkdoc-scangobj)
Program gtkdoc-mkdb found: YES (/usr/bin/gtkdoc-mkdb)
Program gtkdoc-mkhtml found: YES (/usr/bin/gtkdoc-mkhtml)
Program gtkdoc-fixxref found: YES (/usr/bin/gtkdoc-fixxref)
Program gtkdoc-check found: YES (/usr/bin/gtkdoc-check)
Configuring gtkdocentities.ent using configuration
Configuring version.xml using configuration
Program gtkdoc-scan found: YES (/usr/bin/gtkdoc-scan)
Program gtkdoc-scangobj found: YES (/usr/bin/gtkdoc-scangobj)
Program gtkdoc-mkdb found: YES (/usr/bin/gtkdoc-mkdb)
Program gtkdoc-mkhtml found: YES (/usr/bin/gtkdoc-mkhtml)
Program gtkdoc-fixxref found: YES (/usr/bin/gtkdoc-fixxref)
Program gtkdoc-check found: YES (/usr/bin/gtkdoc-check)
Build targets in project: 425
WARNING: Project specifies a minimum meson_version '>= 0.49.2' but uses features which were added in newer versions:
 * 0.50.0: {'Python module path method'}
 * 0.52.0: {'check arg in gnome.gtkdoc'}

Option buildtype is: plain [default: debugoptimized]
Found ninja-1.10.2 at /usr/bin/ninja
```
So as you see inotity checks are with `YES` results.
In other words inotify has been detected .. still no root cause :/

kloczek, 2021-09-24:
I'm preserwing build trees after build of each package so I was able to check config.h content
```console
[tkloczko@barrel x86_64-redhat-linux-gnu]$ grep INOTIFY config.h
#define HAVE_INOTIFY_INIT1 1
#define HAVE_SYS_INOTIFY_H 1
```

hughsie, 2021-09-27:
> Has header "sys/inotify.h" : YES                  <<<< HERE

Hmm, I didn't expect that. So the thing is, we can't just do a `g_debug("directory watches not supported")` or something, as fwupd actually expects to get a notification if a remote gets manually created or deleted.

kloczek, 2021-09-27:
OK, so how can I try to trace that?

hughsie, 2021-09-27:
g_local_file_monitor_new() is falling through to `Unable to find default local file monitor type` -- so it would be good to breakpoint on that, and confirm:

* `is_remote_fs` is FALSE
* `is_directory` is TRUE

then if that's correct, why G_LOCAL_FILE_MONITOR_EXTENSION_POINT_NAME isn't being registered. I'm assuming you've not set `GIO_USE_FILE_MONITOR` or something manually?

kloczek, 2021-09-27:
Nope ..
That is all what I have in build procedure:
```spec
%build
%meson \
        -D agent=true \
        -D build=all \
        -D consolekit=true \
        -D firmware-packager=true \
        -D gpg=true \
        -D gtkdoc=false \
        -D gudev=true \
        -D introspection=true \
        -D lvfs=true \
        -D man=true \
        -D pkcs7=true \
        -D plugin_altos=true \
        -D plugin_amt=true \
        -D plugin_coreboot=true \
        -D plugin_dell=true \
        -D plugin_dummy=true \
        -D plugin_emmc=true \
        -D plugin_flashrom=true \
        -D plugin_modem_manager=true \
        -D plugin_msr=true \
        -D plugin_nvme=false \
        -D plugin_nvme=true \
        -D plugin_redfish=true \
        -D plugin_synaptics=true \
        -D plugin_thunderbolt=true \
        -D plugin_uefi_capsule=true \
        -D plugin_uefi_pk=true \
        -D supported_build=true \
        -D systemd=true \
        -D tests=true \
        -D tpm=true \
        %{nil}
%meson_build

%install
%meson_install

%__mkdir_p --mode=0700 %{buildroot}%{_localstatedir}/lib/fwupd/gnupg

%find_lang %{name}

%check
%meson_test
```

kloczek, 2021-09-27:
Hmm .. just found that `plugin_nvme=false` probably is not right so I've changed that to `plugin_nvme=true` and sent to rebuild and test suite is OK.
```console
Executing(%check): /bin/sh -e /var/tmp/rpm-tmp.juHUjN
+ umask 022
+ cd /home/tkloczko/rpmbuild/BUILD
+ cd fwupd-1.6.4
+ /usr/bin/meson test -C x86_64-redhat-linux-gnu --num-processes 48 --print-errorlogs
ninja: Entering directory `/home/tkloczko/rpmbuild/BUILD/fwupd-1.6.4/x86_64-redhat-linux-gnu'
ninja: no work to do.
 1/32 fwupd-exported-api                     OK               0.08s
 2/32 fwupdplugin-exported-api               OK               0.02s
 3/32 fwupd-context-test                     OK               2.78s
 4/32 fwupd-thread-test                      OK               2.86s
 5/32 fwupd-self-test                        OK               2.89s
 6/32 fwupdplugin-self-test                  OK               2.44s
 7/32 fu-self-test                           OK               2.72s
 8/32 acpi-dmar-self-test                    OK               0.39s
 9/32 acpi-facp-self-test                    OK               0.38s
10/32 acpi-phat-self-test                    OK               0.37s
11/32 ata-self-test                          OK               0.35s
12/32 bcm57xx-self-test                      OK               0.34s
13/32 ccgx-self-test                         OK               0.33s
14/32 dell-self-test                         OK               0.32s
15/32 fu-dfu-self-test                       OK               0.30s
16/32 elantp-self-test                       OK               0.29s
17/32 linux-swap-self-test                   OK               0.26s
18/32 logitech-hidpp-self-test               OK               0.25s
19/32 nitrokey-self-test                     OK               0.24s
20/32 nvme-self-test                         OK               0.23s
21/32 pxi-self-test                          OK               0.22s
22/32 synaptics-mst-self-test                OK               0.18s
23/32 synaptics-prometheus-self-test         OK               0.17s
24/32 synaptics-rmi-self-test                OK               0.16s
25/32 tpm-eventlog-self-test                 OK               0.13s
26/32 uefi-dbx-self-test                     OK               0.10s
27/32 vli-self-test                          OK               0.08s
28/32 lenovo-thinklmi-self-test              OK               0.29s
29/32 wacom-usb-self-test                    OK               0.07s
30/32 redfish-self-test                      OK               0.21s
31/32 uefi-self-test                         OK               0.16s
32/32 thunderbolt-self-test                  OK               3.92s

Ok:                 32
Expected Fail:      0
Fail:               0
Unexpected Pass:    0
Skipped:            0
Timeout:            0

Full log written to /home/tkloczko/rpmbuild/BUILD/fwupd-1.6.4/x86_64-redhat-linux-gnu/meson-logs/testlog.txt
```
However swiching back to `plugin_nvme=false` did not triggered test suite fail.
All what I've changed in mean time was kernel version (from 5.13.16 to 5.15.0-0.rc2) .. hmm 🤔

kloczek, 2021-09-27:
OK. Closing.
Nevertheless that case points that something fishy may be around kernel (what possibly has been fixed in meantime).
Thx for your time.

closed 2021-09-27 by kloczek

vrusinov, 2021-12-05:
I saw the [same error in actual system](https://bugzilla.redhat.com/show_bug.cgi?id=2029179) - not during build/test.

It turned out the system was short on inotify files. I was able to resolve the issue by bumping `fs.inotify.*` up. In particular my system needed `sudo sysctl fs.inotify.max_user_instances=512`.

So the above test failure is likely environmental too, and it makes sense that changing kernel version (which would have required reboot) changed environment/host workload.

One thing fwupd can do better is is to provide a better error message. I was able to figure this out once _something else_ popped "Failed to allocate directory watch: Too many open files" message.

vrusinov, 2021-12-05:
> One thing fwupd can do better is is to provide a better error message

Pardon, the error is actually coming from glib: https://github.com/frida/glib/blob/7dba767c26c081ff56a38827800faa632eec2c87/gio/glocalfilemonitor.c#L858

So it doesn't look like fwupd can do anything here, and it may not be easy to surface actual error in glib - I looked at 
g_local_file_monitor_new + _g_io_module_get_default_type are not really structured for this.

hughsie, 2021-12-05:
Do you think fwupd should check the number of inotify watches at startup? In this case it would have made it much easier to debug. In which case, what API should fwupd use?

vrusinov, 2021-12-05:
> Do you think fwupd should check the number of inotify watches at startup?

Not sure if additional complexity/maintenance for this specific and likely rare problem. Dunno about APIs either - I'm not really familiar with this topic.

I'd say best place to fix this is in glib: it really needs to give something more specific than "unable to find", and I don't think it's impossible to do. Just needs some time and/or familiarity with the code/APIs/target platforms.

vrusinov, 2021-12-05:
FYI - I was unable to report it to glib because it is hosted in Gnome's gitlab, which does not let people with gmail.com accounts to sign up and I've lost my Sunday night motivation at that.

Maybe later.

superm1, 2021-12-06:
> Not sure if additional complexity/maintenance for this specific and likely rare problem is worth it. Dunno about APIs either - I'm not really familiar with this topic.

It seems like maybe just doing the following may be a quick check:

0. (Only run this code on Linux)
1. `inotify_add_watch` on something that fwupd already owns and check the return code
2. If the return code is `ENOSPC` then show a clean error for the user.
3. `inotify_rm_watch` on that something from "1"

hughsie, 2021-12-06:
> inotify_add_watch

But it could be that it's not the 1st `inotify_add_watch()` call -- it could be the 9th that fails...

superm1, 2021-12-06:
> But it could be that it's not the 1st inotify_add_watch() call -- it could be the 9th that fails...

But if we do it on fail path - we can a clean error message.

hughsie, 2021-12-06:
Ahh indeed. It seems easier than I feared, `sudo sysctl fs.inotify.max_user_instances=25` then running the daemon, it's inotify_init() that fails, not inotify_add_watch() -- will prototype something