fwupd/fwupd #3799: Rename the master branch to main

hughsie, 2021-09-24:
Following most of my other projects, I'm thinking we should rename the `master` branch to `main`.  I think we may have to sacrifice a small child to the CI gods, but it's probably worth doing.

hughsie, 2021-09-24:
I've pushed wip/hughsie/master-to-main but we probably want to change the name on github before opening a PR and running all the CI on that.

superm1, 2021-09-24:
How soon are you wanting to do a 1.7.x release?  If imminently can we perhaps do the rename after 1.7.0 is tagged?   I'm fully expecting that there will be things that break that will be non-obvious.

superm1, 2021-09-24:
Also, very important - how does it work on github for a branch rename?  Does `https://github.com/fwupd/fwupd/blob/master/foo` redirect to `https://github.com/fwupd/fwupd/blob/main/foo`?  If not, we're going to have to fix up anything in LVFS that references files, anything on fwupd.github.io that does... any wiki pages that do..  It's gonna be a lot.

superm1, 2021-09-24:
https://github.com/github/renaming
Ah according to this there are redirects.  That's not as concerning then to me.

> Renaming a branch will:
> 
> Re-target any open pull requests
> Update any draft releases based on the branch
> Move any branch protection rules that explicitly reference the old name
> Update the branch used to build GitHub Pages, if applicable
> Show a notice to repository contributors, maintainers, and admins on the repository homepage with instructions to update local copies of the repository
> Show a notice to contributors who git push to the old branch
> **Redirect web requests for the old branch name to the new branch name**
> Return a "Moved Permanently" response in API requests for the old branch name
>

hughsie, 2021-09-24:
> How soon are you wanting to do a 1.7.x release

That's a good question. It's certainly not imminent, although I can hear murmurs of disquiet from various people who have plugins in master then want to get to end-users. I am enjoying having a master branch we can make big changes in tho.

>  Does https://github.com/fwupd/fwupd/blob/master/foo redirect to https://github.com/fwupd/fwupd/blob/main/foo

I don't know. I would hope so. If not, I wonder if we can make master almost like a tag, with a big warning in the README.md that development carried on in main.

> If not, we're going to have to fix up anything in LVFS that references files

That's a small fsck action, no biggie.

superm1, 2021-09-24:
>That's a good question. It's certainly not imminent, although I can hear murmurs of disquiet from various people who have plugins in master then want to get to end-users. I am enjoying having a master branch we can make big changes in tho.

OK in that case let's get this done and get figure out any fallout.

One more thing that comes to mind - do we want any sort of detection in `contrib/setup` for old checkouts to clean things up?  Or do we just hope the Github redirects DTRT?
If they don't then we can ask anyone with a checkout to run `./contrib/setup` again.  I guess we might not know until we merge.

closed 2021-09-24 by superm1