fwupd/fwupd #3805: Absolute Persistence a.k.a The Absolute Backdoor

johannbg, 2021-09-26:
**Is your feature request related to a problem? Please describe.**
Noticed when I was updating the firmware for a security vulnerability via cab files downloaded from the Lenovo side since update was not availabe or being picked up by fwupd yet.

Lenovo provides two cab files for bios update, one update for the `Absolute Persistence Module` if it's enabled and another one for `Non-Absolute-Persistant-Module` encase it has been disabled. 

Now my request is somewhat two fold on one hand the output for the `System Firmware` section and other relevant places should indicate if the system firmware supports absolute persistance and if it's enabled since currently there is no indicator if it's available and enabled/disabled and when you run `fwupdmgr get-details` on both of the provide cab files from Lenovo one get's listed as `Unknown Device` while the other show the `System Firmware` section.

The second part of the RFE is to provide end users the ability to permanently disable the Absolute backdoor, which involves 
some command offers to disable it and displays some descriptive text to users along the lines of it 
`Once you disable the absolute persistance behaviour it can never be turned back on`
followed by some generic steps that involved 
going into the BIOS Setup,find the Absolute Persistence® Module knob ( should be under security ) 
Select Permanently Disabled. then the usually F10 to Save and Exit.

Then the disable command would valide the above steps ( if doable ) or ask for confirmation from the user that this has been done and then install/replacing the existing firmware with the the non-abolute-persistance firmware ( presumably both firmware versions are being stored by vendors in the firmware database )

A side not which it's better to mention since it's not documented anywhere afaik so you may or may not be aware of it but in the case of Lenovo the command will need to extract the GUID string from firmware.metainfo.xml contained with in the cab file and use that string instead of the existing GUID one when replacing the non persistant one.

hughsie, 2021-09-27:
> Lenovo provides two cab files for bios update

I didn't know this. @mrhpearson is this news to you as well? I don't know why a different capsule would be required. I assume the two ESRT tables use different GUIDs for the different types?

> The second part of the RFE is to provide end users the ability to permanently disable the Absolute backdoor

johannbg, 2021-09-27:
Arguably this is the exact tool to do ( + you already can, just not in very userfriendly manner ) and the absolute concept is massive security risk for end users.

hughsie, 2021-09-27:
> and the absolute concept is massive security risk for end users

johannbg, 2021-09-27:
Yes corporations,governments and militaries  might want to have this enabled or disabled  depending on the case but the benefit for the regular consumer is little to none thou they are let to believe there is.

Anyways no need for us to engage in a dialog of the usefulness of this certain industry feature and lets just focus on the user experience for those that dont want or need it. 

Lenovo now provides an opt out from this being installed by default on every purchase and provides 2 firmware files, one that includes the backdoor and surveillance software and another one that exclude it.

hughsie, 2021-09-27:
> one that includes the backdoor and surveillance software and another one that exclude it

Technically we can support this, we have `switch-branch` which can retarget a machine firmware stream, e.g. switching from AMI to coreboot and vice-versa. The question then becomes if Lenovo *wants* to support it.

closed 2021-09-27 by superm1