fwupd/fwupd #3827: --ignore-power does not ignore battery minimal requirements.

johannbg, 2021-10-01:
**Describe the bug**
Just encountered an issue in which a seemingly firmware update on a Lenovo is causing it no longer to charge the battery. 
Now ofcourse this was only found out when the battery had been fully discharged and it only runs on ac power. 

When trying to update/downgrade the firmware fwupd triggers
`Update Error:       Cannot install update when system battery is not at least 25%`

Now I fully understand this error/warning being triggered when a computer was not on AC power but when a computer is already on AC power ( and literally dies if you pull the plug so to speak ) it should arguably not be triggered.

**Steps to Reproduce**
Get a Lenovo laptop, update it's firmware, shake a magic 8 ball and it suddenly stops charging the laptop or at the exact same time the cosmic flows of the universe decide to destroy either the charger,charging port or the battery :smile: 

Lenovo could also be bricking battery`s when warranties are about to expire ( or have expired ) to force users to "upgrade" or doing some shenanigans with the batteries serial nr. and someone screwed up in the firmware update. Such tactics are not unheard of in the manufacturing industry these days.  

**Expected behavior**
No "Update Error:       Cannot install update when system battery is not at least 25%" error being triggered when on AC power.

**Secondary expectations**

That error being ignored when running either
`fwupdmgr install foo.cab --no-safety-check`
or
`fwupdmgr install foo.cab  --ignore-power`
or both.

superm1, 2021-10-01:
What fwupd version? We removed ignore power support in master, so I guess something older.

For now to get by your problem I would suggest to add upower plugin to the plugin disallow list in fwupd.conf

johannbg, 2021-10-02:
@superm1 yeah I'm not runnin master, the version is fwupd-1.6.4-1.fc34.

Disablin the plugin worked like a charm but it had to be placed in /etc/fwupd/daemon.conf followed by a restart of the fwupd daemon so presumably daemon.conf has been replaced/renamed to fwupd.conf as well in master.

superm1, 2021-10-02:
> @superm1 yeah I'm not runnin master, the version is fwupd-1.6.4-1.fc34.
> 
> Disablin the plugin worked like a charm but it had to be placed in /etc/fwupd/daemon.conf followed by a restart of the fwupd daemon so presumably daemon.conf has been replaced/renamed to fwupd.conf as well in master. 

Glad that worked. I just misremembered the name of the conf file when I responded from my phone :)

wmjb, 2021-10-02:
im trying to bypass the same kind of issue with a surface pro 3. i get battery level is too low. the above upower disable doesn't seem to have any effect when using current release

johannbg, 2021-10-02:
What kind of check?

Arguably the only power check that should be performed is the one to check if the computer is already plugged directly into AC power using the standard AC power adapter, as in, it's not connected to a docking station or running on battery, if it is running on battery or through a docking station then refuse to upgrade/downgrate the firmware. I dont think any manufacturer allows or recommends fw updates being performed when on battery or when connected through a docking station.

Anyways adding `;upower` to the end of the DisabledPlugins= line in /etc/fwupd/daemon.conf` wont take an effect until you restart the fwupd service `systemctl restart fwupd` then proceed to diagnose the issue ( which can range from being the charger,charging port, the battery and firmware update ) per the manufacturer instructions.

wmjb, 2021-10-02:
i think the check is being done by the existing firmware. it goes through the fwupd process, but upon reboot fwupd get-history reveals a transient failure due to battery too low.

hughsie, 2021-10-02:
> i think the check is being done by the existing firmware

Yes, I asked all the OEMs what the power condition requirement was for the UEFI update. It's not standardized, hence why we have https://github.com/fwupd/fwupd/blob/main/data/power.quirk

superm1, 2021-10-04:
I think we should close this issue.  As mentioned device inhibition broke this, and so we tore it out of master:
https://github.com/fwupd/fwupd/issues/3778

closed 2022-01-21 by hughsie