fwupd/fwupd #3829: Bitlocker asks for recovery key after firmware updates

pgera, 2021-10-04:
**Describe the bug**
On a dual-boot system with windows, updating firmware with fwupd on linux makes bitlocker on windows ask for recovery on the subsequent boot.

**Steps to Reproduce**
Update system firmware or bios with fwupd

**Expected behavior**
Updating firmware with fwupd should not disrupt bitlocker

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version

client version:	1.5.11
compile-time dependency versions
	gusb:	0.3.4

daemon version:	1.5.11
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): apt

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices

20U9005LUS
│
├─Thunderbolt host controller:
│     Device ID:          3980395e9227783cb5ddbb846c934c9800619da4
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    46.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0109, TBT:0x0109)
│     GUIDs:              32e299b7-9585-594c-833c-63ae28b14809
│                         13a933fd-ab08-5b89-b95f-f0fbd1e736af
│                         4117c99d-0660-5db6-bbfe-65ed036debe5
│                         bb3a7757-29b3-5950-9dba-71cc18c6fb86
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Device stages updates
│   
├─Embedded Controller:
│     Device ID:          0b3db288d2c013fcac57525c52aa6dc2c9934157
│     Current version:    0.1.13
│     Minimum Version:    0.1.13
│     Vendor:             DMI:LENOVO
│     GUIDs:              1ebe8ad3-3d40-4178-92ce-bbb3d0906332
│                         6dd2bd94-436c-5bb1-96a1-c8d0a87e4f09
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          cda6d9cd65fff6c200fd914ff50376263efddb1e
│     Current version:    225.53.1649
│     Minimum Version:    0.0.1
│     Vendor:             DMI:LENOVO
│     GUIDs:              bf778c75-bee3-49fc-ab7e-b84b5b6a5385
│                         5b03f187-8ccf-56d9-a66f-f286c86be3ad
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
[... 135 lines cut ...]
      GUIDs:              7b974856-317d-5538-a915-de0a6353750f
                          12c86995-0b90-5ec5-98f3-7a6ed4ca50e0
                          a39943dd-3afb-54f8-b110-c5a21f071200
                          fccbb6ea-e20e-58ad-bf8a-7fb7d43ff4c2
                          06b4e2aa-91af-508b-b06e-65e3b3189e97
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    
________________________________________________

Devices that have been updated successfully:

 • System Firmware (0.1.19 → 0.1.20)

```

**Additional questions**

- Operating system and version: Ubuntu 20.04 + Windows 10
- Have you tried rebooting? Yes
- Is this a regression? No

hughsie, 2021-10-04:
Isn't that expected if we change the PCR0? Do you have any idea how Windows does this?

pgera, 2021-10-04:
I'm not sure how windows does it. Do they suspend bitlocker before the update ? I grepped for issues in this repository and found this in https://github.com/fwupd/fwupd/issues/1371 in the changelog: "Added a new feature to automatically suspend BitLocker before upgrading the firmware. After the firmware upgrade is complete, BitLocker is automatically enabled."

hughsie, 2021-10-05:
@superm1 any ideas here?

superm1, 2021-10-05:
AFAIK BIOS upgrade tools on Windows do suspend Bitlocker on BIOS updates - yeah.  I don't expect we can do anything like that from Linux to suspend Windows Bitlocker.  We had discussed what we could do _natively_ on Linux in https://github.com/fwupd/fwupd/discussions/2548 but a major problem is that no distributions really do FDE and so no solution has been rallied behind.  See https://0pointer.net/blog/authenticated-boot-and-disk-encryption-on-linux.html for more on this.

IMO the **best** we could do is detect that an NTFS partition has bitlocker enabled and warn a user doing a firmware upgrade is about to break Bitlocker.  This isn't going to be trivial though.  udisks2 doesn't exactly have a bit saying an NTFS volume has bitlocker turned on, and so it's going to mean some sort of probing into the NTFS volume.

hughsie, 2021-10-05:
> detect that an NTFS partition has bitlocker enabled and warn a user doing a firmware upgrade

You answered my next question:

> so it's going to mean some sort of probing into the NTFS volume

I can think of a 100 ways this could go wrong.

hughsie, 2021-10-05:
Matthew Garrett suggested this was because the vendor might have updated the dbx, and hence invalidated PCR-7.

mjg59, 2021-10-05:
https://support.microsoft.com/en-us/topic/prompted-for-bitlocker-recovery-key-after-installing-updates-to-surface-uefi-or-tpm-firmware-on-surface-device-01973e22-1b57-f0db-6f33-03f62127831e talks about this issue a little, so it turns out to be more complicated than I thought - default policy is to seal to PCR 7 and 11, where 11 is entirely under the control of Windows so will do nothing relevant here. However, this is only true if Secure Boot is enabled - otherwise, PCR 0 does come into play. https://github.com/libyal/libbde supports probing the Bitlocker metadata but doesn't support actually mounting a disk with TPM keys (which shouldn't work anyway, because PCR 7).

superm1, 2021-10-05:
Policy actually can be set by group policy as well: https://docs.microsoft.com/en-us/windows/security/information-protection/bitlocker/bitlocker-group-policy-settings
Notably:

```
A platform validation profile consists of a set of PCR indices that range from 0 to 23. The default platform validation profile secures the encryption key against changes to the following:

Core Root of Trust of Measurement (CRTM), BIOS, and Platform Extensions (PCR 0)
Option ROM Code (PCR 2)
Master Boot Record (MBR) Code (PCR 4)
NTFS Boot Sector (PCR 8)
NTFS Boot Block (PCR 9)
Boot Manager (PCR 10)
BitLocker Access Control (PCR 11)
```

mjg59, 2021-10-06:
https://docs.microsoft.com/en-us/windows/security/information-protection/bitlocker/bitlocker-group-policy-settings suggests that EFI without secure boot is 0,2,4 and 11, and says:

" If Secure Boot is on and the firmware measures PCR 7 correctly per the UEFI specification, BitLocker can bind to this information rather than to PCRs 0, 2, and 4 which have the measurements of the exact firmware and Bootmgr images loaded. This reduces the likelihood of BitLocker starting in recovery mode as a result of firmware and image updates, and it provides you with greater flexibility to manage the preboot configuration."

so PCR 0 updates should only trigger issues if secure boot is disabled or if group policy is set to require additional PCRs.

hughsie, 2021-10-06:
Hey maybe it's a `FWUPD_PLUGIN_FLAG_HAS_BITLOCKER` instead?

superm1, 2025-11-01:
> I keep getting this message on my pure-Linux framework laptop 16 with LUKS encryption, do I need to mind?

Please open up your own issue to debug with logs.