fwupd/fwupd #3835: fwupd removes the hidraw when active for the realtek

kelvie, 2021-10-06:
**Describe the bug**
A clear and concise description of what the bug is.
When `fwupd` is running, and I connect my monitor, the hidraw device for it gets added, then deleted (e.g. `/dev/hidraw9`) This is the fwupd log:
```
Oct 05 23:49:49 kelvie-lws fwupd[311285]: 06:49:49:0067 FuEngine             device 39502f4d964cb1e458b324d4a2dc319d0b9f5d23 [HDMI Converter] does not define an update protocol
Oct 05 23:49:49 kelvie-lws fwupd[311285]: 06:49:49:0068 FuEngine             device 39502f4d964cb1e458b324d4a2dc319d0b9f5d23 [HDMI Converter] does not define an update protocol
Oct 05 23:49:49 kelvie-lws fwupd[311285]: 06:49:49:0071 FuDeviceLocker       failed to close device: failed to set clock-mode=0: failed to SetReport: endpoint stalled or request not supported
Oct 05 23:49:49 kelvie-lws systemd[1]: Started Firmware update daemon.
```
After this gets printed, my hidraw device is gone (even get a udev event for it).

**Steps to Reproduce**
Run `fwupd` (e.g. with fwupdmgr) with my monitor plugged in via USB

**Expected behavior**
`/dev/hidraw9` still exists so that I can control my monitor over HID

**fwupd version information**
Please provide the version of the daemon and client.

```
client version: 1.5.11
compile-time dependency versions
        gusb:   0.3.5

daemon version: 1.5.11
```

It came with Ubuntu 21.04

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

The only relevant one is this:
```
fwupdmgr get-devices --show-all-devices
│   
├─HID Device:
│ │   Device ID:          c028c7822752034fe9d52f72af2b3dd23707c65e
│ │   Current version:    1.39
│ │   Vendor:             Realtek (USB:0x0BDA)
│ │   Update Error:       device does not support authentication
│ │   GUIDs:              9eee27e3-b946-5b8f-98a1-10e0367143b6
│ │                       bb9ec3e2-77b3-53bc-a1f1-b05916715627
│ │ 
│ └─HDMI Converter:
│       Device ID:        39502f4d964cb1e458b324d4a2dc319d0b9f5d23
│       Vendor:           USB:0x0BDA
│       GUID:             371b794e-fbb1-5057-8767-21c8ac8f7b40
│       Device Flags:     • Updatable
│     

```
Here's the lsusb output of the device:
```
Bus 005 Device 113: ID 0bda:1100 Realtek Semiconductor Corp. HID Device
```

I see that there is a quirk for this, under rt24hid. I'm guessing what's happening is that the same product ID / chip is being used for multiple devices.

**Additional questions**

- Operating system and version: Ubuntu 21.04 amd64
- Have you tried rebooting? Yes
- Is this a regression? Not sure

hughsie, 2021-10-06:
The `VID_0BDA&PID_1100` is a Realtek reference board, not designed for OEM resale. Can you tell me exactly what monitor you have please?

kelvie, 2021-10-06:
I have a Gigabyte M32U. It's gone through a couple of firmware updates as well.

In my experience it's not uncommon for manufacturers to keep the OEM IDs...

hughsie, 2021-10-07:
@AnyProblem what do you want me to do here? I suspect it's unexpected that Gigabyte used the demo board VID/PID.

kelvie, 2021-10-08:
Yes, it uses HID to control the monitor as well as do firmware updates.