fwupd/fwupd #3945: Need to add some extra fields to the HSI specification

hughsie, 2021-11-03:
I've been asked my the Insights team at Red Hat to add some more fields to the HSI spec to make it more useful to mere mortals. Before I spend a ton of time writing prose and detailed notes, I wanted some feedback on what the docs should look like.

1:
[image]

2:
[image]

3:
[image]

Personally I like 1 the most.

hughsie, 2021-11-03:
@superm1 ideas?

superm1, 2021-11-03:
HSI wasn't supposed to be something that mere mortals could usually solve, It's supposed to be a tool for manufacturers to catch problems or users to tell manufacturers about problems....

hughsie, 2021-11-03:
The usecase here is that we're showing the HSI failures to a server admin, and currently the admin doesn't know:

 * What component this affects
 * What consequence this failure has
 * What they can do about it

At the moment users don't know if they should talk to their OS vendor, their CPU vendor or their BIOS vendor.

I think some or all of that needs to go in, otherwise each failure is pretty meaningless.

superm1, 2021-11-03:
> At the moment users don't know if they should talk to their OS vendor, their CPU vendor or their BIOS vendor.

I disagree.

I think it will _never_ be BIOS vendor.  2 points to support this:
1. people don't buy products from IBVs, they buy them from OEMs
2. OEM has final control what goes in.  It might be broken in reference code or implementation from IBV or CPU vendor, but OEM decides what to override.

In terms of OS vendor, what kind of thing is this really?  No OS vendors do FDE by default.  None of them use the TPM by default.  Hibernation files and encryption are options in the installer.  It's something you or whoever built your kickstart/preseed picked.  The kernel controls lockdown behavior.

To me - everything is talk to your manufacturer or change a knob at install time or runtime.

hughsie, 2021-11-03:
What about the other two things; i.e. explaining the consequence from an attack point of view and (possibly) what the user should do in the BIOS setup?

superm1, 2021-11-03:
>What about the other two things; 

It's really not the business fwupd should be in in my opinion.  I think dumbing the spec down is going to over-simplify potentially complex things that 95% of the time a user can do nothing about.  If your machine doesn't have the IOMMU enabled for whatever reason, or the CPU architecture doesn't support SME what can you really do?
Instead it's better to think of the goal where machines meet a certain level by default and getting the ecosystem caught up to that.

>explaining the consequence from an attack point of view

Mayyyybe.  I sort of feel like all roads lead to "insecure setup" and "insecure setup" leads to you should be cautious what you plug in , cautious who uses it and what software you run.

>what the user should do in the BIOS setup?

What they can do in BIOS setup is also going to be problematic - it will not be in the same place on all BIOSes, or it might not be exposed.  Counter-point - how about instead **fixing things** that can be fixed with `/sys/class/firmware_attributes`?  Then you don't need to say for Vendor XYZ go into Menu ABC, flip knob L.

superm1, 2021-11-03:
BTW - remember we wrote https://github.com/fwupd/fwupd/wiki/Host-security-ID-runtime-issues too.

skontar, 2021-11-04:
To put some things in perspective:

I am a SW engineer. I am not a HW expert, but I am also not completely clueless either.
Lets say that my goal is to explain the detected problem to a "mere mortal" who bought the hardware and convince them to fix it and make them more secure. It may sound dumb, but yes, that is what security people do all the time – convince people to fix security issues on their systems.

From that perspective I need to know:
1) What HW is affected. Yes, most of the people but whole box/server/laptop from a OEM vendor. But some custom build it or have it custom built. I believe that in those cases they need to know which vendor they need to talk to?
2) Worst case impact. It is fine to not be detailed or play with different possibilities. I need to briefly explain what is the worst thing they can expect if they do not get the issue resolved. It is usually also very important to know if the attacker needs to have physical access to the HW. Most of the people would not care in that case.
3) How do I resolve it. Again, I do not need detailed steps. Is that something I can probably change in BIOS? Is that something my HW vendor needs to change? Or can I just download a new firmware from them?

Now, with this perspective, as an exercise I tried to get necessary information from https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.IntelDci.Enabled . It took me maybe 20 minutes digging in the associated references to be able to explain the issue to a "mere mortal" and I am not even sure I got some of it right.

It is frustrating experience. I do not believe that this is the expected outcome of me following a link from `fwupdagent security --force`. I believe that proposed changes would improve usability of it for additional audience, but it is up to you to decide.

skontar, 2021-11-04:
@hughsie I like the #2 the most.

superm1, 2021-11-04:
>But some custom build it or have it custom built. I believe that in those cases they need to know which vendor they need to talk to?

For custom build isn't it still talk to motherboard manufacturer?  I don't know why anyone would go to an IBV or CPU vendor directly.

> I believe that proposed changes would improve usability of it for additional audience, but it is up to you to decide.

I guess in my mind the best parallel to draw is to [Microsoft SCPC](https://docs.microsoft.com/en-us/windows-hardware/design/device-experiences/oem-highly-secure).  If you don't meet the requirements for SCPC you get some messaging in the OS and it's very clear that you need to go to your OEM to fix it.  The end user doesn't see all the details about WHY they aren't meeting them.  But the OEM can run a set of checks to verify what is wrong.  Bring that idea to Linux world - HSI can show up a number in GNOME somewhere and if you aren't satisfied with it and you're technical you run `fwupdmgr security` to see the details.  If you're not technical, you talk to your OEM/manufacturer about getting it improved.

hughsie, 2021-11-05:
>  The end user doesn't see all the details

I guess @skontar is coming from the direction where these HSI results will be shown to the site administrator, i.e. the person who will be actually contacting the OEM with a bugreport or support ticket. At the moment there's not enough information in the spec to educate either the admin, or the level 1 support rep.

Personally, I think the _Impact_ and _Resolution_ do that, although I concede the _Affected Hardware_ isn't so useful given how intertwined this all is.

superm1, 2021-11-05:
Some of these things can't realistically be changed in a firmware update though and are design decisions, and I guess that is getting mixed up here.

Coming back to my previous point - SME and shadow stack - the hardware supports it or it doesn't. So what is opening a ticket going to do here for those?

Similarly some older hardware uses S3 and you can't change it in the field to S2I specifically because it requires an OS reinstallation on Windows to change. So if the system goes out one way it stays.

So my point remains it should be something that at most it should be something you captured a verbose log and sent it over but otherwise don't expect it can be solved for your hardware.

hughsie, 2021-11-05:
> So what is opening a ticket going to do here for those?

Point taken. For this maybe there should be no resolution section at all; you can't do anything.

> otherwise don't expect it can be solved for your hardware

So how about as a compromise, we add the impact section to all attributes, and only add the resolution section if there's something you can reasonably do to change it (**not** "_contact your oem_") -- with the aim of using the firmware_attributes to do using a DBus call in the future.

hughsie, 2021-11-05:
https://github.com/fwupd/fwupd/pull/3958 as a starter

skontar, 2021-11-05:
@hughsie After trying to work with the information and translating it to an Insights recommendation, I agree that _Affected Hardware_ is not useful.

closed 2021-12-05 by hughsie