fwupd/fwupd #3950: `fwupdmgr update` does not update firmware

jakobrs, 2021-11-05:
**Describe the bug**
If I run `fwupdmgr update`, it asks if I want to install a firmware update for my Dell Latitude 3510. If I proceed with the update, it claims to download and install the firmware, but nothing happens when the system is rebooted. No files are created in `/boot`.

**Steps to Reproduce**
```bash
fwupdmgr update
find /boot # to verify that no files have been created
reboot
```

**Expected behavior**
The firmware to be updated

**fwupd version information**
```
client version: 1.7.1
compile-time dependency versions
        gusb:   0.3.8

daemon version: 1.7.1
```

Installed using `pacman -S fwupd`

**fwupd device information**

```
Latitude 3510
│
├─Core™ i5-10210U CPU @ 1.60GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000ea
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         561403e8-143a-5071-ab09-bf5e4c146983
│                         d5092916-1868-5182-8c70-12f83dab6372
│     Device Flags:       • Internal device
│   
├─GP108M [GeForce MX230]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a1
│     Vendor:             NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:              e2b6c51a-7395-5fc3-96d9-74197deec613
│                         f482d53a-20be-51ab-959c-11796ab209d7
│                         c1fe835f-2b15-5039-911f-34c9c617c1f1
│                         aa0023eb-3f41-5de2-a9d5-6996b2eb0752
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Samsung SSD 970 EVO Plus 1TB:
│     Device ID:          3743975ad7f64f8d6575a9ae49fb3a8856fe186f
│     Summary:            NVM Express solid state drive
│     Current version:    2B2QEXM7
│     Vendor:             Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:              0b4d773a-7ac3-58c1-a541-e22ef1cdfe02
│                         c9d531ea-ee7d-5562-8def-c64d0d144813
│                         6e54c992-d302-59ab-b454-2d26ddd63e6d
│                         47335265-a509-51f7-841e-1c94911af66b
│                         f05ecac9-5cc2-5f77-b23b-a08ad8879ada
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    1.3.0
[... 119 lines cut ...]
║ Some new functionality has also been added:                                  ║
║                                                                              ║
║ *Improved the system stability                                               ║
║                                                                              ║
║ Latitude 3510 must remain plugged into a power source for the duration of    ║
║ the update to avoid damage.                                                  ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
Downloading…             [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Waiting…                 [***************************************]
Successfully installed firmware
```

**Additional questions**

- Operating system and version: Arch Linux
- Have you tried rebooting? yes
- Is this a regression? unsure
- Are you using an NVMe disk? yes
- Is secure boot enabled? no
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? I couldn't find any such setting in the BIOS

hughsie, 2021-11-05:
Can you run `sudo /usr/libexec/fwupd/fwupd --verbose` in one terminal and then try to perform the update again in another terminal please. Then please upload the entire verbose daemon log. Thanks!

jakobrs, 2021-11-06:
`fwupdmgr update` refused to do anything if I ran `/usr/lib/fwupd/fwupd --verbose` normally (without using systemd) so I modified the fwupd.service file to launch fwupd with `--verbose` and log to a file.

fwupd.stdout.log: [fwupd.stderr.log](https://github.com/fwupd/fwupd/files/7490037/fwupd.stderr.log)

hughsie, 2021-11-06:
I see:

    07:34:52:0240 FuEngine             disabling plugin because: failed to coldplug using uefi_capsule: ESRT GUID '00000000-0000-0000-0000-000000000000' was not valid

which I'm thinking causes the early-exit of the uefi_capsule plugin and the reason it's not doing the right thing on update. Will build a PR for you to test.

closed 2021-11-09 by hughsie