fwupd/fwupd #3964: Dell XPS 13 9360: Firmware update from 2.15.1 to 2.17.0 failed

paulmenzel, 2021-11-08:
The system had 6 % of battery left but was connected to power.

    $ fwupdmgr update --force

The system rebooted after confirmation as expected. The Dell logo was shown, and the firmware update program started. But when I came back, it said *Firmware update failed*.

```
$ fwupdmgr get-history 
XPS 13 9360
│
├─XPS13 9360 Thunderbolt Controller:
│ │   Device ID:          9516070bdf03a7484d1145383fc0faeb9c51a521
│ │   Previous version:   00.00
│ │   Update State:       Failed
│ │   Update Error:       could not detect device after update: timed out while waiting for device
│ │   Last modified:      2018-10-30 14:51
│ │   GUID:               4d86f168-e1cc-5995-afd3-ae9df6a14f5e
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │ 
│ └─  New version:      21.00
│       License:          Unknown
│     
├─XPS 13 9360 TPM 2.0:
│ │   Device ID:          a60b665c769b019e30f31001e91ab31e24d7a035
│ │   Previous version:   1.3.0.1
│ │   Update State:       Failed
│ │   Update Error:       {error #0} creator.c:98 find_file(): could not find mountpoint: Datei oder Verzeichnis nicht gefunden	{error #1} creator.c:355 efi_generate_file_device_path(): could not canonicalize fs path: Datei oder Verzeichnis nicht gefunden: failed to efi_generate_file_device_path(fw/fwupd-73c2051d-8688-56fb-a93f-d56a9b455e52.cap)
│ │   Last modified:      2018-10-30 18:14
│ │   GUID:               73c2051d-8688-56fb-a93f-d56a9b455e52
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │ 
│ └─  New version:      1.3.1.0
│       License:          Unknown
│     
└─System Firmware:
  │   Device ID:          3f0ecc7823ee99c718693e8ecca38f54e8738dc5
  │   Previous version:   2.15.1
  │   Update State:       Transient failure
  │   Update Error:       failed to update to 0: battery level is too low
  │   Last modified:      2021-11-08 10:57
  │   GUID:               5ffdbc0d-f340-441c-a803-8439c8c0ae10
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─XPS 13 9360:
        New version:      2.17.0
        Remote ID:        lvfs
        Summary:          Firmware for the Dell XPS 13 9360
        License:          Proprietary
        Size:             14,7 MB
        Created:          2021-07-13
        Urgency:          Critical
        Vendor:           Dell
        Flags:            is-upgrade
        Description:      
        This stable release fixes the following issues:
        
        • Firmware updates to address security vulnerabilities.
      
```

Trying it again (with the battery charged more) the update worked. No idea if that was related, but, if related, the user experience with the Dell firmware, that in the end the reason for the failed update, wasn’t shown leaves a lot to be desired.

superm1, 2021-11-08:
You literally ran with --force to override battery checks and expected a different result?

paulmenzel, 2021-11-08:
First, you confirm that there are such checks in place in the Dell UEFI Capsule?

Why is that unreasonable? If forcing to disable the battery check won’t have any effect as the checks still are in place, and the update is going to fail, then why allow that?

Additionally, the firmware update program should say: “Battery too low, policy disallow the upgrade.” instead of just “Firmware update failed.”.

As a user, if I know I have the charger connected, I’d still expect to upgrade the firmware with a low battery.

superm1, 2021-11-08:
> First, you confirm that there are such checks in place in the Dell UEFI Capsule?

Yes.  Look at your messages.
```
  │   Update Error:       failed to update to 0: battery level is too low
```

> Why is that unreasonable? If forcing to disable the battery check won’t have any effect as the checks still are in place, and the update is going to fail, then why allow that?

TOCTOU.  You don't **have to** reboot immediately.  If you're at 42% right now and not going to reboot until the end of the day you might stage the update and then reboot later.  When you reboot later then you'll be charged enough for it succeed.

> Additionally, the firmware update program should say: “Battery too low, policy disallow the upgrade.” instead of just “Firmware update failed.”.

It does, look closer at your Update Error in the history.

> As a user, if I know I have the charger connected, I’d still expect to upgrade the firmware with a low battery.

Nothing fwupd can do here.  It's up to the OEM behavior.

closed 2021-11-08 by superm1

paulmenzel, 2021-11-08:
Thank you for the explanation. I didn’t read the failure reason, as I skipped it after reading the version. Didn’t see that it is the previous version.

    Previous version:   2.15.1

The rest is related to Dell’s UEFI capsule. Too bad, the user experience is that bad. (Not to mention the ugly looking UI, not being fit for the HiDPI screen.)