fwupd/fwupd #3968: modem-manager: Quectel EG25-G fastboot updates fail due to USB disconnections

DylanVanAssche, 2021-11-09:
**Describe the bug**

Quectel EG25-G modem can be upgraded through `fastboot` which is possible with `fwupd`, but as soon as the actual upgrade starts, the bulk transfer of the firmware files fail [1] with `USB error on device 18d1:d00d : No such device (it may have been disconnected) [-4]`. The modem seems to disconnect during the probing and the writing of the firmware over fastboot.

I added a quirk as [done for other modems](https://github.com/fwupd/fwupd/blob/main/plugins/modem-manager/modem-manager.quirk#L2-L5) to handle the handover of the original GUID and the fastboot GUID (see [2]).
This particular modem does not reply `OK` like the [plugin expects](https://github.com/fwupd/fwupd/blob/main/plugins/modem-manager/fu-mm-device.c#L628-L652) so I patched `fwupd` to ignore this as this modem directly enters fastboot mode (see [3]). I plan to rework this as a `modem-manager` plugin quirk.

In [4] I included the verbose log of `fwupd` together with the udev events that are happening which shows the disconnection in between.

[1] https://github.com/fwupd/fwupd/blob/main/plugins/fastboot/fu-fastboot-device.c#L106-L117
[2] Quirk:
```
# Quectel EG25-G
[USB\VID_2C7C&PID_0125]
Summary = Quectel EG25-G modem
CounterpartGuid = USB\VID_18D1&PID_D00D

# Quectel EG25-G in fastboot mode
[USB\VID_18D1&PID_D00D]
Summary = Quectel EG25-G modem (fastboot)
CounterpartGuid = USB\VID_2C7C&PID_0125
```
[3] Temporary patch:
```
--- a/plugins/modem-manager/fu-mm-device.c
+++ b/plugins/modem-manager/fu-mm-device.c
@@ -625,6 +625,12 @@ fu_mm_device_at_cmd(FuMmDevice *self, const gchar *cmd, GError **error)
 					  1500,
 					  FU_IO_CHANNEL_FLAG_SINGLE_SHOT,
 					  error);
+	/* 
+	 * HACK: always assume that booting into fastboot works
+	 * Quectel EG25-G modem does not send a response, it just EOFed
+	 */
+	return TRUE;
+
 	if (at_res == NULL) {
 		g_prefix_error(error, "failed to read response for %s: ", cmd);
 		return FALSE;
```
[4] [log-fwupd-no-such-device.txt](https://github.com/fwupd/fwupd/files/7506860/log-fwupd-no-such-device.txt)

**Steps to Reproduce**
1. Run `fwupdmgr install <CAB file>` on a device with a Quectel EG25-G modem installed such as a PinePhone
2. `fwupd` unpacks the CAB file, switches the modem into fastboot mode, inhibits ModemManager
3. `fwupd` fails to upgrade the firmware because it cannot initiate a bulk transfer: `USB error on device 18d1:d00d : No such device (it may have been disconnected) [-4]`

**Expected behavior**

USB bulk transfer with the firmware starts and the flashing process completes.

**fwupd version information**

```shell
pine64-pinephone:~$ fwupdmgr --version
client version:	1.7.1
compile-time dependency versions
	gusb:	0.3.8
daemon version:	1.7.1
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): Alpine Linux's apk

**fwupd device information**

```shell
Pine64 PinePhone (1.2)
│
├─DA4032:
│     Device ID:          0e0c93c8b4bb222157feedbde8f863e23bd1a8f7
│     Current version:    0x3034313430363139
│     Vendor:             EMMC:0x000045
│     GUIDs:              8c6f96d4-f406-53fa-84ca-a367cd368dfc
│                         add792f2-7053-5638-80e7-20987fd16769
│                         5500994c-1b73-52b4-acf4-bb574b6f3029
│     Device Flags:       • Internal device
│                         • Updatable
│   
└─QUECTEL Mobile Broadband Module:
      Device ID:          976c4a39e87f61e6940ea6a8d39c583cfa99615f
      Summary:            Mobile broadband device
      Current version:    EG25GGBR07A08M2G - 05010820
      Vendor:             QUALCOMM INCORPORATED (USB:0x2C7C)
      GUIDs:              c33a4560-8681-55b6-bbb6-85258f2de149
                          1a2996cb-f86e-5583-a464-e1b96e1c6ae9
                          587bf468-6859-5522-93a7-6cce552a0aa3
                          22ae45db-f68e-5c55-9c02-4557dca238ec
      Device Flags:       • Updatable

```

**Additional questions**

- Operating system and version: postmarketOS edge (Alpine Linux edge)
- Have you tried rebooting? Yes
- Is this a regression? No

superm1, 2021-11-09:
> USB error on device 18d1:d00d : No such device (it may have been disconnected) [-4]

I'm not sure there is a lot that fwupd can do about this.  Anything notable from the kernel on it at this time?

DylanVanAssche, 2021-11-09:
@superm1 Using the `fastboot` tool works fine to perform the upgrade manually.
The kernel only shows the disconnection of the USB for its normal operating mode (as a modem device).
If I run `fastboot devices`, it shows up properly and can be used through the `fastboot` tool.

Could we refresh the device information `fwupd` has about the USB device before performing this action?

closed 2021-12-14 by hughsie