fwupd/fwupd #3972: TPM v1.2 never found

kerneis-anssi, 2021-11-10:
The following line has a typo: https://github.com/fwupd/fwupd/blob/main/plugins/tpm/fu-plugin-tpm.c#L293

It should be "tpm0" instead of "tmp0".

As a result, TPM v1.2 would never be detected.

In addition of fixing this, that's one more reason why this module should be tested by loading the plugin and setting `FWUPD_SYSFSTPMDIR` to `testdatadir` instead of instantiating the TPM v1 and v2 devices. (Another reason is that it would allow testing the HSI checks end-to-end.)

hughsie, 2021-11-10:
@kerneis-anssi you got this or want me to fix it?

hughsie, 2021-11-10:
It's actually worse than you describe, as we never actually add the V1 device... Patch incoming.

closed 2021-11-10 by hughsie

kerneis-anssi, 2021-11-10:
> It's actually worse than you describe, as we never actually add the V1 device... Patch incoming.

Ah, I was wondering what kind of gobject magic loaded the V1 device :-D