fwupd/fwupd #3984: fwupd service core dumps on start

MaSven, 2021-11-13:
**Describe the bug**
Trying to start the fwupd.service results in a core cump of the service. 

**Steps to Reproduce**
```bash
systemctl restart fwupd.service
Job for fwupd.service failed because a fatal signal was delivered causing the control process to dump core.
See "systemctl status fwupd.service" and "journalctl -xeu fwupd.service" for details.

journalctl -u fwupd.service

Nov 13 17:36:41 linux.fritz.box systemd[1]: Starting Firmware update daemon...
Nov 13 17:36:41 linux.fritz.box fwupd[13584]: ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a)
Nov 13 17:36:41 linux.fritz.box fwupd[13584]: 16:36:41:0261 FuEngine             no GUIDs for device (null) [42f81e42b1e21ceb211b345766cfd39439cb242f]
Nov 13 17:36:41 linux.fritz.box fwupd[13584]: 16:36:41:0267 FuEngine             no GUIDs for device (null) [42f81e42b1e21ceb211b345766cfd39439cb242f]
Nov 13 17:36:41 linux.fritz.box fwupd[13584]: 16:36:41:0268 FuEngine             no GUIDs for device (null) [42f81e42b1e21ceb211b345766cfd39439cb242f]
Nov 13 17:36:41 linux.fritz.box systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
Nov 13 17:36:41 linux.fritz.box systemd[1]: fwupd.service: Failed with result 'core-dump'.
Nov 13 17:36:41 linux.fritz.box systemd[1]: Failed to start Firmware update daemon.
```

**Expected behavior**
The service should start normally.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
client version: 1.6.2
compile-time dependency versions
        gusb:   0.3.5
Failed to connect to daemon: Fehler beim Aufruf von StartServiceByName für org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)

```

```shell
zypper info fwupd 
Repository-Daten werden geladen...
Installierte Pakete werden gelesen...

Informationen zu Paket fwupd:
-----------------------------
Repository         : Haupt-Repository (OSS)
Name               : fwupd
Version            : 1.6.2-3.1
Arch               : x86_64
Anbieter           : openSUSE
Installierte Größe : 4,7 MiB
Installiert        : Ja (automatisch)
Status             : aktuell
Quellpaket         : fwupd-1.6.2-3.1.src
Zusammenfassung    : Device firmware updater daemon
Beschreibung       : 
    fwupd is a daemon to allows session software to update device firmware on
    the local machine.

    You can either use a GUI software manager like GNOME Software to view and apply
    updates, the command line tool or the system D-Bus interface directly.
```

Preinstalled on opensuse.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

Does not work, because the daemon does not start.

**Additional questions**

- Operating system and version:
Opensuse Tumbleweed Snapshot openSUSE-20211028-0
- Have you tried rebooting?
Yes
- Is this a regression?
Dont know

hughsie, 2021-11-13:
Does it happen with something newer like 1.7.x?

MaSven, 2021-11-13:
I will try it out.

MaSven, 2021-11-13:
I first tried it with 1.6.3 that I compiled here https://build.opensuse.org/package/show/home:MaSven:branches:Base:System/fwupd and that works.

MaSven, 2021-11-13:
I will make an update request at opensuse also for all the other versions till 1.7.1

hughsie, 2021-11-13:
Thanks, good to hear

closed 2021-11-13 by hughsie

eboye, 2025-01-27:
I think this is related to the same issue:

```bash
× fwupd.service - Firmware update daemon
     Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static)
     Active: failed (Result: core-dump) since Mon 2025-01-27 12:40:19 CET; 28min ago
   Duration: 18h 6min 43.370s
 Invocation: 7ee53b391c214ec7aafd1e001760cec2
       Docs: https://fwupd.org/
    Process: 3528 ExecStart=/usr/lib/fwupd/fwupd (code=dumped, signal=TRAP)
   Main PID: 3528 (code=dumped, signal=TRAP)
   Mem peak: 181.4M (swap: 8.9M)
        CPU: 3.151s

Jan 27 12:40:18 stupid-dragon fwupd[3528]: 11:40:18.195 FuBackend            failed to create device: failed to probe: failed to open /sys/devices/pci0000:00/0000:00:08.1/0000:0e:00.3/uevent: Too many open files
Jan 27 12:40:18 stupid-dragon fwupd[3528]: 11:40:18.195 FuBackend            failed to create device: failed to probe: failed to open /sys/devices/pci0000:00/0000:00:08.1/uevent: Too many open files
Jan 27 12:40:18 stupid-dragon fwupd[3528]: 11:40:18.202 FuBackend            failed to create device: failed to probe: failed to open /sys/devices/pci0000:00/0000:00:08.1/uevent: Too many open files
Jan 27 12:40:18 stupid-dragon fwupd[3528]: 11:40:18.202 FuBackend            failed to create device: failed to probe: failed to open /sys/devices/pci0000:00/0000:00:08.1/0000:0e:00.3/uevent: Too many open files
Jan 27 12:40:18 stupid-dragon fwupd[3528]: 11:40:18.202 FuBackend            failed to create device: failed to probe: failed to open /sys/devices/pci0000:00/0000:00:08.1/uevent: Too many open files
Jan 27 12:40:18 stupid-dragon fwupd[3528]: 11:40:18.785 GLib                 Creating pipes for GWakeup: Too many open files
Jan 27 12:40:19 stupid-dragon systemd-coredump[225112]: [🡕] Process 3528 (fwupd) of user 0 dumped core.
                                                        
                                                        Stack trace of thread 3528:
                                                        #0  0x00007c57b1a47d1b g_logv (libglib-2.0.so.0 + 0x62d1b)
                                                        #1  0x00007c57b1a47f94 g_log (libglib-2.0.so.0 + 0x62f94)
                                                        #2  0x00007c57b1a9ee73 n/a (libglib-2.0.so.0 + 0xb9e73)
                                                        #3  0x00007c57b1a3fb6c g_main_context_new_with_flags (libglib-2.0.so.0 + 0x5ab6c)
                                                        #4  0x00007c57b1c3667c g_dbus_connection_send_message_with_reply_sync (libgio-2.0.so.0 + 0x10167c)
                                                        #5  0x00007c57b1c43491 n/a (libgio-2.0.so.0 + 0x10e491)
                                                        #6  0x00007c57b1c4fa9a n/a (libgio-2.0.so.0 + 0x11aa9a)
                                                        #7  0x00007c57b1c4fc9b g_dbus_proxy_call_sync (libgio-2.0.so.0 + 0x11ac9b)
                                                        #8  0x000059fd5e4f2457 n/a (n/a + 0x0)
                                                        #9  0x00007c57b1c3ced2 n/a (libgio-2.0.so.0 + 0x107ed2)
                                                        #10 0x00007c57b1a42559 n/a (libglib-2.0.so.0 + 0x5d559)
                                                        #11 0x00007c57b1aa5257 n/a (libglib-2.0.so.0 + 0xc0257)
                                                        #12 0x00007c57b1a43287 g_main_loop_run (libglib-2.0.so.0 + 0x5e287)
                                                        #13 0x000059fd5e4f4c1b n/a (n/a + 0x0)
                                                        #14 0x000059fd5e4eb709 n/a (n/a + 0x0)
                                                        #15 0x00007c57b17bbe08 n/a (libc.so.6 + 0x25e08)
                                                        #16 0x00007c57b17bbecc __libc_start_main (libc.so.6 + 0x25ecc)
                                                        #17 0x000059fd5e4ebbd5 n/a (n/a + 0x0)
                                                        ELF object binary architecture: AMD x86-64
Jan 27 12:40:19 stupid-dragon systemd[1]: fwupd.service: Main process exited, code=dumped, status=5/TRAP
Jan 27 12:40:19 stupid-dragon systemd[1]: fwupd.service: Failed with result 'core-dump'.
Jan 27 12:40:19 stupid-dragon systemd[1]: fwupd.service: Consumed 3.151s CPU time, 181.4M memory peak, 8.9M memory swap peak.
```

I will open another one if needed, but I think it's related.

superm1, 2025-01-27:
> I will open another one if needed, but I think it's related.

Please open your own issue.