fwupd/fwupd #3990: 1.7.1: test suite is failing

kloczek, 2021-11-15:
**Describe the bug**
Test suite is failing in two units

**Steps to Reproduce**
Here is set of meson settings which I'm using on build my rpm package:
```spec
%meson \
        -D bluez=true \
        -D build=all \
        -D consolekit=true \
        -D curl=true \
        -D docs=gtkdoc \
        -D firmware-packager=true \
        -D gudev=true \
        -D gusb=true \
        -D introspection=true \
        -D libarchive=true \
        -D lvfs=true \
        -D lzma=true \
        -D man=true \
        -D plugin_altos=true \
        -D plugin_amt=true \
        -D plugin_dell=true \
        -D plugin_dummy=true \
        -D plugin_emmc=true \
        -D plugin_flashrom=true \
        -D plugin_intel_spi=true \
        -D plugin_logitech_bulkcontroller=true \
        -D plugin_modem_manager=true \
        -D plugin_msr=true \
        -D plugin_nvme=true \
        -D plugin_parade_lspcon=true \
        -D plugin_platform_integrity=true \
        -D plugin_realtek_mst=true \
        -D plugin_redfish=true \
        -D plugin_synaptics_rmi=true \
        -D plugin_thunderbolt=true \
        -D plugin_tpm=true \
        -D plugin_uefi_capsule_splash=true \
        -D plugin_uefi_capsule=true \
        -D plugin_uefi_pk=true \
        -D polkit=true \
        -D soup_session_compat=true \
        -D supported_build=true \
        -D systemd=true \
        -D tests=true \
        %{nil}
```
**Expected behavior**
test suite shoud finish without fails/errors.

**fwupd version information**
1.7.1

Test suite output:
```console
+ cd fwupd-1.7.1
+ /usr/bin/meson test -C x86_64-redhat-linux-gnu --num-processes 12 --print-errorlogs
ninja: Entering directory `/home/tkloczko/rpmbuild/BUILD/fwupd-1.7.1/x86_64-redhat-linux-gnu'
ninja: no work to do.
 1/32 fwupd-exported-api                     OK               0.17s
 2/32 fwupdplugin-exported-api               OK               0.04s
 3/32 fwupd-context-test                     OK               1.51s
 4/32 fwupd-thread-test                      OK               1.67s
 5/32 fwupd-self-test                        OK               1.72s
 6/32 fwupdplugin-self-test                  OK               3.05s
 7/32 fu-self-test                           OK               4.53s
 8/32 acpi-dmar-self-test                    OK               0.52s
 9/32 acpi-facp-self-test                    OK               0.50s
10/32 acpi-phat-self-test                    OK               0.46s
[... 137 lines cut ...]
(/home/tkloczko/rpmbuild/BUILD/fwupd-1.7.1/x86_64-redhat-linux-gnu/plugins/uefi-capsule/uefi-self-test:156507): XbSilo-DEBUG: 11:28:34.207: attempting to load /tmp/.ZD7TC1
# XbSilo-DEBUG: failed to load silo: blob too small
(/home/tkloczko/rpmbuild/BUILD/fwupd-1.7.1/x86_64-redhat-linux-gnu/plugins/uefi-capsule/uefi-self-test:156507): XbSilo-DEBUG: 11:28:34.207: failed to load silo: blob too small
# FuQuirks-DEBUG: invalid key names: CcgxFlashRowSize,CcgxFlashSize,CcgxImageKind,CxaudioChipIdBase,CxaudioPatch1ValidAddr,CxaudioPatch2ValidAddr,DellDockBlobBuildOffset,DellDockBlobMajorOffset,DellDockBlobMinorOffset,DellDockBlobVersionOffset,DellDockBoardMin,DellDockHubVersionLowest,DellDockInstallDurationI2C,DellDockUnlockTarget,DellDockVersionLowest,DfuAltName,DfuForceTimeout,DfuForceVersion,ElantpI2cTargetAddress,ElantpIapPassword,ElantpIcPageCount,JabraMagic,LogitechHidppModelId,ParadeLspconAuxDeviceName,PciBcrAddr,RealtekMstDpAuxName,RealtekMstDrmCardKernelName,Rts54I2cSpeed,Rts54RegisterAddrLen,Rts54TargetAddr,SuperioAutoloadAction,SuperioControlPort,SuperioDataPort,SuperioGType,SuperioId,SuperioPort,SuperioTimeout,SynapticsMstDeviceKind,VliDeviceKind,VliSpiCmdChipErase,VliSpiCmdSectorErase,WacomI2cFlashBaseAddr,WacomI2cFlashBlockSize,WacomI2cFlashSize
(/home/tkloczko/rpmbuild/BUILD/fwupd-1.7.1/x86_64-redhat-linux-gnu/plugins/uefi-capsule/uefi-self-test:156507): FuQuirks-DEBUG: 11:28:34.289: invalid key names: CcgxFlashRowSize,CcgxFlashSize,CcgxImageKind,CxaudioChipIdBase,CxaudioPatch1ValidAddr,CxaudioPatch2ValidAddr,DellDockBlobBuildOffset,DellDockBlobMajorOffset,DellDockBlobMinorOffset,DellDockBlobVersionOffset,DellDockBoardMin,DellDockHubVersionLowest,DellDockInstallDurationI2C,DellDockUnlockTarget,DellDockVersionLowest,DfuAltName,DfuForceTimeout,DfuForceVersion,ElantpI2cTargetAddress,ElantpIapPassword,ElantpIcPageCount,JabraMagic,LogitechHidppModelId,ParadeLspconAuxDeviceName,PciBcrAddr,RealtekMstDpAuxName,RealtekMstDrmCardKernelName,Rts54I2cSpeed,Rts54RegisterAddrLen,Rts54TargetAddr,SuperioAutoloadAction,SuperioControlPort,SuperioDataPort,SuperioGType,SuperioId,SuperioPort,SuperioTimeout,SynapticsMstDeviceKind,VliDeviceKind,VliSpiCmdChipErase,VliSpiCmdSectorErase,WacomI2cFlashBaseAddr,WacomI2cFlashBlockSize,WacomI2cFlashSize
# GLib-DEBUG: setenv()/putenv() are not thread-safe and should not be used after threads are created
(/home/tkloczko/rpmbuild/BUILD/fwupd-1.7.1/x86_64-redhat-linux-gnu/plugins/uefi-capsule/uefi-self-test:156507): GLib-DEBUG: 11:28:34.291: setenv()/putenv() are not thread-safe and should not be used after threads are created
Bail out! FATAL-CRITICAL: failed to allocate dbus proxy object: Error calling StartServiceByName for com.intel.tss2.Tabrmd: Timeout was reached
stderr:

** (/home/tkloczko/rpmbuild/BUILD/fwupd-1.7.1/x86_64-redhat-linux-gnu/plugins/uefi-capsule/uefi-self-test:156507): CRITICAL **: 11:28:59.325: failed to allocate dbus proxy object: Error calling StartServiceByName for com.intel.tss2.Tabrmd: Timeout was reached
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――

Summary of Failures:

31/32 lenovo-thinklmi-self-test      FAIL            25.44s   killed by signal 5 SIGTRAP
32/32 uefi-self-test                 FAIL            25.44s   killed by signal 5 SIGTRAP

Ok:                 30
Expected Fail:      0
Fail:               2
Unexpected Pass:    0
Skipped:            0
Timeout:            0
```

superm1, 2021-11-15:
Do you have tpm2-abrmd installed?  If so try to remove tpm2-abrmd to see it helps or not.

superm1, 2021-11-15:
Some other issues that happened from that:
https://github.com/fwupd/fwupd/issues/1634
https://github.com/tpm2-software/tpm2-abrmd/issues/529
https://github.com/fwupd/fwupd/issues/1549

We have no dependency on it.

hughsie, 2021-11-15:
I'm really wondering if tpm2-abrmd is a sane thing to depend on.

superm1, 2021-11-15:
> I'm really wondering if tpm2-abrmd is a sane thing to depend on.

We don't depend on it.

hughsie, 2021-11-15:
gah, i mean the tss2_esys stuff we do use. It seems to break pretty frequently.

superm1, 2021-11-15:
> gah, i mean the tss2_esys stuff we do use. It seems to break pretty frequently.

I don't think it breaks very frequently.  I think we had some growing pains initially, but it's been stable for a while no?

kloczek, 2021-11-19:
Just tested 1.7.2 with or without installed tpm2-abrmd result seems is the same (below is without installed tpm2-abrmd):
```console
+ cd fwupd-1.7.2
+ /usr/bin/meson test -C x86_64-redhat-linux-gnu --num-processes 12 --print-errorlogs
ninja: Entering directory `/home/tkloczko/rpmbuild/BUILD/fwupd-1.7.2/x86_64-redhat-linux-gnu'
ninja: no work to do.
 1/32 fwupd-exported-api                     OK               0.06s
 2/32 fwupdplugin-exported-api               OK               0.02s
 3/32 fwupd-context-test                     OK               0.12s
 4/32 fwupd-self-test                        OK               0.13s
 5/32 fwupd-thread-test                      OK               0.15s
 6/32 fwupdplugin-self-test                  OK               2.50s
 7/32 fu-self-test                           OK               3.91s
 8/32 acpi-dmar-self-test                    OK               0.11s
 9/32 acpi-facp-self-test                    OK               0.11s
10/32 acpi-phat-self-test                    OK               0.11s
11/32 bcm57xx-self-test                      OK               0.09s
12/32 ccgx-self-test                         OK               0.09s
13/32 fu-dfu-self-test                       OK               0.07s
14/32 logitech-hidpp-self-test               OK               0.04s
15/32 elantp-self-test                       OK               0.13s
16/32 linux-swap-self-test                   OK               0.14s
17/32 nitrokey-self-test                     OK               0.10s
18/32 pxi-self-test                          OK               0.08s
19/32 synaptics-prometheus-self-test         OK               0.06s
20/32 synaptics-mst-self-test                OK               0.10s
21/32 synaptics-rmi-self-test                OK               0.09s
22/32 vli-self-test                          OK               0.04s
23/32 uefi-dbx-self-test                     OK               0.04s
24/32 wacom-usb-self-test                    OK               0.05s
25/32 nvme-self-test                         FAIL             0.66s   killed by signal 5 SIGTRAP
>>> G_TEST_BUILDDIR=/home/tkloczko/rpmbuild/BUILD/fwupd-1.7.2/x86_64-redhat-linux-gnu/plugins/nvme G_TEST_SRCDIR=/home/tkloczko/rpmbuild/BUILD/fwupd-1.7.2/plugins/nvme MALLOC_PERTURB_=38 /home/tkloczko/rpmbuild/BUILD/fwupd-1.7.2/x86_64-redhat-linux-gnu/plugins/nvme/nvme-self-test
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――― ✀  ―――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――
stdout:
# random seed: R02Sbe3ab839f6576b09b46fcb7c41213a1b
1..2
# Start of fwupd tests
# GLib-GIO-DEBUG: _g_io_module_get_default: Found default implementation gvfs (GDaemonVfs) for ?gio-vfs?
# XbSilo-DEBUG: attempting to load /tmp/.0ICBD1
# XbSilo-DEBUG: failed to load silo: blob too small
Bail out! FuQuirks-FATAL-WARNING: [INTEL_SPI_CHIPSET\APL] Flags = PCH is invalid: P is not lowercase
stderr:

(/home/tkloczko/rpmbuild/BUILD/fwupd-1.7.2/x86_64-redhat-linux-gnu/plugins/nvme/nvme-self-test:3878819): FuQuirks-WARNING **: 10:36:28.781: [INTEL_SPI_CHIPSET\APL] Flags = PCH is invalid: P is not lowercase
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――

26/32 ata-self-test                          FAIL             0.82s   killed by signal 5 SIGTRAP
>>> G_TEST_SRCDIR=/home/tkloczko/rpmbuild/BUILD/fwupd-1.7.2/plugins/ata MALLOC_PERTURB_=166 G_TEST_BUILDDIR=/home/tkloczko/rpmbuild/BUILD/fwupd-1.7.2/x86_64-redhat-linux-gnu/plugins/ata /home/tkloczko/rpmbuild/BUILD/fwupd-1.7.2/x86_64-redhat-linux-gnu/plugins/ata/ata-self-test
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――― ✀  ―――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――
stdout:
# random seed: R02S11d255f1f205d3e35bbba7986a048554
1..2
# Start of fwupd tests
# Start of ata tests
# GLib-GIO-DEBUG: _g_io_module_get_default: Found default implementation gvfs (GDaemonVfs) for ?gio-vfs?
# XbSilo-DEBUG: attempting to load /tmp/.P03BD1
# XbSilo-DEBUG: failed to load silo: blob too small
Bail out! FuQuirks-FATAL-WARNING: [INTEL_SPI_CHIPSET\APL] Flags = PCH is invalid: P is not lowercase
stderr:

(/home/tkloczko/rpmbuild/BUILD/fwupd-1.7.2/x86_64-redhat-linux-gnu/plugins/ata/ata-self-test:3878788): FuQuirks-WARNING **: 10:36:28.691: [INTEL_SPI_CHIPSET\APL] Flags = PCH is invalid: P is not lowercase
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――

27/32 dell-self-test                         FAIL             0.83s   killed by signal 5 SIGTRAP
>>> G_TEST_SRCDIR=/home/tkloczko/rpmbuild/BUILD/fwupd-1.7.2/plugins/dell MALLOC_PERTURB_=155 G_TEST_BUILDDIR=/home/tkloczko/rpmbuild/BUILD/fwupd-1.7.2/x86_64-redhat-linux-gnu/plugins/dell FWUPD_LOCALSTATEDIR=/tmp/fwupd-self-test/var /home/tkloczko/rpmbuild/BUILD/fwupd-1.7.2/x86_64-redhat-linux-gnu/plugins/dell/dell-self-test
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――― ✀  ―――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――
stdout:
# random seed: R02Sbb7b86acfaa225749ba0a1855c498e72
# GLib-GIO-DEBUG: _g_io_module_get_default: Found default implementation gvfs (GDaemonVfs) for ?gio-vfs?
# XbSilo-DEBUG: attempting to load /tmp/.K6KBD1
[... 145 lines cut ...]
[... cut ...]
28/32 lenovo-thinklmi-self-test      FAIL             0.83s   killed by signal 5 SIGTRAP
29/32 uefi-self-test                 FAIL             0.69s   killed by signal 5 SIGTRAP
30/32 redfish-self-test              FAIL             0.77s   killed by signal 5 SIGTRAP
31/32 tpm-self-test                  FAIL             0.72s   killed by signal 5 SIGTRAP
32/32 thunderbolt-self-test          FAIL             0.75s   killed by signal 5 SIGTRAP

Ok:                 24
Expected Fail:      0
Fail:               8
Unexpected Pass:    0
Skipped:            0
Timeout:            0
```

hughsie, 2021-11-19:
I'm guessing https://github.com/fwupd/fwupd/pull/4011 fixes some of those.

kloczek, 2021-11-19:
Looks like yes :)
Result with dbca6f1b:
```console
+ /usr/bin/meson test -C x86_64-redhat-linux-gnu --num-processes 12 --print-errorlogs
ninja: Entering directory `/home/tkloczko/rpmbuild/BUILD/fwupd-1.7.2/x86_64-redhat-linux-gnu'
ninja: no work to do.
 1/32 fwupd-exported-api                     OK               0.05s
 2/32 fwupdplugin-exported-api               OK               0.02s
 3/32 fwupd-context-test                     OK               1.97s
 4/32 fwupd-thread-test                      OK               2.02s
 5/32 fwupd-self-test                        OK               2.03s
 6/32 fwupdplugin-self-test                  OK               2.66s
 7/32 fu-self-test                           OK               5.73s
 8/32 acpi-dmar-self-test                    OK               0.11s
 9/32 acpi-facp-self-test                    OK               0.11s
10/32 acpi-phat-self-test                    OK               0.10s
11/32 bcm57xx-self-test                      OK               0.09s
12/32 ccgx-self-test                         OK               0.08s
13/32 fu-dfu-self-test                       OK               0.07s
14/32 logitech-hidpp-self-test               OK               0.03s
15/32 elantp-self-test                       OK               0.13s
16/32 linux-swap-self-test                   OK               0.14s
17/32 nitrokey-self-test                     OK               0.12s
18/32 pxi-self-test                          OK               0.10s
19/32 synaptics-prometheus-self-test         OK               0.10s
20/32 synaptics-rmi-self-test                OK               0.12s
21/32 dell-self-test                         OK               0.26s
22/32 synaptics-mst-self-test                OK               0.16s
23/32 ata-self-test                          OK               0.30s
24/32 uefi-dbx-self-test                     OK               0.07s
25/32 nvme-self-test                         OK               0.19s
26/32 lenovo-thinklmi-self-test              OK               0.25s
27/32 vli-self-test                          OK               0.05s
28/32 wacom-usb-self-test                    OK               0.03s
29/32 redfish-self-test                      OK               0.27s
30/32 uefi-self-test                         OK               0.19s
31/32 tpm-self-test                          OK               0.24s
32/32 thunderbolt-self-test                  OK               4.15s

Ok:                 32
Expected Fail:      0
Fail:               0
Unexpected Pass:    0
Skipped:            0
Timeout:            0
```
Thank you.
Closing.

closed 2021-11-19 by kloczek