fwupd/fwupd #3992: Connection refused when trying fwupdmgr refresh

nsauter, 2021-11-15:
**Describe the question**
Is there a Proxy Option? I did not find anything in the `fwupdmgr --help` page so i assume there is none even though it would be very helpful.

**fwupd version information**
```
client version:	1.5.11
compile-time dependency versions
	gusb:	0.3.4

daemon version:	1.5.11
```

hughsie, 2021-11-15:
It's all the standard libcurl ones, so `http_proxy` and friends. See https://github.com/fwupd/fwupd/blob/main/docs/env.md

nsauter, 2021-11-15:
Yeah, thats how i solved my issue right now, but what if my system doesnt need a proxy and if i add one just for `fwupd` my other connections break?

hughsie, 2021-11-15:
you can run it per command too, e.g. `http_proxy=http://foobarbaz:8080 fwupdmgr refresh`

closed 2021-11-15 by hughsie

nsauter, 2021-11-15:
Okay, well. Then i will try this. Thanks.