fwupd/fwupd #4000: STM32F745 failed to wait for detach replug

amfern, 2021-11-16:
**Describe the bug**
Flashing successful, I see error warning, and it makes the process wait for 10 seconds
```
failed to get device before update reload: failed to wait for detach replug: device b0a78eb71f4eeea7df8fb114522556ba8ce22074 did not come back
```

**Steps to Reproduce**
add `Flags = absent-sector-size,will-disappear` to plugins/dfu/dfu.quirk
so it will look like this

```
# STM32F745 dfuse bootloader
[USB\VID_0483&PID_DF11]
Flags = absent-sector-size,will-disappear
Plugin = dfu
DfuForceVersion = 011a
DfuForceTimeout = 5000
```
run the fwupdtool tool
`sudo fwupdtool --verbose install-blob --plugins dfu --no-reboot-check obj/fcos_v3.4.0-a1_HUNTER_V0_DIGITAL_KAKUTEF7.dfu`

**Expected behavior**
fwupdtool exits after flashing without waiting for device to come back and without showing any warning 

**fwupd version information**
```shell
client version:	1.7.1-83-gb9024e80a
compile-time dependency versions
	gusb:	0.3.8

daemon version:	1.7.1-83-gb9024e80a
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): source

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
G5 5590
│
├─BC511 NVMe SK hynix 512GB:
│     Device ID:          63fee790cbe11435e2e48f7408a087565ba0997e
│     Summary:            NVM Express solid state drive
│     Current version:    11001101
│     Vendor:             SK hynix (NVME:0x1C5C)
│     GUIDs:              71575487-e6fa-52be-a95f-62cd22ec72fc
│                         8811e962-e1d5-e811-9f8b-f2801f1b9fd1
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Core™ i7-9750H CPU @ 2.60GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000ea
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         809a0b93-8a12-5338-a571-ad5583acf896
│                         72ec2ff3-49ff-5ec2-bdbb-525badd47543
│     Device Flags:       • Internal device
│   
├─SSD 870 EVO 1TB:
│     Device ID:          5dbeb140337f610d54913c0f43d15fdaa0eafb51
│     Summary:            ATA drive
│     Current version:    SVT01B6Q
│     Vendor:             Samsung (ATA:0x144D, OUI:002538)
│     GUIDs:              812b49e2-b9e4-57ce-a2f1-7941aba15a46
│                         0d5afc53-059c-5a53-a4a4-779643686396
│                         9302adb5-953b-58eb-a606-f0fadd98f5e0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─STM32 BOOTLOADER:
│     Device ID:          b0a78eb71f4eeea7df8fb114522556ba8ce22074
│     Current version:    22.0
│     Vendor:             STMicroelectronics (USB:0x0483)
│     GUIDs:              f7d494e1-3327-5c82-8227-fdb053fce37c
│                         5aea1200-2286-5819-93e6-3b5548b6c88a
│     Device Flags:       • Updatable
│                         • Is in bootloader mode
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    68353
│ │   Minimum Version:    68353
│ │   Vendor:             Dell Inc. (DMI:Dell Inc.)
│ │   Update State:       Success
│ │   GUIDs:              f19aad3e-4f15-4f4d-8518-e7c67cbb777d
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       921af5e5-3d5b-552d-962e-8488d4bb8b20
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  83
│       Minimum Version:  83
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            00fe3755-a4d8-5ef7-ba5f-47979fbb3423
│                         4a6cd2cb-8741-5257-9d1f-89a275dacca7
│                         c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Only version upgrades are allowed
│     
├─TPM:
│     Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│     Summary:            TPM 2.0 Device
│     Current version:    403.1.0.0
│     Vendor:             Intel (TPM:INTC)
│     GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6
│                         34801700-3a50-5b05-820c-fe14580e4c2d
│                         8e1cbc5d-5a11-5149-bfea-b6065d5296ba
│                         03f304f4-223e-54f4-b2c1-c3cf3b5817c6
│                         52d7b679-db28-5bf7-bd87-41d77aeec600
│     Device Flags:       • Internal device
│   
├─TU106M [GeForce RTX 2070 Mobile]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a1
│     Vendor:             NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:              b2347777-f760-55e2-8d8d-2de9e4160afd
│                         bac6ddd9-ee60-52fa-a57e-83e0ee99e648
│                         15e2e65e-fea0-559c-a35e-77f2b3a02795
│                         efb170e4-88c1-5421-bc5f-6bf7ae111797
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
└─Thunderbolt host controller:
      Device ID:          31a852b41a0465049a3054132e9a44ff12ef5cd2
      Summary:            Unmatched performance for high-speed I/O
      Current version:    40.00
      Vendor:             Dell (THUNDERBOLT:0x00D4, TBT:0x00D4)
      GUIDs:              1e863ecb-8ae8-5501-989d-aa0ca6efd19e
                          c5a3688f-5c1d-52db-bfd6-4d8bf7c12d75
                          0660c0bd-ec2e-5588-b32a-21c924b0ac2d
                          2b94a6aa-b12d-5794-aa7f-b9e516fa0a9e
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Device stages updates
```

**Additional questions**

- Operating system and version: archlinux
- Have you tried rebooting? yes
- Is this a regression? no

hughsie, 2021-11-18:
Rather than `will-disappear`, can you try `no-dfu-runtime` instead?

amfern, 2021-11-19:
`no-dfu-runtime` didn't work either.
[Uploading out.log…]()

hughsie, 2021-11-19:
@amfern can you try https://github.com/fwupd/fwupd/pull/4012  please.

amfern, 2021-11-19:
with branch from #4012 i don't see the warnning anymore 
```
sudo fwupdtool install-blob --plugins dfu obj/fcos_v3.4.0-a1_HUNTER_V0_DIGITAL_KAKUTEF7.dfu
[sudo] password for ilya: 
14:45:22:0875 FuQuirks             using DeviceInstanceId= for DeviceInstanceId=USB\VID_0A12&PID_1337 in quirk files is deprecated!
14:45:22:0875 FuQuirks             using DeviceInstanceId= for DeviceInstanceId=USB\VID_0A12&PID_1337&REV_2520 in quirk files is deprecated!
14:45:22:0875 FuQuirks             using DeviceInstanceId= for DeviceInstanceId=USB\VID_0A12&PID_4004 in quirk files is deprecated!
14:45:22:0908 FuQuirks             using Guid= for Guid=8265d473-a6c2-42b4-897b-bc220faa2d32 in quirk files is deprecated!
14:45:22:0909 FuQuirks             using Guid= for Guid=797f8bae-0ea2-4c0f-8a30-7d10ccfacbc0 in quirk files is deprecated!
14:45:22:0909 FuQuirks             using HwId= for HwId=386c2f52-44ec-55d3-b802-47631bfb8451 in quirk files is deprecated!
14:45:22:0909 FuQuirks             using Guid= for Guid=28108d08-5027-42c2-a5b8-92d6ede9b97b in quirk files is deprecated!
Loading…                 [***************************************]
WARNING: This package has not been validated, it may not work properly.
Selected device: STM32 BOOTLOADER
Waiting…                 [***************************************] Less than one minute remaining…
flightcontroller_mark3 on  development [$] via ⍱ took 43s ❯ 
```

closed 2021-11-24 by hughsie